GRIDAPPSD/gridappsd-forum #2: Having problem running the sample app in IEEE 8500  new model

shpoudel, 2019-06-28:
I am having issues running the sample_app for IEEE 8500 new model. I think my issue lies after calling function "gapps.subscribe(listening_to_topic, toggler)". I couldn't quite follow how the instance of GridAPPSD (gapps) is used for subscribing and how it reaches on_message function.

poorva1209, 2019-06-28:
The new 8500 model not yet ready for full simulation run. It will be available with release next week. Can you try your code with old 8500 model and see if you recieve simulation output.

shpoudel, 2019-06-30:
Yes, with old 8500 model, I can receive simulation output.

poorva1209, 2019-08-01:
Please test the new 9500 model in v2019.07.0 and develop tag. Closing this issue.

closed 2019-08-01 by poorva1209