GRIDAPPSD/gridappsd-forum #3: Beta release issue

shpoudel, 2019-07-03:
The beta version works with new IEEE 8500 model and running the default sample-app. However, when I tried to incorporate my change in runsample.py, I get an error which I have attached here as an image.

I also changed docker-compose.yml file accordingly (Line 5): image: sample-app

[image]

shpoudel, 2019-07-03:
This error was gone when I restart the system. Now, just one quick question, why am I not getting any power flow plot in the simulation. Also, if plot appears, which line flow it corresponds to? Because, the system has three substation now instead of single one as in IEEE 123 or original IEEE 8500 node model.

poorva1209, 2019-07-03:
@shpoudel Power plots will be there in the final release this week. @bibiraju : Can you confirm this?

@bibiraju Can we show exactly what we are plotting in UI ?

shpoudel, 2019-07-03:
@poorva1209 Thanks for the reply. Can we have several plots for power flow while displaying? For example, my application will require the display of at least three plots (Power A, Power B, and Power C) for three substations which sum up to 9. Or, Maybe the ABC phase plot can go in the same figure and three such figures for each substation?

poorva1209, 2019-07-03:
@shpoudel Yeah, we are working towards that. It is just that other changes in Viz are taking priority over this. But we will have the option of custom plots included soon.

bibiraju, 2019-07-03:
@shpoudel You see the voltage plots for below components    
      'l2673313',
      'l2876814',
      'l2955047',
      'l3160107',
      'l3254238',
      'm1047574'

The power plots will be available in the next release. As Poorva mentioned, the custom plotting will be added to the visualization soon as well.

shpoudel, 2019-07-03:
Thank you @poorva1209 @bibiraju for the prompt reply. One more question; @poorva1209 Could you please point which code in the respository of gridappsd-python shall I look into for getting the real-time status of any components while the simulation is running? For example, can I create a specific event by opening one switch and read it via the platform? I believe the event option in simulation is for something like this, correct?

poorva1209, 2019-07-03:
- which code in the respository of gridappsd-python shall I look into for getting the real-time status of any components while the simulation is running? 
`You can look into sample app to subscribe to simulation output and check the value for the mrid of the measurement you are interested in.` https://github.com/GRIDAPPSD/gridappsd-sample-app/blob/master/sample_app/runsample.py. `If you are not sure how to query the  mrid of the measurement you are interested in or how to check its value in the simulation output then we can cover that on a call. `
- can I create a specific event by opening one switch and read it via the platform? 
`You can open/close a switch on UI where it is displaying the network model. It shows switches and you can click on them to open/close them. `
- I believe the event option in simulation is for something like this, correct?
`By Event the platform means either Communication Outage or Faults. You can insert lineToGround, LineToLine and lineToLinetoGround fault with switches. `

poorva1209, 2019-08-05:
Closing this as it does not contain any open issues.

closed 2019-08-05 by poorva1209