GRIDAPPSD/gridappsd-forum #4: Query regarding Missing Simulation ID

ArunWSU, 2019-07-03:
Hi,
 I'm Arun, an app developer from Washington State University. I'm having problems querying into the simulation data. Although I tried to execute the query shown in attachment, after the simulation was completed, I get SimulationID not found error. Also, If I go into the "Logs" of "Browse data", sometimes I don't see the simulation ID of the recent execution even after I tried to type in. And, another generic question is sometimes I get watching 201, 400 on my terminal. What does that mean ? Additionally, a general observation of latest 2019.06 release is that running 13busckt simulation is that I get a 650oz simulation output missing warning and the output graph is not visible. And, a minor glitch is that when I try to type in queries in STOMP client, the cursor  keeps automatically moving to the index. Of course, I am following a quick workaround by typing in Notepad and copying the same.
Regards,
Arun

[image]

poorva1209, 2019-07-03:
-  I'm having problems querying into the simulation data. Although I tried to execute the query shown in attachment, after the simulation was completed, I get SimulationID not found error. 
 `We are transitioning from beta relase to final so the query filters are not updated in documentation but are updated in backend. Please try simulation_id instead of hasSimulationId in your query.` 
- If I go into the "Logs" of "Browse data", sometimes I don't see the simulation ID of the recent execution even after I tried to type in.
@bibiraju `Can you look into this issue`.
- sometimes I get watching 201, 400 on my terminal. What does that mean ? 
`These are REST call status messages from storing timeseries data. We are working on removing them from terminal display in future releases. `
- 13busckt simulation is that I get a 650oz simulation output missing warning and the output graph is not visible.
`I will run this locally and reproduce the error. I will assign it to the right person to resolve it. `
- a minor glitch is that when I try to type in queries in STOMP client, the cursor keeps automatically moving to the index
@bibiraju `Can you check and rectify this?`

@ArunWSU  I would suggest making a different issue for different problems. So we can better track and assign them.

poorva1209, 2019-08-01:
> If I go into the "Logs" of "Browse data", sometimes I don't see the simulation ID of the recent execution even after I tried to type in 

This is in process under this issue: https://github.com/gridappsd/goss-gridapps-d/issues/816

poorva1209, 2019-08-19:
Closing this task as all the issues mentioned are resolved.

closed 2019-08-19 by poorva1209