GRIDAPPSD/gridappsd-forum #5: Unable to run 13 bus and 123 bus pv simulations

ArunWSU, 2019-07-04:
Hi,
 Just a few moments ago, I am unable to run the 13bus and 123buspv simulations. I get a FNCS initialization failed error. The one with broker and values in images corresponds to 13 bus and second one with just THREAD-69 ERROR corresponds to 123pv system.
[Jul3_13bus_simulationerror.png](https://images.zenhubusercontent.com/5d1bf0f68871121fe293f312/aac323aa-7b63-4f5b-b029-7c6edc43b005)
[image]
Regards,
Arun

poorva1209, 2019-07-05:
@ArunWSU  I ran v2019.06.beta with both models and they are running for me. Send me steps you followed and I will do it again. 
Which branch of gridappsd-docker?
Did you run ./stop.sh -c 
Then ./run.sh -t v2019.06.beta ?
Any things else selected along with model, like app? weather? houses?

ArunWSU, 2019-07-05:
I am sorry. I found the issue. I was still working on the last commit of the develop branch and trying to run simulations. If I use the Beta version, it fixed the issue. Thanks

closed 2019-07-05 by ArunWSU