GRIDAPPSD/gridappsd-forum #7: Timeseries database output data model has too much nesting

blthayer, 2019-07-25:
When querying the timeseries database via the GridAPPS-D API, the returned json object has a lot of unnecessary nesting which makes parsing it a chore. Unfortunately Zenhub/Github won't let me attach a json file (seriously?!), but I've put a small sample below.

The following sample is the result for getting **a single measurement** from the database:
```json
{
  "data": {
    "measurements": [
      {
        "name": "simulation",
        "points": [
          {
            "row": {
              "entry": [
                {
                  "key": "hasSimulationMessageType",
                  "value": "OUTPUT"
                },
                {
                  "key": "measurement_mrid",
                  "value": "_cb8ec914-b028-4270-bb39-939925d1c820"
                },
                {
                  "key": "angle",
                  "value": "-150.6108656889916"
                },
                {
                  "key": "magnitude",
                  "value": "122.19583713288947"
                },
                {
                  "key": "simulation_id",
                  "value": "1474465980"
                },
                {
                  "key": "time",
                  "value": "1358121602"
                }
              ]
            }
          }
        ]
      }
    ]
  },
  "responseComplete": true,
  "id": "101085167"
}
```
Here's an example of the code I had to write to parse it:
```python
def parse_timeseries(data):
    """Helper to parse platform timeseries data.

    :param data: dictionary with results from calling the timeseries
        API (either for weather data or simulation data). Ultimately,
        this is a return from gridappsd.GridAPPSD.get_response. It is
        assumed that data['data'] is not None, as that check is done
        at a higher level. See gridappsd_platform.py.
    """
    # Ensure data is a dictionary. We won't check its integrity, and
    # let a KeyError get raised if it's incorrectly formatted.
    if not isinstance(data, dict):
        raise TypeError('data must be a dictionary!')

    # The data formatting/data model returned from the platform's time
    # series database is pretty ridiculous. There's all sorts of
    # unnecessary nesting. Since this nesting is present, we have a lot
    # of safety checks we need to do. To keep the code clean/short, I'm
    # just going to use asserts.
    assert 'data' in data.keys()
    assert 'measurements' in data['data'].keys()
    assert len(data['data']) == 1
    assert len(data['data']['measurements']) == 1
    assert 'name' in data['data']['measurements'][0].keys()
    assert 'points' in data['data']['measurements'][0].keys()
    assert isinstance(data['data']['measurements'][0]['points'], list)
    
    # Alrighty, there's our initial checks. Now move on.

    # Initialize our data list. This is eventually going to be used to
    # create a Pandas DataFrame.
    dl = []
    # Loop over the "points," which contain the data we need.
    for point in data['data']['measurements'][0]['points']:
        # We're expecting a dictionary with a single entry.
        assert len(point) == 1

        # Extract the row.
        row = point['row']

        # Again, we're expecting a dictionary with a single entry.
        assert len(row) == 1

        # Extract the "entry"
        entry = row['entry']

        # Create a dictionary for this entry.
        this_entry = {}

        # Loop over the values in entry.
        for d in entry:
            # We should have just 'key' and 'value' keys.
            assert len(d) == 2

            # Use the key as well, a key, and the value as well, a
            # value. The fact that I have to do this seems crazy.
            this_entry[d['key']] = d['value']

        # Append to our list.
        dl.append(this_entry)

    # Create our DataFrame.
    df = pd.DataFrame(dl)
    # Index it by time.
    return df.set_index(keys='time')
```
I don't have time to propose a new schema, but you can see from both the json and the code that most of the nesting is truly pointless.

blthayer, 2019-08-13:
@ericstephan @poorva1209 @craig8 

I've just started using `v2018.08.0` which I saw in the release notes flattens the response from the timeseries API. 

It would appear there are duplicate fields `data` and `measurements`. Here's a Python console session where I'm debugging the `parse_timeseries` method I gave above in the original issue:

```python
>>> data.keys()
dict_keys(['data', 'responseComplete', 'id'])
>>> data['data'].keys()
dict_keys(['data', 'measurements'])
>>> data['data']['data'] == data['data']['measurements']
True
```

If you want me to file a separate ticket for this bug, let me know. For now, I'm planning to use the `measurements` field.

blthayer, 2019-08-13:
@ericstephan @poorva1209 @craig8 
I found a bug - the API is returning duplicates of the same measurement output over and over and over. Evidence (again from debugging `parse_timeseries`):

```python
>>> d0 = data['data']['measurements'][0]
>>> for d in data['data']['measurements'][1:]:
...     print(d == d0)
... 
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
True
```

blthayer, 2019-08-13:
Adding app-critical and bug labels as this update seems to have broken things.

ericstephan, 2019-08-15:
I've checked in the code and talked to Poorva about this.   Poorva asked me to remove the "data" root object and pass back just the array of results back.  I can send same results if you are interested.

blthayer, 2019-08-15:
@ericstephan, thank you! 

@afisher1 was going to talk to @tdtalbot about the bug which returns the same measurement over and over - any update on when that piece of things might be fixed?

poorva1209, 2019-08-17:
@blthayer Eric has made changes to flatten the json as well as resolve the bug that is returning same measurement or weather from timeseries data store. I couldn't test it out as I am on travel . I will update this first thing on Monday. 

@afisher1 @tdtalbot Just FYI.

ericstephan, 2019-08-17:
I will be available Monday when/if anything is needed.

Eric

Jeffrey-Simpson, 2019-08-19:
I am getting errors on the query:
{"queryMeasurement":"weather",
"queryFilter":{"startTime":"1357048800000000",
                        "endTime":"1357048860000000"},
"responseFormat":"JSON"}

```
{
    "error": {
        "message": "com.google.gson.JsonSyntaxException: java.lang.IllegalStateException: Expected BEGIN_OBJECT but was BEGIN_ARRAY at line 1 column 2 path $\n\tat com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$Adapter.read(ReflectiveTypeAdapterFactory.java:200)\n\tat com.google.gson.Gson.fromJson(Gson.java:810)\n\tat com.google.gson.Gson.fromJson(Gson.java:775)\n\tat com.google.gson.Gson.fromJson(Gson.java:724)\n\tat com.google.gson.Gson.fromJson(Gson.java:696)\n\tat gov.pnnl.goss.gridappsd.dto.TimeSeriesEntryResult.parse(TimeSeriesEntryResult.java:36)\n\tat gov.pnnl.goss.gridappsd.data.ProvenTimeSeriesDataManagerImpl.query(ProvenTimeSeriesDataManagerImpl.java:124)\n\tat gov.pnnl.goss.gridappsd.data.ProvenTimeSeriesDataManagerImpl.handle(ProvenTimeSeriesDataManagerImpl.java:112)\n\tat gov.pnnl.goss.gridappsd.data.DataManagerImpl.processDataRequest(DataManagerImpl.java:149)\n\tat gov.pnnl.goss.gridappsd.process.ProcessEvent.onMessage(ProcessEvent.java:213)\n\tat pnnl.goss.core.client.GossClient$1.run(GossClient.java:321)\n\tat java.lang.Thread.run(Thread.java:748)\nCaused by: java.lang.IllegalStateException: Expected BEGIN_OBJECT but was BEGIN_ARRAY at line 1 column 2 path $\n\tat com.google.gson.stream.JsonReader.beginObject(JsonReader.java:387)\n\tat com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$Adapter.read(ReflectiveTypeAdapterFactory.java:189)\n\t... 11 more\n"
    },
    "responseComplete": true,
    "id": "eb2bf753-145f-4472-812f-a1700817948e"
}
```

On the console I get:
```
Thread-46 ERROR [gov.pnnl.goss.gridappsd.log.LogManagerImpl] - 1566230284286|ProcessEvent|1331885624|ERROR|system|ERROR
com.google.gson.JsonSyntaxException: java.lang.IllegalStateException: Expected BEGIN_OBJECT but was BEGIN_ARRAY at line 1 column 2 path $
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$Adapter.read(ReflectiveTypeAdapterFactory.java:200)
	at com.google.gson.Gson.fromJson(Gson.java:810)
	at com.google.gson.Gson.fromJson(Gson.java:775)
	at com.google.gson.Gson.fromJson(Gson.java:724)
	at com.google.gson.Gson.fromJson(Gson.java:696)
	at gov.pnnl.goss.gridappsd.dto.TimeSeriesEntryResult.parse(TimeSeriesEntryResult.java:36)
	at gov.pnnl.goss.gridappsd.data.ProvenTimeSeriesDataManagerImpl.query(ProvenTimeSeriesDataManagerImpl.java:124)
	at gov.pnnl.goss.gridappsd.data.ProvenTimeSeriesDataManagerImpl.handle(ProvenTimeSeriesDataManagerImpl.java:112)
	at gov.pnnl.goss.gridappsd.data.DataManagerImpl.processDataRequest(DataManagerImpl.java:149)
	at gov.pnnl.goss.gridappsd.process.ProcessEvent.onMessage(ProcessEvent.java:213)
	at pnnl.goss.core.client.GossClient$1.run(GossClient.java:321)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.IllegalStateException: Expected BEGIN_OBJECT but was BEGIN_ARRAY at line 1 column 2 path $
	at com.google.gson.stream.JsonReader.beginObject(JsonReader.java:387)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$Adapter.read(ReflectiveTypeAdapterFactory.java:189)
	... 11 more

Thread-44 ERROR [gov.pnnl.goss.gridappsd.log.LogManagerImpl] - 1566230284286|ProcessEvent|1331885624|ERROR|system|ERROR
com.google.gson.JsonSyntaxException: java.lang.IllegalStateException: Expected BEGIN_OBJECT but was BEGIN_ARRAY at line 1 column 2 path $
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$Adapter.read(ReflectiveTypeAdapterFactory.java:200)
	at com.google.gson.Gson.fromJson(Gson.java:810)
	at com.google.gson.Gson.fromJson(Gson.java:775)
	at com.google.gson.Gson.fromJson(Gson.java:724)
	at com.google.gson.Gson.fromJson(Gson.java:696)
	at gov.pnnl.goss.gridappsd.dto.TimeSeriesEntryResult.parse(TimeSeriesEntryResult.java:36)
	at gov.pnnl.goss.gridappsd.data.ProvenTimeSeriesDataManagerImpl.query(ProvenTimeSeriesDataManagerImpl.java:124)
	at gov.pnnl.goss.gridappsd.data.ProvenTimeSeriesDataManagerImpl.handle(ProvenTimeSeriesDataManagerImpl.java:112)
	at gov.pnnl.goss.gridappsd.data.DataManagerImpl.processDataRequest(DataManagerImpl.java:149)
	at gov.pnnl.goss.gridappsd.process.ProcessEvent.onMessage(ProcessEvent.java:213)
	at pnnl.goss.core.client.GossClient$1.run(GossClient.java:321)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.IllegalStateException: Expected BEGIN_OBJECT but was BEGIN_ARRAY at line 1 column 2 path $
	at com.google.gson.stream.JsonReader.beginObject(JsonReader.java:387)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$Adapter.read(ReflectiveTypeAdapterFactory.java:189)
	... 11 more

```

poorva1209, 2019-08-19:
I am still testing this one. I will send out an email as soon as this is resolved in develop as well as release containers.

closed 2019-08-20 by poorva1209