GRIDAPPSD/gridappsd-forum #9: Unsuccessful queries in GridAPPS-D platform

ArunWSU, 2019-07-26:
HI,
 The Object dictionary query in powergrid API doesn't work even though I tried multiple Object types and Object id's(10 and the sensor simulator service (2) in timeseries API, when I tried to execute through STOMP client utility. For 2, the response I get is data:null, even though I get data through simulation query in same API. Can you tell me the difference between GridAPPS-D sensor simulation service and sensor data? Just so you know, sometimes I get no response from query execution, may be because of incorrect syntax and nothing shows up on terminal as well. So, if I refresh my browser, the previous simulation data is lost.

1. {  "requestType": "QUERY_OBJECT_DICT" ,"modelId": "_49AD8E07-3BF9-A4E2-CB8F-C3722F837B62" ,  "objectType":  "LoadBreakSwitch", "resultFormat": "JSON"}
2. {"queryMeasurement": "gridappsd-sensor-simulator","queryFilter": {"simulation_id": "1447666058"},"responseFormat": "JSON"}

Regards,
Arun

poorva1209, 2019-07-26:
1. To query model data change the topic in Stomp client to goss.gridappsd.process.request.data.powergridmodel 
2. To query timeseries data (simulation/service/weather) changethe topic in Stomp client to goss.gridappsd.process.request.data.timeseries

> Can you tell me the difference between GridAPPS-D sensor simulation service and sensor data? 

Not sure what you mean here. Simulation output is coming from GridLAB-D and sensor service output is coming from this service : https://github.com/GRIDAPPSD/gridappsd-sensor-simulator

> Just so you know, sometimes I get no response from query execution, may be because of incorrect syntax and nothing shows up on terminal as well. So, if I refresh my browser, the previous simulation data is lost.

Try with the topic changes. If you still don't receive a response back, send me the exact topic and query you are using. Viz app doesn't keep the data if the browser is refreshed.

ArunWSU, 2019-07-26:
Hi, 
Even in the first comment, I used the topics that you suggested as in documentation. But, 1 seems to be fixed now if I refresh and try again. 2 still doesn't work
Topic: goss.gridappsd.process.request.data.timeseries
Request:
{"queryMeasurement": "gridappsd-sensor-simulator","queryFilter": {"simulation_id": "33041270"},"responseFormat": "JSON"}
Response
{
    "data": null,
    "responseComplete": true,
    "id": "751866136"
}
But, this one works for the same timeseries topic
{"queryMeasurement": "simulation","queryFilter": {"simulation_id": "33041270"},"responseFormat": "JSON"}

poorva1209, 2019-07-30:
gridappsd-sensor-simulator doesn't run by default. Did you add it as a pre-requisite in sample_app config file?

poorva1209, 2019-08-08:
Documentation on how to start a service added here : https://gridappsd.readthedocs.io/en/master/using_gridappsd/index.html#how-to-start-a-service

closed 2019-08-19 by poorva1209