GRIDAPPSD/gridappsd-forum #10: Proven "number" filters could be more useful

blthayer, 2019-07-30:
[Documentation link](https://gridappsd.readthedocs.io/en/latest/using_gridappsd/index.html#timeseries-api)

For both the weather and simulation data time series queries, there are some of filters that are documented like `<measurement Field>[number]`. These `[number]` query filters appear to be for exact matches only, which is rarely useful (especially when simulation output is reported to an excessive precision as it is now. Side note: reducing simulator output precision could possibly help with throughput issues). It would be much more useful to have arithmetic comparisons in place. E.g. `give me all values < x` or `give me all values < x and > y`. 

@poorva1209 @ericstephan @craig8

poorva1209, 2019-08-01:
Similar for measurement mrid - useful to have multiple at the same time.

blthayer, 2019-08-01:
@ericstephan - let me know if you need more details, but here are some additional thoughts:

- The list size for measurement mrids should be unlimited if possible.
    - I think it's on the user to be smart about making queries that won't fill up their memory.
    - This is in line with other database API's I've used: they tend to give you a `fetchall` method which you use at your own risk. I don't think we should get too ambitious with our Python API in implementing some sort of `fetch_multiple` where you specify how many rows to grab at a time.
    - It would be nice if the server handled queries that were too long without the user knowing.
- For `number` fields, we should be able to specify all combinations of inequalities (e.g. openleft, openright, both closed, both open, less than, greater than).