GRIDAPPSD/gridappsd-forum #11: Make timestamps consistent across the platform

blthayer, 2019-07-30:
@poorva1209 - assigning to you for redirection as needed.

So far, I've come across three different timestamp conventions in the platform:

1. Header for simulation output (and probably other headers that go out on the GOSS bus): **milliseconds** since the epoch.
2. Simulation output message: **seconds** since the epoch.
3. Time filters for Proven queries: **microseconds** since the epoch. HOWEVER: Proven returns timestamps in the output in **seconds** since the epoch. This is exceptionally egregious.

While most applications won't care about anything sub-second, I know there are counter examples (like the UAF app), so perhaps we should standardize on microseconds since the epoch. Presumably we aren't using anything 32 bit?

poorva1209, 2019-10-08:
> 1. Header for simulation output (and probably other headers that go out on the GOSS bus): **milliseconds** since the epoch.

We won't control the header timestamp for now. 

> 2. Simulation output message: **seconds** since the epoch.

That will remain in second. 

3. Time filters for Proven queries: **microseconds** since the epoch. HOWEVER: Proven returns timestamps in the output in **seconds** since the epoch. This is exceptionally egregious.

Proven queries and response both changes to have timestamp in second. 

@Tonya: Tag proven:test as proven:develop. I have test it.

tonya1, 2019-10-08:
Merged #1123 and updated proven:develop

closed 2019-10-08 by tonya1