GRIDAPPSD/gridappsd-forum #14: Capacitors are not controllable in new 9500 node model

blthayer, 2019-07-30:
@poorva1209 @temcdrm @afisher1 

Given my best interpretation of CIM, it's my understanding that for a capacitor to be controllable, it must have an associated RegulatingControl object. All but one capacitor in the 8500 node model have RegulatingControl, but in the new 9500 model, none do.

Fire up the platform with v2019.06.beta. With your web browser, visit http://localhost:8889/bigdata/#query. Enter the following query (which can be found in the [Powergrid-Models repo](https://github.com/GRIDAPPSD/Powergrid-Models/blob/develop/blazegraph/queries.txt)):
```sparql
PREFIX r:  <http://www.w3.org/1999/02/22-rdf-syntax-ns#>
PREFIX c:  <http://iec.ch/TC57/CIM100#>
SELECT ?name ?basev ?nomu ?bsection ?bus ?conn ?grnd ?phs ?ctrlenabled ?discrete ?mode ?deadband ?setpoint ?delay ?monclass ?moneq ?monbus ?monphs ?id ?fdrid WHERE {
 ?s r:type c:LinearShuntCompensator.
# feeder selection options - if all commented out, query matches all feeders
VALUES ?fdrid {"_4F76A5F9-271D-9EB8-5E31-AA362D86F2C3"}  # 8500 node
 ?s c:Equipment.EquipmentContainer ?fdr.
 ?fdr c:IdentifiedObject.mRID ?fdrid.
 ?s c:IdentifiedObject.name ?name.
 ?s c:ConductingEquipment.BaseVoltage ?bv.
 ?bv c:BaseVoltage.nominalVoltage ?basev.
 ?s c:ShuntCompensator.nomU ?nomu. 
 ?s c:LinearShuntCompensator.bPerSection ?bsection. 
 ?s c:ShuntCompensator.phaseConnection ?connraw.
   bind(strafter(str(?connraw),"PhaseShuntConnectionKind.") as ?conn)
 ?s c:ShuntCompensator.grounded ?grnd.
 OPTIONAL {?scp c:ShuntCompensatorPhase.ShuntCompensator ?s.
 ?scp c:ShuntCompensatorPhase.phase ?phsraw.
   bind(strafter(str(?phsraw),"SinglePhaseKind.") as ?phs) }
 OPTIONAL {?ctl c:RegulatingControl.RegulatingCondEq ?s.
          ?ctl c:RegulatingControl.discrete ?discrete.
          ?ctl c:RegulatingControl.enabled ?ctrlenabled.
          ?ctl c:RegulatingControl.mode ?moderaw.
           bind(strafter(str(?moderaw),"RegulatingControlModeKind.") as ?mode)
          ?ctl c:RegulatingControl.monitoredPhase ?monraw.
           bind(strafter(str(?monraw),"PhaseCode.") as ?monphs)
          ?ctl c:RegulatingControl.targetDeadband ?deadband.
          ?ctl c:RegulatingControl.targetValue ?setpoint.
          ?s c:ShuntCompensator.aVRDelay ?delay.
          ?ctl c:RegulatingControl.Terminal ?trm.
          ?trm c:Terminal.ConductingEquipment ?eq.
          ?eq a ?classraw.
           bind(strafter(str(?classraw),"CIM100#") as ?monclass)
          ?eq c:IdentifiedObject.name ?moneq.
          ?trm c:Terminal.ConnectivityNode ?moncn.
          ?moncn c:IdentifiedObject.name ?monbus.
          }
 ?s c:IdentifiedObject.mRID ?id. 
 ?t c:Terminal.ConductingEquipment ?s.
 ?t c:Terminal.ConnectivityNode ?cn. 
 ?cn c:IdentifiedObject.name ?bus
}
ORDER by ?name
```

Results from running this query are below:
```
name | basev | nomu | bsection | bus | conn | grnd | phs | ctrlenabled | discrete | mode | deadband | setpoint | delay | monclass | moneq | monbus | monphs | id | fdrid
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
capbank0a | 12470 | 7200 | 0.0077160494 | r42246 | Y | true | A | true | true | reactivePower | -500000 | -50000 | 100 | ACLineSegment | cap_3a | q16642 | A | _2199D08B-9352-2085-102F-6B207E0BEBA3 | _4F76A5F9-271D-9EB8-5E31-AA362D86F2C3
capbank0b | 12470 | 7200 | 0.0077160494 | r42246 | Y | true | B | true | true | reactivePower | -500000 | -50000 | 101 | ACLineSegment | cap_3b | q16642 | B | _5C01CD02-852C-04A5-BA0B-D9C7B2094BB6 | _4F76A5F9-271D-9EB8-5E31-AA362D86F2C3
capbank0c | 12470 | 7200 | 0.0077160494 | r42246 | Y | true | C | true | true | reactivePower | -500000 | -50000 | 102 | ACLineSegment | cap_3c | q16642 | C | _9BCE2975-44FB-F2FC-5720-90A7C84F5BC4 | _4F76A5F9-271D-9EB8-5E31-AA362D86F2C3
capbank1a | 12470 | 7200 | 0.005787037 | r42247 | Y | true | A | true | true | reactivePower | -375000 | -37500 | 100 | ACLineSegment | cap_2a | q16483 | A | _57546BB3-C069-6088-1569-8166FDDA934F | _4F76A5F9-271D-9EB8-5E31-AA362D86F2C3
capbank1b | 12470 | 7200 | 0.005787037 | r42247 | Y | true | B | true | true | reactivePower | -375000 | -37500 | 101 | ACLineSegment | cap_2b | q16483 | B | _97E41339-8930-8B3B-421D-32DBAFCC2A3B | _4F76A5F9-271D-9EB8-5E31-AA362D86F2C3
capbank1c | 12470 | 7200 | 0.005787037 | r42247 | Y | true | C | true | true | reactivePower | -375000 | -37500 | 102 | ACLineSegment | cap_2c | q16483 | C | _05DB8C45-CF83-09F8-5E1C-85634A96720F | _4F76A5F9-271D-9EB8-5E31-AA362D86F2C3
capbank2a | 12470 | 7200 | 0.005787037 | r20185 | Y | true | A | true | true | reactivePower | -375000 | -37500 | 100 | ACLineSegment | cap_1a | l2823592 | A | _174D9F0D-5699-FB32-FAE0-226544F12364 | _4F76A5F9-271D-9EB8-5E31-AA362D86F2C3
capbank2b | 12470 | 7200 | 0.005787037 | r20185 | Y | true | B | true | true | reactivePower | -375000 | -37500 | 101 | ACLineSegment | cap_1b | l2823592 | B | _A255AECD-B084-B740-D23B-21160C831719 | _4F76A5F9-271D-9EB8-5E31-AA362D86F2C3
capbank2c | 12470 | 7200 | 0.005787037 | r20185 | Y | true | C | true | true | reactivePower | -375000 | -37500 | 102 | ACLineSegment | cap_1c | l2823592 | C | _BC0109A6-D424-E27B-371F-D593FEFD5BE5 | _4F76A5F9-271D-9EB8-5E31-AA362D86F2C3
capbank3 | 12470 | 12471.12 | 0.0057867083 | r18242 | Y | true |   |   |   |   |   |   |   |   |   |   |   | _CDEE25DE-0FF0-3AE7-5238-24DE1EAA2D64 | _4F76A5F9-271D-9EB8-5E31-AA362D86F2C3
```

Now, change the query by replacing `"_4F76A5F9-271D-9EB8-5E31-AA362D86F2C3"` with `"_AAE94E4A-2465-6F5E-37B1-3E72183A4E44"` to get results for the 9500 node model. Results are below:
```
name | basev | nomu | bsection | bus | conn | grnd | phs | ctrlenabled | discrete | mode | deadband | setpoint | delay | monclass | moneq | monbus | monphs | id | fdrid
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
capbank0a | 12470 | 7200 | 0.0077160494 | r42246 | Y | true | A |   |   |   |   |   |   |   |   |   |   | _9C742CAC-56EB-F4AA-2A40-B68F41ACDA54 | _AAE94E4A-2465-6F5E-37B1-3E72183A4E44
capbank0b | 12470 | 7200 | 0.0077160494 | r42246 | Y | true | B |   |   |   |   |   |   |   |   |   |   | _40B85852-05FE-70D8-81CA-769066AE3126 | _AAE94E4A-2465-6F5E-37B1-3E72183A4E44
capbank0c | 12470 | 7200 | 0.0077160494 | r42246 | Y | true | C |   |   |   |   |   |   |   |   |   |   | _192A1490-EB2F-557C-D4C2-2E63DCAFB5B1 | _AAE94E4A-2465-6F5E-37B1-3E72183A4E44
capbank1a | 12470 | 7200 | 0.005787037 | r42247 | Y | true | A |   |   |   |   |   |   |   |   |   |   | _A397C73B-35E4-C2ED-8B94-7BB70390FF24 | _AAE94E4A-2465-6F5E-37B1-3E72183A4E44
capbank1b | 12470 | 7200 | 0.005787037 | r42247 | Y | true | B |   |   |   |   |   |   |   |   |   |   | _9B7B350E-3E03-29A9-FDC4-0213E712B396 | _AAE94E4A-2465-6F5E-37B1-3E72183A4E44
capbank1c | 12470 | 7200 | 0.005787037 | r42247 | Y | true | C |   |   |   |   |   |   |   |   |   |   | _FA38140A-970E-BD61-84F3-6BB623A5DA91 | _AAE94E4A-2465-6F5E-37B1-3E72183A4E44
capbank2a | 12470 | 7200 | 0.005787037 | r20185 | Y | true | A |   |   |   |   |   |   |   |   |   |   | _03F18E11-D96A-5A05-3366-83579B60CA98 | _AAE94E4A-2465-6F5E-37B1-3E72183A4E44
capbank2b | 12470 | 7200 | 0.005787037 | r20185 | Y | true | B |   |   |   |   |   |   |   |   |   |   | _A6CF8323-7447-0397-C7D6-FC54F86CB4A4 | _AAE94E4A-2465-6F5E-37B1-3E72183A4E44
capbank2c | 12470 | 7200 | 0.005787037 | r20185 | Y | true | C |   |   |   |   |   |   |   |   |   |   | _99FE5F41-F952-2E90-55A4-4C551F598AC3 | _AAE94E4A-2465-6F5E-37B1-3E72183A4E44
capbank3 | 12470 | 12471.12 | 0.0057867083 | r18242 | Y | true |   |   |   |   |   |   |   |   |   |   |   | _FC0FC61F-4D23-0785-4E63-369B31ECFF38 | _AAE94E4A-2465-6F5E-37B1-3E72183A4E44
```
While the formatting isn't great, it's clear all the attributes which make the capacitors controllable are missing.

This can be confirmed by taking a look at the GridLAB-D models:
`8500`:
```
object capacitor {
  name "cap_capbank2a";
  parent "r20185";
  phases AN;
  phases_connected AN;
  cap_nominal_voltage 7200.00;
  capacitor_A 300000.00;
  switchA CLOSED;
  control MANUAL; // VAR;
  VAr_set_low -225000.00;
  VAr_set_high 150000.00;
	remote_sense "line_cap_1a";
  pt_phase A;
	control_level BANK;
  dwell_time 100.00;
}
```

`9500`:
```
object capacitor {
  name "cap_capbank2a";
  parent "r20185";
  phases AN;
  phases_connected AN;
  cap_nominal_voltage 7200.00;
  capacitor_A 300000.00;
  switchA CLOSED;
}
```

poorva1209, 2019-08-01:
This is provided by the model itself. Should be discussed with Alexander and Shiva to figure out if these capacitors should have local smart control. @temcdrm will discuss this on App standup call this Friday.

blthayer, 2019-08-01:
@poorva1209 - this had to do with the issue where Shiva removed the control stuff from the model. When it get put back in there was some sort of debacle where something broke in the GridLAB-D --> OpenDSS --> CIM --> GridLAB-D conversion pipeline.

I think Craig has mentioned that he has a working version of the model in `develop`, but this issue is for `v2019.07.0`. Will there be a `v2019.07.1` release soon?

temcdrm, 2019-08-02:
Please check that the July 18 version of the 9500-node XML file has been loaded into the platform. Outside of the platform, the query does produce expected results on that file, i.e., nine individual phase capacitors are controllable, and a three-phase "capbank3" is not.  See attached.

The capcontrol objects in the source OpenDSS files had been re-enabled on July 1. There were unrelated updates from IncSys on July 18, with patches on those by PNNL. This query on controllable capacitors should have been working on any platform build after July 1.

[image]
[image]

blthayer, 2019-08-03:
@poorva1209 @craig8 @temcdrm 
I confirmed that the queries above in platform version `v2019.07.0` yield the same results as when I originally filed the issue for `v2019.06.beta`. Will the fixed model be in the `v2019.08.0` release? When will that be available? I'm always hesitant to use `develop` - I kept having the rug shifting underneath me when I was running `develop` several months ago, so have opted to stick with the tagged versions.

blthayer, 2019-08-06:
I can confirm this is now fixed when running the platform with the `develop` tag. I'll close this once the next release is out.

closed 2019-08-19 by poorva1209