GRIDAPPSD/gridappsd-forum #16: Document sensor service

blthayer, 2019-07-30:
### Description of Issue
There isn't currently any documentation for the [sensor service](https://github.com/GRIDAPPSD/gridappsd-sensor-simulator). Doing a search for "sensor" in Read the Docs brings up only two page hits, and neither are related to the sensor service. Documentation should describe:
1. Where the repo is for reference
2. How the sensor service works and what it does
3. How to configure the platform to use the sensor service
4. How to configure the sensor service itself.

blthayer, 2019-07-31:
Adding the app-critical label since it's non-trivial to figure out how to use the sensor service...

poorva1209, 2019-07-31:
@craig8 We already have an issue on board to document sensor service. Can you prioritize this and do this today ?

poorva1209, 2019-07-31:
@blthayer sensor service is included in gridappsd container so it run it 
1. Go into sample_app container by
  docker exec -it gridappsddocker_sample_app_1 bash
2. Edit sample_app.config to include gridappsd-sensor-simulator under pre-reqs key. 
3. Exit sample app container
4. Go into gridappsd container and start platform. 
   You should see last lines on platform terminal as sample_app registraton message with sensor 
   service in pre-reqs. 

There is no other way of configuring sensor-service right now other than chaning the source code in gridappsd container under services folder.

blthayer, 2019-07-31:
@poorva1209 - thank you!

poorva1209, 2019-09-23:
@craig8 Update sensor service query data request and response here: https://gridappsd.readthedocs.io/en/latest/using_gridappsd/index.html#query-sensor-service-data

closed 2019-10-07 by poorva1209