GRIDAPPSD/gridappsd-forum #21: Sensor service bugs

blthayer, 2019-08-01:
__Overview__:
I got the sensor service running, and was able to query the time series database to get output. Upon inspecting the data returned, something seemed wrong.

__Details__:
I ran the sensor simulator with a 30 second `interval` (the default in the source code). However, the data I got back from querying the time series database was on a 3 second interval (just like the simulator).

__Possible Issue/Solution__:
Take a look at the `on_simulation_message` function in `sensor_simulator.py` in the `develop` branch of the `gridappsd-sensor-simulator` repository. For your convenience, I've reproduced it here:

```python
    def on_simulation_message(self, headers, message):
        """ Listens for simulation messages off the gridappsd message bus
        If the message is a measurements message then the function will inspect each
        non string value and generate a new value for the passed data.
        :param headers:
        :param message:
        :return:
        """
        mobj = message.copy()
        mout = mobj.copy()
        mout['measurements'] = []

        t = mobj['message']['timestamp'] 
        _log.debug(f"Processing Timestamp: {t}")
        for measurement in mobj['message']['measurements']:
            remove = False
            mcopy = measurement.copy()
            for prop, value in mcopy.items():
                # TODO: this only processes 'magnitude' and 'value'
                #       it needs to also process 'angle' but with different noise
                if prop in ('measurement_mrid', 'angle'):
                    continue

                new_value = self.get_new_value(t, value)

                if not new_value:
                    remove = True
                else:
                    mcopy[prop] = new_value

            if not remove:
                mout['measurements'].append(mcopy)

        # Publish new data back out to the message bus
        self._gapps.send(self.output_topic, json.dumps(mout))
```

Notice in the begging of the function, a copy of `message` is taken. First off, it would be wise to make this a `deepcopy`, as the `message` object isn't flat, and has mutable types underneath the top level.

Second, notice that `mout` has the field `measurements` which is initialized to an empty list. *HOWEVER*, the `for` loop is looping over `mobj['message']['measurements']`, *not* `mobj['message']`. I suspect this is resulting in the sensor service simply regurgitating the input message.

Third, there's still an outstanding `TODO` in which angles are not handled at all. This is not good, as it could result in reporting back magnitude averages but angle instantaneous values. 

Finally, the sensor service does not report what type of aggregation it reports. From reading the code, it would seem that *only* the averages are put back on the message bus. Is this desirable? Or should it be putting out min, max, *and* average. At any rate, there should be some sort of field like `aggregation_type` that specifies if the sensor service output is min, max, average, or something else (standard deviation would be useful in the future).

__Summary__:
1. Take deepcopies rather than copies.
2. Fix inconsistency in looping (`mout['measurements']` vs. `mobj['message']['measurements']`).
3. Handle angles in addition to magnitudes.
4. Report what type of aggregation comes out of the sensor service.

blthayer, 2019-08-01:
@craig8 here's a little demo of tracking magnitude and phase separately vs. tracking a full blown complex number:

```python
import cmath
>>> a = 5 + 1j*1
>>> b = 5 + 1j*3
>>> (cmath.phase(a) + cmath.phase(b)) / 2
0.36890753006023247
>>> cmath.phase((a + b)/2)
0.3805063771123649
>>> (abs(a) + abs(b)) / 2
5.464985704219043
>>> abs(a + b) / 2
5.385164807134504
```
You can see the results here are not equal.

poorva1209, 2019-08-20:
@Craig I think we need to pick this issue in current release as this is app-critical. Let's talk when you are back to plan again.

craig, 2019-08-21:
NO!

blthayer, 2019-10-04:
Dropping in to note that the current version is still only tracking voltage magnitudes, and not doing any processing with angles. This is okay for my use case, but something we'll need to fix long term.

closed 2019-10-22 by poorva1209