GRIDAPPSD/gridappsd-forum #22: Weather data in the influxdb database is mountain time.

Jeffrey-Simpson, 2019-08-08:
The time should be move 6 hours forward during Daylight Saving Time so that weather data lines up with the time of day that the simulation environment is running in.  It should be shifted 7 forward for Standard Time.

For example a  query at 1374489000 which is GMT Monday, July 22, 2013 10:30:00 AM gives these values:
```
AvgWindDirection                                  281.7
AvgWindSpeed                                    12.1824
DATE                                          7/22/2013
Diffuse                                   -0.0723454105
DirectCH1                                  0.0836627672
GlobalCM22                                 -0.147733296
MST                                               04:30
TowerDryBulbTemp                                 73.148
TowerRH                                           21.77
lat                                             39.74 N
long                                           105.18 W
place               Solar Radiation Research Laboratory
time                                         1374489000
```
When really we want the values 6 hours into the future  at 1374514200 to returned

```
AvgWindDirection                                   88.5
AvgWindSpeed                                      5.089
DATE                                          7/22/2013
Diffuse                                      8.31660309
DirectCH1                                    84.1498419
GlobalCM22                                   86.1813791
MST                                               10:30
TowerDryBulbTemp                                 83.588
TowerRH                                           39.62
lat                                             39.74 N
long                                           105.18 W
place               Solar Radiation Research Laboratory
time                                         1374510600
```

I can change the date in the Solar Forecasting repo or it can be change when it is put into InfluxDB

poorva1209, 2019-08-20:
Please test this with latest release 2019.08.1 and close.

Jeffrey-Simpson, 2019-08-20:
Hello,
I am testing  with ./run.sh -t v2019.08.1 and but I am still getting results in MST.
Also I getting only ten values back instead of a whole for this script.
Thanks,
-Jeff

poorva1209, 2019-08-20:
Did you run ./stop.sh -c before ./run,sh ?

Jeffrey-Simpson, 2019-08-21:
Yes.
I will try again this morning.

poorva1209, 2019-09-30:
@Jeffrey-Simpson Please test this with the latest release.

Jeffrey-Simpson, 2019-09-30:
Hello,
It looks good to me!
Thanks,
-Jeff

poorva1209, 2019-10-01:
Thanks @Jeffrey-Simpson . I am closing this issue.

closed 2019-10-01 by poorva1209