GRIDAPPSD/gridappsd-forum #23: Missing EnergyConsumer phases/measurements, 9500 node model, v2019.08.1

blthayer, 2019-09-01:
Platform version: `v2019.08.1`
Model: `9500 node`

In the 9500 node model, all energy consumers are split phase 120V/240V, and the individual phases (`s1`, `s2`) each have two `Analog` measurements: one for `PNV` and one for `VA`. 

It follows that each `EnergyConsumer` should have exactly 4 measurements associated with it. This was true prior to the `v2019.08.1` release (at least in the July release which I was previously using. I never used `2018.08.0` since it had major bugs in it).

The following EnergyConsumers only have `2` measurements instead of `4`:
```
2221108403a0
2221108403b0
2221108403c0
2221108404a0
2221108404b0
2221108404c0
2221108405a0
2221108405b0
2221108405c0
2221108406a0
2221108406b0
2221108407b0
2226061820b0
```
Note these names are off by a letter at the end (`a` or `b`), but if you do a search in the .glm of the model, they'll come right up.

Execute the following blazegraph query (sorry for the lack of human readability, this came out of my code):
```sparql
# Get EnergyConsumers and their nominal voltage
PREFIX r: <http://www.w3.org/1999/02/22-rdf-syntax-ns#>
PREFIX c: <http://iec.ch/TC57/CIM100#>
PREFIX xsd: <http://www.w3.org/2001/XMLSchema#>
SELECT ?name ?bus ?basev ?conn (group_concat(distinct ?phs;separator=",") as ?phases) WHERE { ?s r:type c:EnergyConsumer. VALUES ?feeder_mrid {"_AAE94E4A-2465-6F5E-37B1-3E72183A4E44"} ?s c:Equipment.EquipmentContainer ?fdr. ?fdr c:IdentifiedObject.mRID ?feeder_mrid. ?s c:IdentifiedObject.name ?name. ?s c:ConductingEquipment.BaseVoltage ?bv. ?bv c:BaseVoltage.nominalVoltage ?basev. ?s c:EnergyConsumer.phaseConnection ?connraw. bind(strafter(str(?connraw),"PhaseShuntConnectionKind.") as ?conn) OPTIONAL { ?ecp c:EnergyConsumerPhase.EnergyConsumer ?s. ?ecp c:EnergyConsumerPhase.phase ?phsraw. bind(strafter(str(?phsraw),"SinglePhaseKind.") as ?phs) } ?t c:Terminal.ConductingEquipment ?s. ?t c:Terminal.ConnectivityNode ?cn. ?cn c:IdentifiedObject.name ?bus } GROUP BY ?name ?bus ?basev ?p ?q ?conn ORDER by ?name 
```
The above query returns `2537` results, one result for each phase of each energy consumer (`s1` or `s2`). **This appears to be the root of the problem:** Notice `2537` is not divisible by `2`, so not all EnergyConsumer's have two phases in blazegraph. If we use a data analysis tool (`pandas` in Python in my case) to group the results of the above query by `bus`, we find that there are `13` buses which only have a single energy consumer phase associated with them. These buses are printed below:

```
sx1108403a
sx1108403b
sx1108403c
sx1108404a
sx1108404b
sx1108404c
sx1108405a
sx1108405b
sx1108405c
sx1108406a
sx1108406b
sx1108407b
sx1108410b
``` 
Furthermore, these buses match with the EnergyConsumers which only have two measurements associated with them (I only checked the first one, so I'm making an assumption about the rest).

If we investigate the GridLAB-D model output from the platform (via the [Configuration File API](https://gridappsd.readthedocs.io/en/latest/using_gridappsd/index.html#request-gridlab-d-base-file)), the associated `triplex_load` objects have both of their phases in tact. Here's an example:
```
object triplex_load {
  name "ld_2221108403a0a";
  parent "sx1108403a";
  phases AS;
  nominal_voltage 120.09;
  base_power_1 14940.13;
  base_power_2 14940.13;
  current_pf_1 0.97;
  current_fraction_1 1.00;
  current_pf_2 0.97;
  current_fraction_2 1.00;
}
```
**Bottom line**: There are EnergyConsumers with missing phase information in blazegraph, but both phases are present in the corresponding GridLAB-D model.

Assigning to @poorva1209 to reassign to the appropriate person. Tagging @afisher1 and @temcdrm as they're involved in the model conversion pipeline. Please tag anyone else as appropriate.

temcdrm, 2019-09-05:
The 9500-node loads are actually not all split-phase. In the repository UnbalancedLoads.dss file from July 18, sx1108403c is part of a three-phase unbalanced load at 120 Volts.  So, the described CIM and GLM outputs are probably as expected for those three-phase unbalanced loads. The first 9 lines are part of 3 unbalanced secondary loads. 

The last 4 lines are specified in UnbalancedLoads.dss as being on phase .1 (i.e., s1) only.  There is no matching unbalanced load on phase .2 (i.e., s2). Again, the CIM and GLM outputs are probably as expected for this situation.

In the 8500-node model, we were using balanced loads on the secondary, and that could have looked different to app developers.  The original EPRI UnbalancedLoads.dss file doesn't have these 13 problematic loads. Therefore, IncSys and/or WSU may be able to "fix" this by editing the UnbalancedLoads.dss file.

AAndersn, 2019-09-05:
Tom, you are correct: these are loads that we added in increase the load of the feeder.

Please confirm that the solution is adding a set of .2 loads to the DSS file.

closed 2019-10-03 by poorva1209