GRIDAPPSD/gridappsd-forum #29: Duplicate redis service

AAndersn, 2019-09-10:
When starting Gridapps-D using `./run.sh -t v2019.09.0`, docker returns an error:

```
Starting the docker containers
Creating gridappsddocker_redis_1      ... error
........
ERROR: for redis  Cannot start service redis: driver failed programming external connectivity on endpoint gridappsddocker_redis_1: Error starting userland proxy: listen tcp 0.0.0.0:6379: bind: address already in use
ERROR: Encountered errors while bringing up the project.
 
Error starting containers
Exiting
```

After running `sudo service redis stop`, I am then able to run `./run.sh -t v2019.09.0`, and gridapps-d and all docker containers start.

poorva1209, 2019-09-10:
@tonya1 Please look at this issue and see if we check for duplicate redis service and ask user to close it instead of showing this error.

craig8, 2019-09-19:
@AAndersn I believe this means that you have a service outside of the container that is redis.  You have two options.  You can remove the external redis service (or stop it like you did) or you can change the redis port and the connection string that redis uses to that port.  This however isn't a bug, but a configuration issue with your specific environment.

closed 2019-09-19 by craig8