GRIDAPPSD/gridappsd-forum #31: 9500 node model: possibly bad initial topology

blthayer, 2019-09-29:
As has been mentioned in the past, I use GridLAB-D "off to the side" as a tool for solving the powerflow and getting data out.

I query the platform to get the "base" model, and then add the necessary components (e.g. clock, powerflow module, etc.) in order to get the model in shape to run. This has been working until I tried platform version `v2019.09.1` (note I never used `v2019.09.0`, so this issue may have been present there).

Now, I get a cryptic error from GridLAB-D when trying to run the model:
```
"ERROR    [2013-04-01 12:00:00 UTC] : sync_substation(obj=8737;sourcebus): NR: A line's phase was flagged as not full three-phase, but wasn't: (line_ln6352701-1) 7 7 0 3"
...
```

Note I'm running the latest GridLAB-D build from develop.

By doing a quick Google search, it seems Frank helped someone back in May on a related issue:
https://sourceforge.net/p/gridlab-d/discussion/842562/thread/c926ca2ec6/

Unfortunately, ZenHub is lacking in some critical features - it won't let me attach a .glm file. So, I tried to hack it by changing the file extension to .txt. No dice - it then complains the file is too large. So, I'll email the file out to everyone I tag (thanks ZenHub!). 

I've modified the model slightly by commenting out everything MySQL related so that whoever takes this on doesn't have to have GridLAB-D built to interface with MySQL. Please let me know if you run into any issues.

Since the model runs in the platform, I'm presumably missing something, but it's incredibly unclear to me what that may be.

@afisher1 @poorva1209 @shpoudel @temcdrm

afisher1, 2019-09-30:
That error is because the GridLAB-D version you are using is out of date. Are you using the version of gld that is on the platform? If not you need to update to the latest version of the develop branch.

blthayer, 2019-09-30:
@afisher1 - I just finished triple-checking (rebuilding container, deleting and recreating PyCharm interpreter, etc.) and I'm still getting this error.

My container contains the latest from GridLAB-D's `develop` branch. As evidence, here's the output from running `gridlabd --version` inside my container:

```
GridLAB-D 4.1.0-18165 (Lugo [76964ea7:develop:Mod]) 64-bit LINUX RELEASE
```

I was able to get gridlab-d running inside the platform container (`v2019.09.1`) (had to export a valid `GLPATH`), and here's the version information I get:
```
GridLAB-D 4.1.0-18165 (Lugo [76964ea7:develop]) 64-bit LINUX RELEASE
```

I then copied the model into the container and tried to run it. I get the following messages (same as in my container):
```
gridappsd@0aba39b17a77:/tmp$ gridlabd test_9500.glm 
WARNING  [INIT] : object name '221-242079' does not follow strict naming rules and may not link correctly during load time
WARNING  [INIT] : last warning message was repeated 39 times
WARNING  [INIT] : Node:cap_capbank2a does not have the same nominal voltage as its parent - copying voltage from parent.
WARNING  [INIT] : last warning message was repeated 9 times
WARNING  [INIT] : Using a 2x2 z-matrix, instead of geometric values, is an under-determined system. Ground and/or neutral currents will be incorrect.
WARNING  [INIT] : last warning message was repeated 1274 times
WARNING  [INIT] : solarpanel: no climate data found, using static data
ERROR    [2012-01-01 00:00:00 UTC] : sync_substation(obj=8737;sourcebus): NR: A line's phase was flagged as not full three-phase, but wasn't: (line_ln6352701-1) 7 7 0 3
ERROR    [2012-01-01 00:00:00 UTC] : 2012-01-01 00:00:00 UTC: object sourcebus stopped its clock (exec)!
ERROR    [2012-01-01 00:00:00 UTC] : exec halted: synchronization failed
FATAL    [2012-01-01 00:00:00 UTC] : shutdown after simulation stopped prematurely
FATAL    [2012-01-01 00:00:00 UTC] : environment startup failed: Success
WARNING  [2012-01-01 00:00:00 UTC] : last warning message was repeated 353 times
```

afisher1, 2019-10-01:
You are running the 9500 node. Due to the meshed topology of the system we have to add the following objects to the model. You can see this in the model_startup.glm file. In fact regardless of whether the model is meshed or not we add these objects as they don't break anything if the system is truly radial.

module reliability;
object fault_check {
    name fault_check_object;
    check_mode ONCHANGE;
    eventgen_object external_event_handler;
    strictly_radial FALSE;
    grid_association TRUE;
}
object eventgen {
    name external_event_handler;
    use_external_faults TRUE;
}

These objects would not be found in the CIM model as they are GridLAB-D specific and needed to solve meshed models and provide a way to inject externally sourced fault messages to the gridlab-d model. I guess I'm a little unawares on what gets sent to the application who requests the simulation model file for running off to the side. I was under the impression that basically everything found in the temp simulation id folder was sent to the application.

afisher1, 2019-10-01:
When I add those objects to the file you sent the simulation solves for me.

blthayer, 2019-10-01:
@afisher1 - Thank you! I'll give this a go tomorrow when I have time and close this issue.

blthayer, 2019-10-02:
@afisher1 - that fixed it, thank you. I've updated my code and tests, and am comfortable closing this issue. Are there any gotchas I should be aware of? I'll effectively be always adding those objects to any model I want to run.

closed 2019-10-02 by blthayer