GRIDAPPSD/gridappsd-forum #32: Fault in 9500 node model

shpoudel, 2019-10-07:
@poorva1209 and @afisher1 , I worked with 9500 node model to test if faults are working. However, I have an error showing up in the console before I could verify it. I have attached the error as a picture here. Could you please let me know the reasons for such an error?

[image]

Note: I added a fault to happen 60 seconds after the simulation has started

poorva1209, 2019-10-07:
@shpoudel : Can you run the simulation multiple times and get the same error every time? This is something the FNCS do randomly so it shouldn't repeat with every simulation.

shpoudel, 2019-10-08:
@poorva1209 : I ran the simulation couple of times and it was not a success. However, again I encountered a new type of error in the console. Could you please look into it?

[image]

poorva1209, 2019-10-08:
@afisher1 Can you check the new error message.

poorva1209, 2020-03-09:
This is a random thrown by FNCS co-Simulator, not by GridAPPS-D platform. This would be irrelevant once we move to HELICS in March/April release. Closing this issue.

closed 2020-03-09 by poorva1209