GRIDAPPSD/gridappsd-forum #33: Ability to run time series simulations with load profile

ArunWSU, 2019-10-09:
Hi,
  I'm Arun, an app developer from WSU. To build and run my application, I need a time series with load power variations by assigning load shapes. Though I tried running simulation with houses, the voltage magnitude and load power measurements are constant as shown in the image. Kindly guide me on this, if the feature currently exists.

[image]

Regards,
Arun.

poorva1209, 2019-10-09:
@temcdrm Would this issue be resolved once we include internal house loads with scheduled load file? Or is this something different?

Jeffrey-Simpson, 2019-10-10:
You can build a 123 CIM model will pv and secondary loads here:
https://github.com/GRIDAPPSD/Powergrid-Models/tree/develop/blazegraph/test/dss/NREL

You might have to comment out the some of the loads in the IEEE123Master_fixedVR.dss so that there are no duplicate objects added.

Copy the following to opens_cmd_123pv_house_cim100.txt and run it with opendsscmd 
```
compile IEEE123Master_fixedVR.dss
solve
!redirect capacitor.dss
redirect IEEE123CommLoads.dss       // 3-phase and 1-phase delta loads, always use this
// use one but not both of the following
//redirect IEEE123PrimaryLoads.dss    // 1-phase wye loads on the primary
redirect IEEE123SecondaryLoads.dss  // 1-phase wye loads converted to secondary service

redirect pvsystems_high_pvs.dss

set maxiterations=20
solve

export summary ieee123_pv_reg_house_CIM100.csv

!export CIM100 cdpsmcombined file=ieee123_pv_reg_CIM100.xml sid=61683F36-350B-FAE0-9A22-752190519A0A fid=EBDB5A4A-543C-9025-243E-8CAD24307380 sgrid=3193578C-6CEF-F7EE-D78D-38D80D450636 geo=ieee rgnid=C9801073-17EA-61E4-F0F7-AC27F1499838
export CIM100 cdpsmcombined file=ieee123_pv_reg_house_CIM100.xml fid=E407CBB6-8C8D-9BC9-589C-AB83FBF0826D sub=ieee123_Substation sid=1BC1F4C0-4060-8913-B36B-A473AEE5D38B subgeo=medium sgrid=B6CEC5E5-D2FB-0AC0-803C-0A6FB3D0C864 geo=ieee rgnid=73C512BD-7249-4F50-50DA-D93849B89C43

```

ArunWSU, 2019-10-15:
Hi, 
Sorry for the late follow-up. I was havingh trouble installing openDSS in linux. I used "Procedure to compile OpenDSS in your local Linux distribution", v1.0, 2017 to build opendsscmd. I'm using version 8, which is working when I verified for IEEE 13 bus system. 
In step 1, I faced one error is that in your "master_VR" you have defined as "IEEELinecodes.DSS", which doesn't redirect unless it is .dss. Now, I'm getting the error CIM100 export error , when I do step 1 replacing that with the code that you have given as in attachment.  @Jeffrey-Simpson Can you help me on this?

[image]

Also, one general feedback from WSU-ADMS team is that documentation mentions the use of player file for time varying simulations. We understand that you are trying to fix the same for IEEE 9500 node model. For the November demo, do kindly let us know when the time varying simulation for IEEE 9500 feeder will work.

Jeffrey-Simpson, 2019-10-15:
Hello,
You should use version 1.2.
You can also just use the opendsscmd that is in the gridappsd docker container.

docker ps
docker cp <dir-to-copy> <containerid>:<dir-location>
docker exec -it <containerid> bash
opendsscmd <file>

[Image removed by sender. Export error]<https://user-images.githubusercontent.com/37948729/66862969-e2015c80-ef46-11e9-8ce9-4d92c5ed82a1.png>

ArunWSU, 2019-10-15:
Hi, 
I followed the steps that you had outlined. I had the containers running and used the container gridappsd: v2019.09.1 for running opendsscmd. Then, I created a directory in it and copied all the all the files into it using the docker cp command. I opened and checked the files after docker exec command. If I run the opendsscmd, it still shows the same error as the previous image with two additional lines.
FNCS Connected
...
Error: Unknown Export command: "cim100"
Last Error: Error: Unknown Export command: "cim100"
An unhandled exception occurred at $0000000000448F67:
EFCreateError: Unable to create file "opendsscmd.ini"
  $0000000000448F67
  $0000000000448DCC

Jeffrey-Simpson, 2019-10-15:
Hello,
I tried  and I get the same error in the docker container. Maybe it is not as up to date as I thought.

Can you download and try one of these versions?
https://sourceforge.net/projects/electricdss/files/OpenDSSCmd/

Thanks,
-Jeff

ArunWSU, 2019-10-15:
Hi,
I also saw that link previously when I googled opendsscmd. But, it is only an executable run file and is not linked to any application. Hence, it doesn't work. I will update you if I can find some way to link and update it to the previous installation.
Arun

Jeffrey-Simpson, 2019-10-15:
The installer run in the container and should run on your machine.
You will have to run as super user I think

docker cp ~/Downloads/opendsscmd-1.2.2-linux-x64-installer.run 0f2e229c94a2:/tmp/gridappsd_tmp/

chmod 777 opendsscmd-1.2.2-linux-x64-installer.run

 ./opendsscmd-1.2.2-linux-x64-installer.run

ArunWSU, 2019-10-17:
Hi,
I still didn't get an XML file as I get an Error 303: File not found after following the steps that you have outlined above. Just to make a clarification, will doing the steps only generate a secondary model in the platform or will it help in generating time series measurements? Kindly provide me a tested out working fix.
Thanks and regards,
Arun.

[image]

Jeffrey-Simpson, 2019-10-17:
It builds a model with secondary loads.
Since you copied the files into the docker container they will not have the correct file permissions.
Try logging into the container as root.
Docker exec -it -u root <ID>
And running it.
Also make sure you are not importing things twice like the pv_high file or you will get those duplicate errors.
Thanks,
-Jeff

[Image removed by sender. opendss_cmd_1 2 2_execution_error]<https://user-images.githubusercontent.com/37948729/66986127-183bfa80-f073-11e9-9043-3528f27819fe.png>

poorva1209, 2019-11-18:
@ArunWSU @Jeffrey-Simpson Is this issue still open?

ArunWSU, 2019-11-18:
Hi.
We were able to see the time series variations in the new release. So, I'm closing this issue.
Thanks and regards,
Arun.

closed 2019-11-18 by ArunWSU