GRIDAPPSD/gridappsd-forum #34: Simulation configuration does not handle ZIP fractions correctly

blthayer, 2019-10-09:
@temcdrm @afisher1 @poorva1209

I've confirmed this is an issue both through the GUI and when starting a simulation with my own code. The steps to reproduce are quite simple:

1. Start platform
2. In the GUI, select the 8500 node feeder
3. In the "simulation" tab, change `z_fraction` from `"0"` to `"1"` and change `i_fraction` from `"1"` to `"0"`.
4. Click "submit"
5. After the one line diagram loads, click the play button to start the simulation
6. Get the ID of the main platform Docker container (via `docker container ls`) then enter the main container via `docker exec -it <container ID> bash`
7. Navigate to `/tmp/gridappsd_tmp/<simulation ID>`
8. Once the simulation has started, you should be able to see a `model_base.glm` file in that directory.
9. Check the fraction values in the glm file like this `cat model_base.glm | grep impedance_fraction` and like this `cat model_base.glm | grep current_fraction`

You'll see output like:
```
...
  impedance_fraction_1 0.50;
  impedance_fraction_2 0.50;
  impedance_fraction_1 0.50;
  impedance_fraction_2 0.50;
  impedance_fraction_1 0.50;
  impedance_fraction_2 0.50;
  impedance_fraction_1 0.50;
...
```
and
```
...
  current_fraction_2 0.50;
  current_fraction_1 0.50;
  current_fraction_2 0.50;
  current_fraction_1 0.50;
  current_fraction_2 0.50;
  current_fraction_1 0.50;
  current_fraction_2 0.50;
...
``` 

Notice that doesn't match how we configured the simulation.

temcdrm, 2019-10-09:
Is there a log of what arguments get passed to the CIMImporter?  This output would be expected if both -z=1 and -i=1.  

The GUI shouldn't really call these fractions.  Whenever z, i, or p is specified, then the fractions are calculated as, for example, i/(z+i+p).  So you could input these as fractions and it would work if they all sum to 1.  However, you could also input percent or some other scheme.  Also, if you just say z=0.5 and leave i=0, p=0 (default), you will actually get an impedance_fraction of 1.

poorva1209, 2019-10-18:
@tdtalbot In GLDAllConfigurationHandler we are changing iFraction to 1 even if user 0 value in simulation request. Here is the code snippet (line 165)
```
double iFraction = GridAppsDConstants.getDoubleProperty(parameters, IFRACTION, 0);
		if(iFraction==0){
			iFraction = 1;
			bWantZip = true;
		}
```
This is causing the problem that Brandon mentioned in this issue. I just wanted to make sure that this is not done for some specific reason before removing 'iFraction=1' line.'

closed 2019-10-18 by poorva1209