GRIDAPPSD/gridappsd-forum #35: Errors when commanding equipment

blthayer, 2019-10-13:
@poorva1209 @craig8 
# Version information
Platform version: `v2019.09.1`
Model: 9500 (MRID: `'_AAE94E4A-2465-6F5E-37B1-3E72183A4E44'`)

# Error from platform
```
Thread-140 ERROR [gov.pnnl.proven.api.exchange.RestExchange] - 
Error while calling REST Service
Thread-140 ERROR [gov.pnnl.proven.api.exchange.RestExchange] - Exception: javax.ws.rs.InternalServerErrorException: HTTP 500 Internal Server Error
javax.ws.rs.InternalServerErrorException: HTTP 500 Internal Server Error
	at org.glassfish.jersey.client.JerseyInvocation.convertToException(JerseyInvocation.java:1098)
	at org.glassfish.jersey.client.JerseyInvocation.translate(JerseyInvocation.java:883)
	at org.glassfish.jersey.client.JerseyInvocation.lambda$invoke$1(JerseyInvocation.java:767)
	at org.glassfish.jersey.internal.Errors.process(Errors.java:316)
	at org.glassfish.jersey.internal.Errors.process(Errors.java:298)
	at org.glassfish.jersey.internal.Errors.process(Errors.java:229)
	at org.glassfish.jersey.process.internal.RequestScope.runInScope(RequestScope.java:414)
	at org.glassfish.jersey.client.JerseyInvocation.invoke(JerseyInvocation.java:765)
	at org.glassfish.jersey.client.JerseyInvocation$Builder.method(JerseyInvocation.java:456)
	at org.glassfish.jersey.client.JerseyInvocation$Builder.post(JerseyInvocation.java:357)
	at gov.pnnl.proven.api.exchange.RestExchange.addProvenData(RestExchange.java:293)
	at gov.pnnl.proven.api.producer.ProvenSession.sendMessage(ProvenSession.java:191)
	at gov.pnnl.proven.api.producer.Producer.sendMessage(Producer.java:379)
	at gov.pnnl.proven.api.producer.ProvenProducer.sendBulkMessage(ProvenProducer.java:133)
	at gov.pnnl.goss.gridappsd.data.ProvenTimeSeriesDataManagerImpl$1.onMessage(ProvenTimeSeriesDataManagerImpl.java:172)
	at pnnl.goss.core.client.GossClient$1.run(GossClient.java:321)
	at java.lang.Thread.run(Thread.java:748)

```

# To reproduce
1. Start simulation with 9500 node feeder.
2. Create a `GridAPPSD` object (Python), call it `gad` for the sake of these steps. 
3. Define the following json string as `msg`:
```json
{
	"simulation_id": "509727846",
	"message": {
		"timestamp": 1570997372,
		"difference_mrid": "19254d1e-e842-44cc-8d1e-85a45f933d09",
		"reverse_differences": [
			{
				"object": "_229AACF5-22E9-9E5E-3E40-3CB05ACFCE56",
				"attribute": "TapChanger.step",
				"value": -3
			},
			{
				"object": "_32751BF8-77D2-E518-3B47-78FDC02F603A",
				"attribute": "TapChanger.step",
				"value": -3
			},
			{
				"object": "_D24B8F9E-4D9D-9F24-1E6A-645DD3DFF075",
				"attribute": "TapChanger.step",
				"value": -3
			},
			{
				"object": "_931818A1-4485-C354-518E-3761DEC9ACB6",
				"attribute": "TapChanger.step",
				"value": -2
			},
			{
				"object": "_74A30019-0582-3894-96A7-84D42330C0AB",
				"attribute": "TapChanger.step",
				"value": -2
			},
			{
				"object": "_2CB8138B-BC29-9093-5F25-4DFA2BEC742C",
				"attribute": "TapChanger.step",
				"value": -3
			},
			{
				"object": "_A3C89D06-E86E-A137-6500-2A6164F637A0",
				"attribute": "TapChanger.step",
				"value": -1
			},
			{
				"object": "_B0CCD269-2ED2-1217-AC3A-634797F3851F",
				"attribute": "TapChanger.step",
				"value": -2
			},
			{
				"object": "_652E7D66-C447-C725-65BD-DC1FD25A0A51",
				"attribute": "TapChanger.step",
				"value": -1
			},
			{
				"object": "_1BBF7067-962E-E8E7-9C1F-FED0CFA1FCA4",
				"attribute": "TapChanger.step",
				"value": 6
			},
			{
				"object": "_9E9ADF4E-550C-1184-A14F-3F7878DAA0C3",
				"attribute": "TapChanger.step",
				"value": 5
			},
			{
				"object": "_FFBFE6B6-8965-F8A5-C4CF-0BD57A0F1168",
				"attribute": "TapChanger.step",
				"value": 1
			},
			{
				"object": "_E0EABCA2-20F9-2BCC-A51F-2BD70F25520D",
				"attribute": "TapChanger.step",
				"value": 9
			},
			{
				"object": "_DC247627-511C-5CF2-3676-F1FFA58EBD89",
				"attribute": "TapChanger.step",
				"value": 11
			},
			{
				"object": "_AC676D9E-8197-81C9-B9D9-09A004FC4CAF",
				"attribute": "TapChanger.step",
				"value": 3
			},
			{
				"object": "_971B59D8-E0B0-64AF-5DC1-1C4243A9E6B8",
				"attribute": "TapChanger.step",
				"value": 6
			},
			{
				"object": "_F87D3E02-55F2-74C9-F18E-7A047E549540",
				"attribute": "TapChanger.step",
				"value": 10
			},
			{
				"object": "_10063B37-BC31-031A-33A5-7BF5A74E628F",
				"attribute": "TapChanger.step",
				"value": 5
			}
		],
		"forward_differences": [
			{
				"object": "_229AACF5-22E9-9E5E-3E40-3CB05ACFCE56",
				"attribute": "TapChanger.step",
				"value": 16
			},
			{
				"object": "_32751BF8-77D2-E518-3B47-78FDC02F603A",
				"attribute": "TapChanger.step",
				"value": 16
			},
			{
				"object": "_D24B8F9E-4D9D-9F24-1E6A-645DD3DFF075",
				"attribute": "TapChanger.step",
				"value": 16
			},
			{
				"object": "_931818A1-4485-C354-518E-3761DEC9ACB6",
				"attribute": "TapChanger.step",
				"value": 16
			},
			{
				"object": "_74A30019-0582-3894-96A7-84D42330C0AB",
				"attribute": "TapChanger.step",
				"value": 16
			},
			{
				"object": "_2CB8138B-BC29-9093-5F25-4DFA2BEC742C",
				"attribute": "TapChanger.step",
				"value": 16
			},
			{
				"object": "_A3C89D06-E86E-A137-6500-2A6164F637A0",
				"attribute": "TapChanger.step",
				"value": 16
			},
			{
				"object": "_B0CCD269-2ED2-1217-AC3A-634797F3851F",
				"attribute": "TapChanger.step",
				"value": 16
			},
			{
				"object": "_652E7D66-C447-C725-65BD-DC1FD25A0A51",
				"attribute": "TapChanger.step",
				"value": 16
			},
			{
				"object": "_1BBF7067-962E-E8E7-9C1F-FED0CFA1FCA4",
				"attribute": "TapChanger.step",
				"value": 16
			},
			{
				"object": "_9E9ADF4E-550C-1184-A14F-3F7878DAA0C3",
				"attribute": "TapChanger.step",
				"value": 16
			},
			{
				"object": "_FFBFE6B6-8965-F8A5-C4CF-0BD57A0F1168",
				"attribute": "TapChanger.step",
				"value": 16
			},
			{
				"object": "_E0EABCA2-20F9-2BCC-A51F-2BD70F25520D",
				"attribute": "TapChanger.step",
				"value": 16
			},
			{
				"object": "_DC247627-511C-5CF2-3676-F1FFA58EBD89",
				"attribute": "TapChanger.step",
				"value": 16
			},
			{
				"object": "_AC676D9E-8197-81C9-B9D9-09A004FC4CAF",
				"attribute": "TapChanger.step",
				"value": 16
			},
			{
				"object": "_971B59D8-E0B0-64AF-5DC1-1C4243A9E6B8",
				"attribute": "TapChanger.step",
				"value": 16
			},
			{
				"object": "_F87D3E02-55F2-74C9-F18E-7A047E549540",
				"attribute": "TapChanger.step",
				"value": 16
			},
			{
				"object": "_10063B37-BC31-031A-33A5-7BF5A74E628F",
				"attribute": "TapChanger.step",
				"value": 16
			}
		]
	}
}
```

4. Assuming you've stored the simulation ID as `sim_id` and have imported `topics`, run `gad.send(topic=topics.simulation_input_topic(sim_id), message=msg)`

# Useful next steps
So that we can isolate the issue, it would be good if someone could send the command into a running simulation using something other than the `gridappsd` Python package.

# Other notes
I had the same issue when following a similar procedure but with the 8500 node system (different MRIDs and message, of course)

poorva1209, 2019-10-14:
@blthayer I think this is a documentation issue. I tried the following and it is working in v2019.09.1. I updated the documentation too here: https://gridappsd.readthedocs.io/en/latest/using_gridappsd/index.html#send-input-to-simulation

While debugging this we captured another issue with Proven that if all the values are of string type then it treats them as "tags" instead of field values. I am creating another issue for that but it shouldn't affect your input message here and shouldn't block you. @ericstephan FYI.

```
{
	"command": "update",
	"input": {
		"simulation_id": "509727846",
		"message": {
			"timestamp": 1570997372,
			"difference_mrid": "19254d1e-e842-44cc-8d1e-85a45f933d09",
			"reverse_differences": [{
				"object": "_229AACF5-22E9-9E5E-3E40-3CB05ACFCE56",
				"attribute": "TapChanger.step",
				"value": -3
			}, {
				"object": "_32751BF8-77D2-E518-3B47-78FDC02F603A",
				"attribute": "TapChanger.step",
				"value": -3
			}, {
				"object": "_D24B8F9E-4D9D-9F24-1E6A-645DD3DFF075",
				"attribute": "TapChanger.step",
				"value": -3
			}, {
				"object": "_931818A1-4485-C354-518E-3761DEC9ACB6",
				"attribute": "TapChanger.step",
				"value": -2
			}, {
				"object": "_74A30019-0582-3894-96A7-84D42330C0AB",
				"attribute": "TapChanger.step",
				"value": -2
			}, {
				"object": "_2CB8138B-BC29-9093-5F25-4DFA2BEC742C",
				"attribute": "TapChanger.step",
				"value": -3
			}, {
				"object": "_A3C89D06-E86E-A137-6500-2A6164F637A0",
				"attribute": "TapChanger.step",
				"value": -1
			}, {
				"object": "_B0CCD269-2ED2-1217-AC3A-634797F3851F",
				"attribute": "TapChanger.step",
				"value": -2
			}, {
				"object": "_652E7D66-C447-C725-65BD-DC1FD25A0A51",
				"attribute": "TapChanger.step",
				"value": -1
			}, {
				"object": "_1BBF7067-962E-E8E7-9C1F-FED0CFA1FCA4",
				"attribute": "TapChanger.step",
				"value": 6
			}, {
				"object": "_9E9ADF4E-550C-1184-A14F-3F7878DAA0C3",
				"attribute": "TapChanger.step",
				"value": 5
			}, {
				"object": "_FFBFE6B6-8965-F8A5-C4CF-0BD57A0F1168",
				"attribute": "TapChanger.step",
				"value": 1
			}, {
				"object": "_E0EABCA2-20F9-2BCC-A51F-2BD70F25520D",
				"attribute": "TapChanger.step",
				"value": 9
			}, {
				"object": "_DC247627-511C-5CF2-3676-F1FFA58EBD89",
				"attribute": "TapChanger.step",
				"value": 11
			}, {
				"object": "_AC676D9E-8197-81C9-B9D9-09A004FC4CAF",
				"attribute": "TapChanger.step",
				"value": 3
			}, {
				"object": "_971B59D8-E0B0-64AF-5DC1-1C4243A9E6B8",
				"attribute": "TapChanger.step",
				"value": 6
			}, {
				"object": "_F87D3E02-55F2-74C9-F18E-7A047E549540",
				"attribute": "TapChanger.step",
				"value": 10
			}, {
				"object": "_10063B37-BC31-031A-33A5-7BF5A74E628F",
				"attribute": "TapChanger.step",
				"value": 5
			}],
			"forward_differences": [{
				"object": "_229AACF5-22E9-9E5E-3E40-3CB05ACFCE56",
				"attribute": "TapChanger.step",
				"value": 16
			}, {
				"object": "_32751BF8-77D2-E518-3B47-78FDC02F603A",
				"attribute": "TapChanger.step",
				"value": 16
			}, {
				"object": "_D24B8F9E-4D9D-9F24-1E6A-645DD3DFF075",
				"attribute": "TapChanger.step",
				"value": 16
			}, {
				"object": "_931818A1-4485-C354-518E-3761DEC9ACB6",
				"attribute": "TapChanger.step",
				"value": 16
			}, {
				"object": "_74A30019-0582-3894-96A7-84D42330C0AB",
				"attribute": "TapChanger.step",
				"value": 16
			}, {
				"object": "_2CB8138B-BC29-9093-5F25-4DFA2BEC742C",
				"attribute": "TapChanger.step",
				"value": 16
			}, {
				"object": "_A3C89D06-E86E-A137-6500-2A6164F637A0",
				"attribute": "TapChanger.step",
				"value": 16
			}, {
				"object": "_B0CCD269-2ED2-1217-AC3A-634797F3851F",
				"attribute": "TapChanger.step",
				"value": 16
			}, {
				"object": "_652E7D66-C447-C725-65BD-DC1FD25A0A51",
				"attribute": "TapChanger.step",
				"value": 16
			}, {
				"object": "_1BBF7067-962E-E8E7-9C1F-FED0CFA1FCA4",
				"attribute": "TapChanger.step",
				"value": 16
			}, {
				"object": "_9E9ADF4E-550C-1184-A14F-3F7878DAA0C3",
				"attribute": "TapChanger.step",
				"value": 16
			}, {
				"object": "_FFBFE6B6-8965-F8A5-C4CF-0BD57A0F1168",
				"attribute": "TapChanger.step",
				"value": 16
			}, {
				"object": "_E0EABCA2-20F9-2BCC-A51F-2BD70F25520D",
				"attribute": "TapChanger.step",
				"value": 16
			}, {
				"object": "_DC247627-511C-5CF2-3676-F1FFA58EBD89",
				"attribute": "TapChanger.step",
				"value": 16
			}, {
				"object": "_AC676D9E-8197-81C9-B9D9-09A004FC4CAF",
				"attribute": "TapChanger.step",
				"value": 16
			}, {
				"object": "_971B59D8-E0B0-64AF-5DC1-1C4243A9E6B8",
				"attribute": "TapChanger.step",
				"value": 16
			}, {
				"object": "_F87D3E02-55F2-74C9-F18E-7A047E549540",
				"attribute": "TapChanger.step",
				"value": 16
			}, {
				"object": "_10063B37-BC31-031A-33A5-7BF5A74E628F",
				"attribute": "TapChanger.step",
				"value": 16
			}]
		}
	}
}
```

blthayer, 2019-10-15:
@poorva1209 @craig8 
Thanks for updating the documentation. I did initially have the messages sending in the format Poorva suggested, but changed it when I saw it was different from the documentation.

I think my primary issue is this "internal server error" causes the `gridappsd-python` code to come crashing down. Here's a stack trace:

```
Exception in thread StompReceiverThread-1:
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.7/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.7/site-packages/stomp/transport.py", line 347, in __receiver_loop
    self.process_frame(f, frame)
  File "/usr/local/lib/python3.7/site-packages/stomp/transport.py", line 193, in process_frame
    self.notify(frame_type, f.headers, f.body)
  File "/usr/local/lib/python3.7/site-packages/stomp/transport.py", line 249, in notify
    rtn = notify_func(headers, body)
  File "/usr/local/lib/python3.7/site-packages/gridappsd/goss.py", line 244, in on_message
    self._callback(header, msg)
  File "/usr/local/lib/python3.7/site-packages/gridappsd/simulation.py", line 154, in __on_platformlog
    if self.simulation_id == message['processId']:
KeyError: 'processId'
``` 

I went ahead and patched this in `gridappsd-python`, you can find the commit [here](https://github.com/GRIDAPPSD/gridappsd-python/commit/7bcb44feec01dc4d6f3f78ebad2b57ebf896f059). 

By making that patch, I'm able to see that the simulation does indeed continue running (I couldn't tell before since everything just died due to the error in `gridappsd-python`). 

We should probably beef up the error handling throughout that repository.

I'm going to close this issue, but I suggest @poorva1209 and @craig8 (et al) open a new issue for error handling.

closed 2019-10-15 by blthayer