GRIDAPPSD/gridappsd-forum #36: PV inverter setpoints are being reset by the weather player file

Jeffrey-Simpson, 2019-10-14:
I believe that PV inverter setpoints are being reset by the weather player file. I noticed that after our algorithm changes the p and q set point that the set points get changed back after a minute which lines up with the frequency of the data in the player file.

ftuffner, 2019-10-23:
@Jeffrey-Simpson I'm having some issues duplicating this issue, but I'm not sure I am looking at it right.  When you mention the "weather player file", is that playing directly into the weather object, or are you using the "PLAYERVALUE" mode of the solar object?

Jeffrey-Simpson, 2019-10-23:
Hello,
It should be a  weather csv playing into a climate object.

This is the start file:

clock {
     timezone "UTC0";
     starttime '2013-07-22 16:30:00';
     stoptime '2013-07-22 16:35:00';
}
#set maximum_synctime=3600
#set suppress_repeat_messages=1
#set relax_naming_rules=1
#set profiler=1
#set minimum_timestep=0.1
module connection;
module generators;
module tape;
module powerflow {
     line_capacitance TRUE;
     solver_method NR;
}
module climate;
module reliability;
object fncs_msg {
     name 173895120;
     message_type JSON;
     configure model_outputs.json;
     option "transport:hostname 127.0.0.1, port 57816";
}
object recorder {
     parent 173895120;
     property message_type;
     file 173895120.csv;
     interval 1;
}
object fault_check {
     name fault_check_object;
     check_mode ONCHANGE;
     eventgen_object external_event_handler;
     output_filename fault_check_output.txt;
}
object eventgen {
     name external_event_handler;
     use_external_faults TRUE;
}
object csv_reader {
 name CSVREADER;
 filename "model_weather.csv";
}
object climate {
 name "Weather Data";
 tmyfile "model_weather.csv";
 reader CSVREADER;
}; 
class player {
   double value;
}
object player {
   name ieeezipload;
   file ieeezipload.player;
   loop 0;
}
#define VSOURCE=66395.28095680696
#include "/tmp/gridappsd_tmp/173895120/model_base.glm"

poorva1209, 2020-03-09:
@afisher1 @Jeffrey-Simpson is this still relevant?

closed 2020-12-07 by Jeffrey-Simpson