GRIDAPPSD/gridappsd-forum #37: Simulation output not updated

shpoudel, 2019-10-21:
@poorva1209 : I am subscribing to simulation output to check the switch status every time I get a message from the platform. Even if I changed the status of some switch manually, the output remains the same. It was working fine when I gave the demo a few weeks back on the IEEE 123-node system. Is there something I should be aware of? 

Note: I am checking the status of LoadBreakSwitch.

poorva1209, 2019-10-21:
What release version are you using ?

shpoudel, 2019-10-21:
I am using 2019.09.1

poorva1209, 2019-10-21:
Can you also send met the message you are sending to change the switch status,.

shpoudel, 2019-10-21:
I am changing it from visualization. However, I also tried it by sending the following code. It is just a sample code for opening Switch5 in IEEE 123 node. I get a message in console that the message is passed but I don't see the corresponding change in simulation output back. 

 op = ['_F2599E92-5611-4112-B522-3D3044A8055F']
        if self.flag_sw == 0:
            for sw_mrid in op:
                self._open_diff.add_difference(sw_mrid, "Switch.open", 1, 0)
                msg = self._open_diff.get_message()
                print (msg)
                self._gapps.send(self._publish_to_topic, json.dumps(msg))  
                self.flag_sw = 1

shpoudel, 2019-10-21:
Following are the snapshots:
[image]
[image]

poorva1209, 2019-10-31:
v2019.10.0 has been updated to resolve this issue. Please test it again by stopping and pulling the containers again.

closed 2019-11-01 by poorva1209