GRIDAPPSD/gridappsd-forum #39: Implement simulation clock topic/service

blthayer, 2019-10-28:
@poorva1209 @craig8 
Not a high priority issue, just something I'm thinking about since I'm finding myself needing to implement my own version of this.

It would be quite helpful if there was a topic one could subscribe to that simply emitted the current simulation time. At present, one has to listen to the simulation log and look for messages which say "incrementing to <time step>." Then, one has to take this time step and add it to the known simulation starting timestamp (which you have to listen for) to get the present simulation time. It seems like a simple and useful service would emit both the time step and time for each simulation time step.

blthayer, 2019-10-28:
Update: You can find my implementation [here](https://github.com/GRIDAPPSD/pyvvo/blob/develop/pyvvo/gridappsd_platform.py) (search for "class SimulationClock") and corresponding tests [here](https://github.com/GRIDAPPSD/pyvvo/blob/develop/tests/test_gridappsd_platform.py) (search for "SimulationClockTestCase").

poorva1209, 2019-10-28:
Log message is not a good of checking for simulation clock time. 
There is a queue where fncs bridge publishes simulation time that is subscribed by Test Manager only. We can convert that into topic so applications/services can also subscribe to it. I will add the details here soon.

AAndersn, 2022-05-12:
@garydblack Let's add this old ask into the set of App Toolbox services for discussion this month. Somebody brought it up again recently and but it was unclear if this was going to go into the platform or toolbox pipeline