GRIDAPPSD/gridappsd-forum #40: TimeoutError in version 2019.10.0

shpoudel, 2019-10-28:
@poorva1209 @craig8 
I am getting the time out error while using the get_response() function from goss.py to query for the object measurement MRIDs.  It was working fine for v2019.09.1

blthayer, 2019-10-30:
@shpoudel 

This is a simple fix - up the `timeout` parameter. The signature of the `get_response` function is as follows ([source](https://github.com/GRIDAPPSD/gridappsd-python/blob/develop/gridappsd/goss.py)):

```python
get_response(self, topic, message, timeout=5)
```

You can see the default timeout is 5 seconds, which is relatively low. There are times when the platform is not very responsive (e.g. if it's in the middle of generating the 9500 node model in order to start a simulation).

I had to up my timeouts starting in either `v2019.09.0` or `v2019.09.1`, I don't remember which.

@craig8 @poorva1209 I propose we close this ticket if @shpoudel is okay with the fix.

shpoudel, 2019-10-30:
@blthayer Thanks for the reply. However, I am not using the default timeout. I have given timeout = 90 seconds in my code. Do you remember what value you used there? I remember someone in last week's meeting mentioned 90 seconds should be more than enough.

poorva1209, 2019-10-30:
@shpoudel What model and object type are you using for measurement queries. 90 seconds should be more than enough for the response. Can you check if your influxdb and proven containers are up and running?

shpoudel, 2019-10-30:
@poorva1209 I am using a 9500 node model and object type for me are LoadBreakSwitch and EnergyConsumer. Yes, the containers you mentioned are up. I changed it into 180 seconds and it worked a few times and again the same error came up. It seems so random to me.

poorva1209, 2019-11-14:
@shpoudel Is it working now for you ?

shpoudel, 2019-11-14:
This is working as of now for v2019.10.0. I was not able to make this work with develop tag.

poorva1209, 2019-11-14:
So you are getting timeout error even when starting the application once the simulation has started ?

shpoudel, 2019-11-14:
Yes, I was getting that even when starting the application once the simulation has started. So, I moved to v2019.10.0 from develop and it solves as desired now.

poorva1209, 2019-11-14:
ok, once you are done making changes and testing. we can setup a call and test together on develop.

shpoudel, 2019-11-14:
Sure. Sounds good!

poorva1209, 2019-11-14:
I am closing this issue. Once you do the test with develop after updating the app. And if it is still not working then let's make another issue with develop tag.

closed 2019-11-14 by poorva1209