GRIDAPPSD/gridappsd-forum #42: Memory leak in gridappsd-python/simulation

blthayer, 2019-10-30:
@craig8 

The `Simulation` class in gridappsd-python/simulation uses regular sets to track callbacks, which can/does lead to memory leaks. A good explanation can be found [here](https://stackoverflow.com/a/14922132/11052174).

Fortunately, the fix is simple: replace `set()` with `weakref.WeakSet()`.

I'll fix this and push to develop.

craig8, 2019-10-30:
formality of a pull request eh :)

craig8, 2019-10-30:
```
Traceback (most recent call last):
  File "/home/osboxes/repos/gridappsd-python-log/run_simulation.py", line 128, in <module>
    sim.add_onmesurement_callback(onmeasurment)
  File "/home/osboxes/repos/gridappsd-python-log/gridappsd/simulation.py", line 111, in add_onmesurement_callback
    self.__filterable_measurement_callback_set.add((callback, device_filter))
  File "/usr/lib/python3.6/_weakrefset.py", line 84, in add
    self.data.add(ref(item, self._remove))
TypeError: cannot create weak reference to 'tuple' object
```

with a tuple we get this error...so going to have to do something different.

blthayer, 2019-11-08:
I've reverted the changes, and am just going to close this issue for now. If we want to use some of my workarounds later, we can.

closed 2019-11-08 by blthayer