GRIDAPPSD/gridappsd-forum #44: No power consumption by load(s) in 9500 node model

blthayer, 2019-10-31:
@poorva1209 @craig8 
@Jeffrey-Simpson has also observed this behavior.

## The Problem and How to Reproduce
Title sums it up: I'm looking at power output (VA measurements) for a given load and getting 0 values, but the voltage (PNV measurements) are not zero, and close to nominal. Since the voltage is not 0, the load is indeed connected to the system.

Here's my run configuration:
```
{
  "power_system_config": {
    "GeographicalRegion_name": "_73C512BD-7249-4F50-50DA-D93849B89C43",
    "SubGeographicalRegion_name": "_A1170111-942A-6ABD-D325-C64886DC4D7D",
    "Line_name": "_AAE94E4A-2465-6F5E-37B1-3E72183A4E44"
  },
  "application_config": {
    "applications": []
  },
  "simulation_config": {
    "start_time": "1358121600",
    "duration": "300",
    "simulator": "GridLAB-D",
    "timestep_frequency": "1000",
    "timestep_increment": "1000",
    "run_realtime": false,
    "simulation_name": "ieee8500",
    "power_flow_solver_method": "NR",
    "model_creation_config": {
      "load_scaling_factor": "1",
      "schedule_name": "ieeezipload",
      "z_fraction": "0",
      "i_fraction": "1",
      "p_fraction": "0",
      "randomize_zipload_fractions": false,
      "use_houses": false
    }
  },
  "test_config": {
    "events": [],
    "appId": ""
  },
  "service_configs": [
    {
      "id": "gridappsd-sensor-simulator",
      "user_options": {
        "sensors-config": {
          "_cd29cea1-8517-4a44-89f0-1a91042830a1": {
            "nominal-value": 120,
            "perunit-confidence-band": 0.01,
            "aggregation-interval": 30,
            "perunit-drop-rate": 0
          },
          "_17593302-8b33-4fd4-b853-d83eabbb522e": {
            "nominal-value": 120,
            "perunit-confidence-band": 0.01,
            "aggregation-interval": 30,
            "perunit-drop-rate": 0
          },
          "_d34e9c1d-84c6-40cb-829d-14576db951ca": {
            "nominal-value": 120,
            "perunit-confidence-band": 0.01,
            "aggregation-interval": 30,
            "perunit-drop-rate": 0
          },
          "_04f69915-92ec-4116-bccb-8fe0a5f4eea9": {
            "nominal-value": 120,
            "perunit-confidence-band": 0.01,
            "aggregation-interval": 30,
            "perunit-drop-rate": 0
          }
        },
        "random-seed": 42,
        "default-aggregation-interval": 30,
        "passthrough-if-not-specified": false,
        "default-perunit-confidence-band": 0.01,
        "default-perunit-drop-rate": 0
      }
    }
  ]
}
```

Take a look at the following measurements: 
```
['_cd29cea1-8517-4a44-89f0-1a91042830a1', '_17593302-8b33-4fd4-b853-d83eabbb522e', '_d34e9c1d-84c6-40cb-829d-14576db951ca', '_04f69915-92ec-4116-bccb-8fe0a5f4eea9']
```

Notice the voltage measurements are reading ~120V, but the power measurements are at 0.

## Hypothesis
I think this is getting messed up by the player file. Here's what I'm seeing in the player file:
```
gridappsd@b702b8f5469c:/tmp/gridappsd_tmp/1257221190$ cat ieeezipload.player 
2019-01-14 00:00:00 UTC,0.425364
+1m,0.42153
+1m,0.418488
+1m,0.39988
+1m,0.382085
+1m,0.366393
```

And here's the clock in the model:
```
gridappsd@b702b8f5469c:/tmp/gridappsd_tmp/1257221190$ head model_startup.glm 
clock {
     timezone "UTC0";
     starttime '2013-01-14 00:00:00';
     stoptime '2013-01-14 00:05:00';
}
#set maximum_synctime=3600
#set suppress_repeat_messages=1
#set relax_naming_rules=1
#set profiler=1
#set minimum_timestep=0.1
```

I suspect this time mismatch is causing the loads to go to 0.

poorva1209, 2019-10-31:
Ok...working on it. I think I know what the issue is. It should be a quick fix.

AAndersn, 2019-11-01:
Brandon, I checked by running a simulation on the 10.0 release, and I am getting 3.6 MW of load fed by the 115-69kV xfmr. I am currently checking the DG output values because the don't look right in the base case.

shpoudel, 2019-11-01:
I am also getting 0 values for VA measurements. 

Also, where can I find the logic for converting **start time** and  **end time** to timestamp?

blthayer, 2019-11-01:
@AAndersn Thanks for the observations!

@shpoudel - assuming you're using Python:
```
from datetime import datetime
dt = datetime(2019, 11, 1, 15, 33)
timestamp = int(dt.timestamp()) 
```

@poorva1209 - any update on this issue?

poorva1209, 2019-11-01:
I have resolved this issue and made an update on the develop tag. Please pull the develop tag container and test it again.

blthayer, 2019-11-01:
@poorva1209 - I seem to be getting non-zero VA measurements from EnergyConsumers when I run the platform with the `develop` tag. So this seems fixed to me.

shpoudel, 2019-11-01:
Same here. Working with **develop** tag gave me VA measurements. 
Thanks, @blthayer and @poorva1209

shpoudel, 2019-11-01:
Randomly, I am getting the following error:
**Process Initialization error: No load schedule data in time series data store. Setting useClimate = False.**

poorva1209, 2019-11-01:
> Randomly, I am getting the following error:
> **Process Initialization error: No load schedule data in time series data store. Setting useClimate = False.**

I will create another issue for this and try to reproduce it.

closed 2019-11-01 by poorva1209