GRIDAPPSD/gridappsd-forum #45: Random error for no load data

poorva1209, 2019-11-01:
Randomly, I am getting the following error:
Process Initialization error: No load schedule data in time series data store. Setting useClimate = False.

Re-running the run.sh solve this.

shpoudel, 2019-11-01:
@poorva1209 and @tonya1: Please let me know when this is fixed.

tonya1, 2019-11-04:
@shpoudel  and @poorva1209 Could I get a little more information on the error?

Where are you seeing the error? Is it after running run.sh?  Is it Is it after starting a simulation?  Which model are you using?  

Thanks,

Tonya

shpoudel, 2019-11-04:
@tonya1: I am getting this error after doing ./run-gridappsd.sh and when I click start on the viz. It occurs for both the 123 bus and 9500 node model.

closed 2022-02-03 by poorva1209