GRIDAPPSD/gridappsd-forum #47: Time out error

therahuljha, 2019-11-05:
@poorva1209 I am facing time out error whenever I try to query the regulator and capacitor mrids for 9500 node. I am running in develop  version. 

 raise TimeoutError("Request not responded to in a timely manner!")
gridappsd.goss.TimeoutError: Request not responded to in a timely manner!

therahuljha, 2019-11-05:
I tried by changing time out to 180. But, still I am getting TimeoutError.

poorva1209, 2019-11-05:
@therahuljha Please add the query request to the comment.

therahuljha, 2019-11-05:
topic = "goss.gridappsd.process.request.data.powergridmodel"
    message = {
        "modelId": model_mrid,
        "requestType": "QUERY_OBJECT_MEASUREMENTS",
        "resultFormat": "JSON",
        "objectType": "PowerTransformer"}     
    obj_msr_reg = gapps.get_response(topic, message, timeout=180)
 
    message = {
        "modelId": model_mrid,
        "requestType": "QUERY_OBJECT_MEASUREMENTS",
        "resultFormat": "JSON",
        "objectType": "LinearShuntCompensator"}     
    obj_msr_cap = gapps.get_response(topic, message, timeout=180)
 
    message = {
        "modelId": model_mrid,
        "requestType": "QUERY_OBJECT_MEASUREMENTS",
        "resultFormat": "JSON",
        "objectType": "EnergyConsumer"}     
    obj_msr_loads = gapps.get_response(topic, message, timeout=180)
    regulators = get_regulators_mrids(gapps, model_mrid)  
    capacitors = get_capacitors_mrids(gapps, model_mrid)

poorva1209, 2019-11-05:
Thanks, I will try it now.

poorva1209, 2019-11-14:
@therahuljha Try starting your application once the simulation has started running.

therahuljha, 2019-11-14:
Yes, I tried that. It is working fine with version 2019.10.0

poorva1209, 2019-11-14:
ok. thanks. Closing this issue.

closed 2019-11-14 by poorva1209