GRIDAPPSD/gridappsd-forum #48: Inverters exceeding rated power in simulation

blthayer, 2019-11-06:
@AAndersn @afisher1 @poorva1209 @craig8 @temcdrm @Jeffrey-Simpson 

Platform version: `v2019.10.0`

It would appear that during the course of a normal simulation, there are several inverters exceeding their rated power, and some exceed it by a significant amount.

Here's my run configuration:

```json
{
  "power_system_config": {
    "GeographicalRegion_name": "_73C512BD-7249-4F50-50DA-D93849B89C43",
    "SubGeographicalRegion_name": "_A1170111-942A-6ABD-D325-C64886DC4D7D",
    "Line_name": "_AAE94E4A-2465-6F5E-37B1-3E72183A4E44"
  },
  "application_config": {
    "applications": []
  },
  "simulation_config": {
    "start_time": "1358179200",
    "duration": "5",
    "simulator": "GridLAB-D",
    "timestep_frequency": "1000",
    "timestep_increment": "1000",
    "run_realtime": false,
    "simulation_name": "ieee8500",
    "power_flow_solver_method": "NR",
    "model_creation_config": {
      "load_scaling_factor": "1",
      "schedule_name": "ieeezipload",
      "z_fraction": "0",
      "i_fraction": "1",
      "p_fraction": "0",
      "randomize_zipload_fractions": false,
      "use_houses": false
    }
  },
  "test_config": {
    "events": [],
    "appId": ""
  },
  "service_configs": []
}
```
Inside my application code, I listen to the simulation output, filter to get VA measurements associated with inverters, and then update my application's internal representation of the inverters with the latest measurement data. I just augmented my internal inverter models to check the incoming measurement against the inverter's rated power (ratedS in the CIM), and was surprised to find that the measurements emitted from the simulation corresponding to the run configuration above result in significant overloads. Here's my logging output:

```
 17:23:22 [WARNING] [InverterSinglePhase]: Equipment pv_1041, phase S2, had its state updated to exceed its rated power by 26.34%! New P: 1010.76, New Q: -0.14, New |S|: 1010.76, Rated |S|: 800.00.
 17:23:22 [WARNING] [InverterSinglePhase]: Equipment pv_1039, phase S2, had its state updated to exceed its rated power by 361.98%! New P: 461.98, New Q: -0.03, New |S|: 461.98, Rated |S|: 100.00.
 17:23:22 [WARNING] [InverterSinglePhase]: Equipment pv_1091, phase S2, had its state updated to exceed its rated power by 87.34%! New P: 187.34, New Q: -0.00, New |S|: 187.34, Rated |S|: 100.00.
 17:23:22 [WARNING] [InverterSinglePhase]: Equipment pv_1103, phase S2, had its state updated to exceed its rated power by 54.00%! New P: 461.99, New Q: -0.03, New |S|: 461.99, Rated |S|: 300.00.
 17:23:22 [WARNING] [InverterSinglePhase]: Equipment pv_1014, phase S1, had its state updated to exceed its rated power by 204.05%! New P: 304.05, New Q: 0.01, New |S|: 304.05, Rated |S|: 100.00.
 17:23:22 [WARNING] [InverterSinglePhase]: Equipment pv_1151, phase S2, had its state updated to exceed its rated power by 295.96%! New P: 1583.82, New Q: -0.35, New |S|: 1583.82, Rated |S|: 400.00.
 17:23:22 [WARNING] [InverterSinglePhase]: Equipment pv_1032, phase S1, had its state updated to exceed its rated power by 6.20%! New P: 212.40, New Q: 0.01, New |S|: 212.40, Rated |S|: 200.00.
```

For your convenience, I've attached two .csv files which can be used to map measurements to inverters and names: [https://app.zenhub.com/files/174419221/9e3a84d9-2af0-4c8f-b90c-8e34c6d5c8c5/download](https://app.zenhub.com/files/174419221/9e3a84d9-2af0-4c8f-b90c-8e34c6d5c8c5/download)

Alternatively, you could use the dictionary created by the platform when a simulation is started.

shpoudel, 2019-11-06:
@blthayer: I don't know how the rating for inverters is assigned in GridLAB-D. Do you know the multiplier for PV at the time instant you logged your output? Stating your problem in a different way, it seems like PV is producing more power than it's kVA rating when getting a multiplier for variable power generation. 

Also, while looking into the .DSS files for PV, it appears that PVs were created in accordance with unbalanced loads and the warning you have observed are on that phase of unbalanced loads where it has less kW in compared to another (Phase S1 and S2 in your warning). 

So, if the PV multiplier is the same as load, then it's confirmed the warning is because of PVs and loads defined in two different ways when handling the split-phase transformer. I hope you understand what I mean.

temcdrm, 2019-11-06:
Here is the GridLAB-D syntax for one of the overloaded inverters, its paired unbalanced inverter, and the triplex_meter they both share:

object triplex_meter {

  name "sx2766721a_pvmtr";

  parent "sx2766721a";

  phases AS;

  nominal_voltage 120.09;

}
object inverter {
  name "inv_pv_pv_1039";
  parent "sx2766721a_pvmtr";
  phases AS;
  generator_status ONLINE;
  four_quadrant_control_mode CONSTANT_PQ;
  inverter_type FOUR_QUADRANT;
  inverter_efficiency 1.0;
  power_factor 1.0;
  V_base 120.000;
  rated_power 100.000;
  P_Out 60.000;
  Q_Out 0.000;
  object solar {
    name "pv_pv_1039";
    generator_mode SUPPLY_DRIVEN;
    generator_status ONLINE;
    panel_type SINGLE_CRYSTAL_SILICON;
    efficiency 0.2;
    rated_power 100.000;
  };
}

object inverter {

  name "inv_pv_pv_1038";

  parent "sx2766721a_pvmtr";

  phases AS;

  generator_status ONLINE;

  four_quadrant_control_mode CONSTANT_PQ;

  inverter_type FOUR_QUADRANT;

  inverter_efficiency 1.0;

  power_factor 1.0;

  V_base 120.000;

  rated_power 11000.000;

  P_Out 8810.000;

  Q_Out 0.000;

  object solar {

    name "pv_pv_1038";

    generator_mode SUPPLY_DRIVEN;

    generator_status ONLINE;

    panel_type SINGLE_CRYSTAL_SILICON;

    efficiency 0.2;

    rated_power 11000.000;

  };

}

And the original OpenDSS syntax for two unbalanced inverters on the same secondary node (file PV_10penDSSPV.dss).

New PVsystem.PV_1038 phases=1 Bus1=SX2766721A.1 kV=0.12 kVA=11 Pmpp=8.81

New PVsystem.PV_1039 phases=1 Bus1=SX2766721A.2 kV=0.12 kVA=0.1 Pmpp=0.06

This raises two questions:

  1.  Why model these inverters unbalanced in OpenDSS to begin with?  The whole system is to be balanced on the secondary.  Even if the loads are unbalanced on the secondary, the PV system would still be balanced, i.e., one unit connected at 240 volts, representable as two identical units connected at 120 volts.
  2.  The solar object pv_pv_1039 has rated_power of 100 Watts.  That should never produce 461 Watts if driven by a weather file, so, how could it?

_________________________

Tom McDermott, Ph.D., PE, FIEEE
Chief Engineer, Systems Team
Electricity Infrastructure Group
Pacific Northwest National Laboratory
Phone: 509-375-4434
Thomas.McDermott@pnnl.gov

shpoudel, 2019-11-06:
@temcdrm: I have the answer to your first question. The PV files were created by Alexander and me when we were working with an unbalanced load for 9500 nodes. I remember a couple of weeks back you changed the loads to a balanced one. At that time, it was not in our memory that PV files need to be changed as they were created long back in accordance with unbalanced loads.

Yes, It makes sense- "Even if the loads are unbalanced on the secondary, the PV system would still be balanced". Maybe we overlooked the clear definition of split-phase transformer.

blthayer, 2019-11-06:
So, am I correct in that @shpoudel and/or @AAndersn are planning to modify the residential PV to be connected at 240V instead of 120V?

We still need someone to investigate the primary part of the issue, however - how/why are these inverters outputting above their rated power and/or above the rated power of their PV objects? As I first step, it would be good to know if someone can reproduce/confirm the measurements I'm seeing. If it's reproducible, then we need to track down the issue.

@shpoudel - do you have time to reproduce this?

shpoudel, 2019-11-06:
@blthayer: Sorry. I am busy with the PESGM deadline until tomorrow. I can look into it tomorrow evening or early Friday morning only.

Jeffrey-Simpson, 2019-11-06:
Hello,
I am seeing this as well now that Brandon mentioned it.
Here are my PV output for P.
Thanks,
-Jeff

blthayer, 2019-11-06:
@Jeffrey-Simpson - not sure if you tried to attach something to the email or what, but not seeing any of your outputs.

poorva1209, 2019-11-14:
@Jeffrey-Simpson @blthayer @temcdrm @shpoudel What is the status of this issue?

shpoudel, 2019-11-14:
I created a pull request to modify the residential PV to be connected at 240V instead of 120V. But, I am not sure if this will resolve the issue.

blthayer, 2019-11-14:
@poorva1209 - we need to assign it to someone so it can get fixed :) And that someone is not me - I've sunk way too much time into this project and am playing catch-up on my school work.

So in addition to @shpoudel's modification, we need to see what might be causing those inverters to inject more power than their rating. There could be all sorts of possible causes here.

poorva1209, 2019-11-19:
@shpoudel @Jeffrey-Simpson develop tag is updated with latest blazegraph changes. Please test it and let us know if this issue is still present. Please clear your browser cache before testing.

shpoudel, 2019-11-19:
@poorva1209 : Unfortunatrly, I am not able to work on develop tag with the time out error appearing while I run queries. I switched to develop tag to try one more time and got following error in terminal. Do you have any idea what this error means?

[image]

poorva1209, 2019-11-19:
@shpoudel Let's set up a call and debug your develop container.  What time would work for you for tomorrow ?

shpoudel, 2019-11-19:
@poorva1209: Anytime after 1:00 PM works for me.

shpoudel, 2019-11-21:
@blthayer @poorva1209 @temcdrm: I went through this again this morning and would like to point a few things:

1. I ran the queries for inverters with develop tag. It seems to me the blazegraph has not been updated and the inverter's rating I got is still the same based on the old PV definition. 
2. I noticed why the solars were producing more power than it's rated capacity and this lead inverters to exceed the capacity. As Tom pointed the PVs were unbalanced in the DSS file. This is what happened for PV_1039.

New PVsystem.PV_1039 phases=1 Bus1=SX2766721A.2 kV=0.12 kVA=0.1 Pmpp=0.06

As we see the PV_1039 is at Bus SX2766721A.2. There is another PV named PV_1038 at SX2766721A.1 which has a rated power of 8.81kW. Now, the first PV got multiplied with a value from weather file with a rated power of the second PV. This shouldn't have happened as they were separate PV. But, this holds true for all sample examples Brandon provided. 

Now, I believe, if we update the PV with new files, this problem would be solved as PV will be balanced and get multiplied with proper weather files.

poorva1209, 2019-11-21:
@shpoudel Did you execute ./stop.sh -c before pulling new develop containers?

shpoudel, 2019-11-21:
Yes, I did.

poorva1209, 2019-11-21:
@tonya1 Can you work with @shpoudel today to figure out the issue of why he is not seeing updated values in the model in develop tag.

shpoudel, 2019-11-21:
Let me give you more details on what I did before you actually start looking into it.
1. My container is running the latest develop tag.
2. I ran the following query for inverters:

PREFIX r: <http://www.w3.org/1999/02/22-rdf-syntax-ns#>
PREFIX c: <http://iec.ch/TC57/CIM100#>
PREFIX xsd: <http://www.w3.org/2001/XMLSchema#>
SELECT ?inverter_mrid ?inverter_name ?inverter_mode ?inverter_max_q ?inverter_min_q ?inverter_p ?inverter_q ?inverter_rated_s ?inverter_rated_u ?phase_mrid ?phase_name ?phase_p ?phase_q (group_concat(distinct ?phs;separator="\n") as ?phases) 
WHERE {
# Update for your feeder, or remove for all feeders.
VALUES ?feeder_mrid {"_AAE94E4A-2465-6F5E-37B1-3E72183A4E44"}
?s r:type c:PowerElectronicsConnection.
?s c:Equipment.EquipmentContainer ?fdr.
?fdr c:IdentifiedObject.mRID ?feeder_mrid.
?s c:IdentifiedObject.mRID ?inverter_mrid.
?s c:IdentifiedObject.name ?inverter_name.
?s c:PowerElectronicsConnection.p ?inverter_p.
?s c:PowerElectronicsConnection.q ?inverter_q.
?s c:PowerElectronicsConnection.ratedS ?inverter_rated_s.
?s c:PowerElectronicsConnection.ratedU ?inverter_rated_u.
OPTIONAL {
?s c:PowerElectronicsConnection.inverterMode ?inverter_mode.
?s c:PowerElectronicsConnection.maxQ ?inverter_max_q.
?s c:PowerElectronicsConnection.minQ ?inverter_min_q.
}
OPTIONAL {
?pecp c:PowerElectronicsConnectionPhase.PowerElectronicsConnection ?s.
?pecp c:IdentifiedObject.mRID ?phase_mrid.
?pecp c:IdentifiedObject.name ?phase_name.
?pecp c:PowerElectronicsConnectionPhase.p ?phase_p.
?pecp c:PowerElectronicsConnectionPhase.q ?phase_q.
?pecp c:PowerElectronicsConnectionPhase.phase
?phsraw bind(strafter(str(?phsraw),"SinglePhaseKind.") as ?phs)
}
}
GROUP BY ?inverter_mrid ?inverter_name ?inverter_rated_s ?inverter_rated_u ?inverter_p ?inverter_q ?phase_mrid ?phase_name ?phase_p ?phase_q ?inverter_mode ?inverter_max_q ?inverter_min_q
ORDER BY ?inverter_mrid

AAndersn, 2019-12-05:
@temcdrm @poorva1209  @tonya1 Please provide an update if Shiva's DSS file changes have been imported into Blazegraph. When he last checked, the PV file definitions were still the same (unbalanced).

tonya1, 2019-12-05:
@shpoudel I am able to use your query with blazegraph.  I get back results but I am not sure how to interpret them to verify the data is correct.  What should I look for?

shpoudel, 2019-12-05:
@tonya1: You should get **inverter_rated_u** as 208 and phases as s1 s2 for each PV from the query result. 

Also, another check would be: See the full query list and search for "pv_3_s1" or "pv_1_s1". You should not find any words like that in the result.

tonya1, 2019-12-07:
@shpoudel I generated the XML from your pull request and created an updated blazegraph container.  Can you test the gridappsd/blazegraph:balanced container?

AAndersn, 2019-12-07:
Tonya, thanks for updating the containers. I'll now be able to create the
final JSON events files with stable MRIDs for the eval demo.

Has this been added to the ~develop release?

shpoudel, 2019-12-07:
@tonya1 Thanks for updating the blazegraph container. I ran the query with newer image (gridappsd/blazegraph:balanced) and PVs are balanced now.

However, I can see 4 measurement mrid with a single PV. I guess there should be 2 for each PV if defined from s1 and s2 (Similar to balanced Load). Let me check what different values those 4 ids contain in the simulation output. I will try to verify by this weekend if this has been resolved fully.

blthayer, 2019-12-10:
@shpoudel - I'll try to find time to look into this next week (the week of 12/16), but I may not have time. Are we still getting overloads?

shpoudel, 2019-12-10:
@blthayer: I didn't see any overloads after the blazegraph was updated with balanced PV. But, I would like this to be verified by you once you have time for this.

 The log output you presented while opening this ticket should be fixed with balanced PV. 

https://github.com/GRIDAPPSD/gridappsd-forum/issues/48#issuecomment-557206765

Does this comment make sense to you and during unbalance? I believe this was the reason for inverters exceeding the rated power for unbalance cases.

AAndersn, 2019-12-10:
@shpoudel Is the blazegraph:balanced part of the -develop container?

shpoudel, 2019-12-10:
@AAndersn: I don't know if it is part of -develop. But, if you are looking for updated blazegraph to update MRIDs and create JSON files, you have to change the image in docker-compose.yml.

image: gridappsd/blazegraph${GRIDAPPSD_TAG} to
image: gridappsd/blazegraph:balanced

@tonya1: Is this correct explanation to Alexander question?

poorva1209, 2020-03-09:
@shpoudel @blthayer @AAndersn Is this issue still relevant?

blthayer, 2020-03-10:
@poorva1209 I suspect it is, but haven't tested since the fall and won't have time to for a while. Maybe @shpoudel or @AAndersn might have time to test it.

shpoudel, 2020-04-02:
@poorva1209 @tonya1: Yes, this has been fixed once Tonya made the PV balanced with a new blazegraph image. I ran the simulation with a newer image (gridappsd/blazegraph:balanced) and PVs are balanced now. Also, I see the PV generation is within the limit. 

So, it seems this issue is fixed to me. However, I would like it to be verified by @blthayer or @Jeffrey-Simpson as you guys are actively working on PV inverters.

poorva1209, 2020-04-02:
@shpoudel Thanks for the quick update. Can you also check it with gridappsd/blazegraph:develop to make sure those changes are merged in that image too.

shpoudel, 2020-04-02:
Sure @poorva1209, I will get back to you soon.

shpoudel, 2020-04-02:
@poorva1209: While running the query, it seems like those changes are not merged into gridappsd/blazegraph:develop

AAndersn, 2020-04-02:
Shiva,

We did not merge those changes into the develop tag prior to the EIOC
evaluation so as not to change the MRIDs of the model.

Now is a good time to check the impact of doing a final model update.

Thanks,

Alexander
*----*
*Alexander Anderson, PhD*

Chair, IEEE Smart Village Partner Engagement
IEEE GHTC 2019 Industry Program Chair
[image: Inline image 1]
Join us in our mission to empower 50 million people by 2025:
https://smartvillage.ieee.org/join-us/

temcdrm, 2020-04-02:
Which branch of Powergrid-Models was this done on?

shpoudel, 2020-04-02:
@AAndersn: Yes, that change was not included for MRID to keep consistent.

@temcdrm: This belongs to develop tag:

https://github.com/GRIDAPPSD/Powergrid-Models/tree/develop/blazegraph/test/dss/WSU

blthayer, 2020-04-02:
@shpoudel - unfortunately I will not be able to verify this for some time, as I'm not actively working for PNNL right now (I'll be back in mid June). I can make exceptions for urgent matters, but this doesn't seem to fall into that category :)

poorva1209, 2020-06-09:
@Jeffrey-Simpson We have updated the develop to use the balanced PV version of blazegraph models.  Can you please check that and see If inverters are not exceeding rated power.

Jeffrey-Simpson, 2020-06-11:
I put a check in the DER-Dispatch to check for PV power greater the ratings. None produce more than they are rated. So we can close this.

poorva1209, 2020-06-11:
Thanks @Jeffrey-Simpson. I am closing this.

closed 2020-06-11 by poorva1209