GRIDAPPSD/gridappsd-forum #49: Regulator outage event doesn't work and platform throws error

blthayer, 2019-11-08:
Platform version: `v2019.10.0`

Simulation run configuration:
```json
{
  "power_system_config": {
    "GeographicalRegion_name": "_73C512BD-7249-4F50-50DA-D93849B89C43",
    "SubGeographicalRegion_name": "_A1170111-942A-6ABD-D325-C64886DC4D7D",
    "Line_name": "_AAE94E4A-2465-6F5E-37B1-3E72183A4E44"
  },
  "application_config": {
    "applications": []
  },
  "simulation_config": {
    "start_time": "1358179200",
    "duration": "1200",
    "simulator": "GridLAB-D",
    "timestep_frequency": "1000",
    "timestep_increment": "1000",
    "run_realtime": true,
    "simulation_name": "ieee8500",
    "power_flow_solver_method": "NR",
    "model_creation_config": {
      "load_scaling_factor": "1",
      "schedule_name": "ieeezipload",
      "z_fraction": "0",
      "i_fraction": "1",
      "p_fraction": "0",
      "randomize_zipload_fractions": false,
      "use_houses": false
    }
  },
  "test_config": {
    "events": [
      {
        "message": {
          "forward_differences": [
            {
              "object": "_CE091BBD-77FD-4E8D-8815-EEF79D540108",
              "attribute": "TapChanger.step",
              "value": 10
            }
          ],
          "reverse_differences": [
            {
              "object": "_CE091BBD-77FD-4E8D-8815-EEF79D540108",
              "attribute": "TapChanger.step",
              "value": 5
            }
          ]
        },
        "event_type": "ScheduledCommandEvent",
        "occuredDateTime": 1358179260,
        "stopDateTime": 1358180260
      },
      {
        "allOutputOutage": false,
        "allInputOutage": false,
        "tag": "mc45mjk2",
        "inputList": [
          {
            "name": "vreg3_a",
            "type": "Regulator",
            "mRID": [
              "_CE091BBD-77FD-4E8D-8815-EEF79D540108"
            ],
            "attribute": "TapChanger.step",
            "phases": [
              {
                "phaseLabel": "A",
[... 56 lines cut ...]
87b163e191eb        mysql/mysql-server:5.7            "/entrypoint.sh mysq…"   5 minutes ago       Up 5 minutes (healthy)   0.0.0.0:3306->3306/tcp, 33060/tcp                                                                                                              gridappsd-docker_mysql_1
35175147d5dd        gridappsd/pyvvo:mysql-latest      "docker-entrypoint.s…"   5 minutes ago       Up 5 minutes             3306/tcp, 33060/tcp                                                                                                                            gridappsd-docker_pyvvo-db_1
7786922ad4d0        redis:3.2.11-alpine               "redis-server --appe…"   5 minutes ago       Up 5 minutes             0.0.0.0:6379->6379/tcp                                                                                                                         gridappsd-docker_redis_1
1621c5ab696e        aaa5a6a31acc                      "/bin/sh"                6 minutes ago       Created                                                                                                                                                                 pycharm_helpers_PY-192.7142.42
```

And after the error:
```
thay838@ubuntu:~$ docker ps -a
CONTAINER ID        IMAGE                             COMMAND                  CREATED             STATUS                   PORTS                                                                                                                                          NAMES
4a1e7cb59234        gridappsd/pyvvo:latest            "python -u /opt/.pyc…"   5 minutes ago       Up 5 minutes             0.0.0.0:53581->53581/tcp                                                                                                                       pyvvo_pyvvo_1
27443f7f11f3        gridappsd/pyvvo:mysql-latest      "docker-entrypoint.s…"   5 minutes ago       Up 5 minutes             3306/tcp, 33060/tcp                                                                                                                            pyvvo_db_1
b7061fb11673        gridappsd/viz:v2019.10.0          "npm start"              6 minutes ago       Up 6 minutes             0.0.0.0:8080->8082/tcp                                                                                                                         gridappsd-docker_viz_1
6d81a31fdb3b        gridappsd/pyvvo:latest            "python /usr/src/gri…"   6 minutes ago       Up 6 minutes                                                                                                                                                            gridappsd-docker_pyvvo_1
e3f1df71da3f        gridappsd/gridappsd:v2019.10.0    "/gridappsd/entrypoi…"   6 minutes ago       Up 6 minutes             0.0.0.0:8000->8000/tcp, 8001-8887/tcp, 0.0.0.0:8888->8888/tcp, 0.0.0.0:61613-61614->61613-61614/tcp, 0.0.0.0:61616->61616/tcp, 8889-9000/tcp   gridappsd-docker_gridappsd_1
5effe522e0e6        gridappsd/blazegraph:v2019.10.0   "/docker-entrypoint.…"   7 minutes ago       Up 6 minutes             0.0.0.0:8889->8080/tcp                                                                                                                         gridappsd-docker_blazegraph_1
a598a2b0e6f2        gridappsd/influxdb:v2019.10.0     "/entrypoint.sh infl…"   7 minutes ago       Up 6 minutes             0.0.0.0:8086->8086/tcp                                                                                                                         gridappsd-docker_influxdb_1
d9c7950dcedf        gridappsd/proven:v2019.10.0       "java -jar /opt/paya…"   7 minutes ago       Up 6 minutes             4848/tcp, 8009/tcp, 8181/tcp, 0.0.0.0:18080->8080/tcp                                                                                          gridappsd-docker_proven_1
87b163e191eb        mysql/mysql-server:5.7            "/entrypoint.sh mysq…"   7 minutes ago       Up 6 minutes (healthy)   0.0.0.0:3306->3306/tcp, 33060/tcp                                                                                                              gridappsd-docker_mysql_1
35175147d5dd        gridappsd/pyvvo:mysql-latest      "docker-entrypoint.s…"   7 minutes ago       Up 6 minutes             3306/tcp, 33060/tcp                                                                                                                            gridappsd-docker_pyvvo-db_1
7786922ad4d0        redis:3.2.11-alpine               "redis-server --appe…"   7 minutes ago       Up 6 minutes             0.0.0.0:6379->6379/tcp                                                                                                                         gridappsd-docker_redis_1
1621c5ab696e        aaa5a6a31acc                      "/bin/sh"                8 minutes ago       Created                                                                                                                                                                 pycharm_helpers_PY-192.7142.42

```
Additionally, it's worth noting that my application can still command that regulator after the 60 second mark.

Jeffrey-Simpson, 2019-11-08:
I don't know about REST error.

But does the tap pos change?

The outage event should like this if you are sending it from the Simulation run configuration:
Outage_event = {
    "allOutputOutage": False,
    "allInputOutage": False,
    "inputOutageList": [{"objectMRID": ""_CE091BBD-77FD-4E8D-8815-EEF79D540108", "attribute": " TapChanger.step"},
                        ],
    "outputOutageList": [],
    "event_type": "CommOutage",
    "occuredDateTime": 1358179260
    "stopDateTime": 1358180260
}

Thanks,
-Jeff

https://github.com/Jeffrey-Simpson https://github.com/craig8 https://github.com/poorva1209

At https://github.com/poorva1209's request, here is my output of docker ps -a before the error:

AAndersn, 2019-11-18:
I will try testing the event with the develop and 10.0 branch and see if the error is reproducible. When Jeff and I ran the event together during a Zoom meeting, we did not encounter this error.

poorva1209, 2020-06-23:
**Note - Please use blazegraph:v2020.03.0 tag to test this**

@afisher1 @Jeffrey-Simpson I tested Event2.json with 9500 model and TapChanger.step value is not changing as expected. 

Steps to reproduce:
1.	Configure 9500 simulation on Viz
2.	Set start time as 2013-07-14 08:09:00
3.	Set duration as 3600
4.	Upload Event2.json in Test Configuration (It contains one ScheduledCommentEvent and one CommOutage event)
5.	Start the simulation
 
Based on the event file, I am expecting measurement_mrid _0f8202ca-a4bf-4c7e-9302-601919c09992 to change from 5 to 10 at 2013-07-14 08:10:00 but it remains 5.

@singha42 : FYI.

Jeffrey-Simpson, 2020-06-23:
Hello,
Okay where is the repo with Event2.json you are using located?
Thanks,
-Jeff

poorva1209, 2020-06-23:
@Jeffrey-Simpson @afisher1 I just sent Event2.json over email to you both as GitHub was not allowing me to attach it here.

AAndersn, 2020-06-23:
@Jeffrey-Simpson Check if you are using the right model - we just pushed a model update with new MRIDs less than two weeks ago. I have not created new event files with the updated MRIDs

poorva1209, 2020-06-23:
> @Jeffrey-Simpson Check if you are using the right model - we just pushed a model update with new MRIDs less than two weeks ago. I have not created new event files with the updated MRIDs

@Jeffrey-Simpson @AAndersn That's why I added this note in my testing steps:  **Note - Please use blazegraph:v2020.03.0 tag to test this**

Jeffrey-Simpson, 2020-06-23:
Hello,

First I believe I know what is wrong. The CommOutage is taking place at the same time as the reverse command is sent so it will block the change to back to the original tap setting.

Second I have not been able to test through the viz yet since it looks like some of the CommOutage  information is lost by the time it gets to ProcessEvent.
Namely the ‘event_type’ seems to get lost for the CommOutage. I am using develop for all tags except blazegraph.

{'power_system_config': {'GeographicalRegion_name': '_73C512BD-7249-4F50-50DA-D93849B89C43',
  'SubGeographicalRegion_name': '_A1170111-942A-6ABD-D325-C64886DC4D7D',
  'Line_name': '_AAE94E4A-2465-6F5E-37B1-3E72183A4E44'},
'application_config': {'applications': []},
'simulation_config': {'start_time': '1592945413',
  'duration': '120',
  'simulator': 'GridLAB-D',
  'timestep_frequency': '1000',
  'timestep_increment': '1000',
  'run_realtime': True,
  'simulation_name': 'test9500new',
  'power_flow_solver_method': 'NR',
  'model_creation_config': {'load_scaling_factor': '1',
   'schedule_name': 'ieeezipload',
   'z_fraction': '0',
   'i_fraction': '1',
   'p_fraction': '0',
   'randomize_zipload_fractions': False,
   'use_houses': False}},
'test_config': {'events': [{'allInputOutage': False,
    'allOutputOutage': False,
    'inputOutageList': [{'objectMRID': '_E44571D4-52CE-4ACE-9012-37DEBF17FCF8',
      'attribute': 'TapChanger.step'}],
    'outputOutageList': [],
    'occuredDateTime': 1373811010,
    'stopDateTime': 1373814000},
   {'message': {'forward_differences': [{'object': '_E44571D4-52CE-4ACE-9012-37DEBF17FCF8',
       'attribute': 'TapChanger.step',
       'value': 10}],
     'reverse_differences': [{'object': '_E44571D4-52CE-4ACE-9012-37DEBF17FCF8',
       'attribute': 'TapChanger.step',
       'value': 5}]},
    'event_type': 'ScheduledCommandEvent',
    'occuredDateTime': 1373810940,
    'stopDateTime': 1373811060}],
  'appId': ''},
'service_configs': []}

Thanks,
-Jeff

poorva1209, 2020-06-23:
@Jeffrey-Simpson 

For first  - We never reach a point of checking the value change back to original because it never gets updated in the first place. It always remains 5. Are you seeing it changed to 10 at 8:10:00 time? 
For second - Bibi and Nhuy are working on resolving the error right now.

Jeffrey-Simpson, 2020-06-23:
Hello,
Okay.  If I test with just the Scheduled event it works fine.  The tap goes to 10 and then back to 5. I don’t know if attachments work in the github issues so I will cc you also.
Thanks,
-Jeff

Jeffrey-Simpson, 2020-06-26:
Hello,
I tested to day with the latest develop tag and the viz. It works fine.

If it is still not working for you then I think it might be something with the time in the event file not being handle the same as the simulation start time in the viz.
Try moving the start time back an hour and see if it works. If this the case we should probably fix it to be handle the same.

Thanks,
-Jeff

closed 2021-03-11 by poorva1209