GRIDAPPSD/gridappsd-forum #50: CPLEX tested and not working

shpoudel, 2019-11-12:
@tonya1: I checked with the wsu-restoration container that you created for the WSU application. I tested and found the CPLEX is still not working as desired.

tonya1, 2019-11-12:
@shpoudel Could you provide some examples I could use to verify that cplex is working?

shpoudel, 2019-11-12:
@tonya1: Inside the container you have created, there is a file named **check.py** which calls CPLEX while it is executing python check.py. If it runs, then cplex is working.

tonya1, 2019-11-21:
@shpoudel I am running into some trouble getting the correct python version for our older version of CPLEX.  

Which version of CPLEX are you running?  and which versions of python does it support?  Maybe something like  
   ls -l /opt/ibm/ILOG/CPLEX_Studio1261/cplex/python/

shpoudel, 2019-11-21:
@tonya1 : Following is the version of CPLEX and python in my windows machine. 

[image]

Please let me know if you need any more detailed information as you get from (ls -l in linux).

shpoudel, 2019-11-21:
In the folder you mentioned: It has 2.7, 3.5, and 3.6 in windows. But no idea how python 3.7 is working on it.

tonya1, 2019-11-22:
cc: @Jon8080 

@shpoudel 
I have built a WSU container with python 3.4 ( the python version supported in the EIOC version of cplex)  I was able to run the cplex setup.py within the container but I am still not able to successfully run the check.py script.

```
python3 check.py
Traceback (most recent call last):
  File "check.py", line 37, in <module>
    seating_model.solve(CPLEX(msg = 1))
NameError: name 'CPLEX' is not defined
```

Thanks for providing your version information.  We have version 12.6.1.0 in the EIOC, will this still work with your code?

Now that I see you are running a windows version of cplex, I don't think it can be mounted within a linux container and work.  I can keep working on this version in the EIOC at PNNL, if you have any suggestions for getting the check.py script to work, please send them my way.

shpoudel, 2019-11-22:
@tonya1: I updated check.py to simple problem than before and also modified few lines to handle the optimization model. Now, the error is different from what you got before and it looks like:

raise PulpSolverError("PuLP: cannot execute "+self.path)
pulp.solvers.PulpSolverError: PuLP: cannot execute cplex

While there are different reasons for this error, this also comes up when CPLEX is not installed properly. One quick check for me would be to just type **cplex** in the container terminal and it should print something like "Welcome to IBM....". I don't know about docker much but in other terminal it  will print some message.

tonya1, 2019-11-22:
cc: @Jon8080

@shpoudel, I am able to run cplex within the container using the cplex command.  I don't see the updates to the check.py in the repository.  I have been using the master branch, should I be using a different branch?

```
root@befc36a99ae1:/usr/src/gridappsd-restoration# cplex

Welcome to IBM(R) ILOG(R) CPLEX(R) Interactive Optimizer 12.6.1.0
  with Simplex, Mixed Integer & Barrier Optimizers
5725-A06 5725-A29 5724-Y48 5724-Y49 5724-Y54 5724-Y55 5655-Y21
Copyright IBM Corp. 1988, 2014.  All Rights Reserved.

Type 'help' for a list of available commands.
Type 'help' followed by a command name for more
information on commands.

CPLEX> quit
```

shpoudel, 2019-11-22:
@tonya1: I think I know what's happening. Will you be able to hold a meeting for half an hour anytime today? We are in a different container, I guess, and that's why I am not seeing CPLEX. I believe we can fix this together.

tonya1, 2020-01-06:
We were able to get this to work using the following dockerfile.  

# Use the base application container to allow the application to be controlled
# from the gridappsd container.
FROM gridappsd/app-container-base:3.4
 
# Add the TIMESTAMP variable to capture the build information from
# the travis docker build command and add them to the image.
ARG TIMESTAMP
RUN echo $TIMESTAMP > /dockerbuildversion.txt
 
 
RUN apt-get update \
    && apt-get install -y \
       libatlas3-base
 
 
# Pick a spot to put our application code
# (note gridappsd-python is located at /usr/src/gridappsd-python)
# and is already installed in the app-container-base environment.
WORKDIR /usr/src/gridappsd-restoration
 
# Add dependencies to the requirements.txt file before
# uncommenting the next two lines
# COPY requirements.txt ./
# RUN RUN pip install --no-cache-dir -r requirements.txt
RUN pip install networkx numpy pulp
 
# RUN apt-get install software-properties-common
# RUN add-apt-repository ppa:openjdk-r/ppa
# RUN apt-get update
# RUN apt-get install -y openjdk-8-jdk
 
 
# Copy all of the source over to the container.
COPY . .
 
# RUN chmod +x cplex_studio129.linux-x86-64.bin
# RUN ./cplex_studio129.linux-x86-64.bin
# Use a symbolic link to the sample app rather than having to
# mount it at run time (note can still be overriden in docker-compose file)
RUN ln -s /usr/src/gridappsd-restoration/wsu_res_app.config /appconfig
 
CMD cd /opt/ibm/ILOG/CPLEX_Studio1261/cplex/python/3.4/x86-64_linux/ ; python setup.py install ; cd /usr/src/gridappsd-restoration ; python /usr/src/gridappsd-python/register_app.py

closed 2020-01-06 by tonya1