GRIDAPPSD/gridappsd-forum #51: Error while running EnergyConsumer query

shpoudel, 2019-11-14:
@poorva: Could you please look into the following error message and point any probable cause? I was running a query for EnergyConsumer in 9500 nodel model (v2019.10.0) and got the following error in terminal.

{'error': {'message': 'org.apache.jena.query.QueryParseException: Lexical error at line 4, column 125.  Encountered: "\\n" (10), after : "\\""\n\tat org.apache.jena.sparql.lang.ParserSPARQL11.perform(ParserSPARQL11.java:109)\n\tat org.apache.jena.sparql.lang.ParserSPARQL11.parse$(ParserSPARQL11.java:52)\n\tat org.apache.jena.sparql.lang.SPARQLParser.parse(SPARQLParser.java:34)\n\tat org.apache.jena.query.QueryFactory.parse(QueryFactory.java:147)\n\tat org.apache.jena.query.QueryFactory.create(QueryFactory.java:79)\n\tat org.apache.jena.query.QueryFactory.create(QueryFactory.java:52)\n\tat org.apache.jena.query.QueryFactory.create(QueryFactory.java:40)\n\tat gov.pnnl.goss.gridappsd.data.handlers.BlazegraphQueryHandler.query(BlazegraphQueryHandler.java:95)\n\tat gov.pnnl.goss.gridappsd.data.BGPowergridModelDataManagerImpl.queryResultSet(BGPowergridModelDataManagerImpl.java:202)\n\tat gov.pnnl.goss.gridappsd.data.BGPowergridModelDataManagerImpl.query(BGPowergridModelDataManagerImpl.java:177)\n\tat gov.pnnl.goss.gridappsd.data.BGPowergridModelDataManagerHandlerImpl.handle(BGPowergridModelDataManagerHandlerImpl.java:32)\n\tat gov.pnnl.goss.gridappsd.data.DataManagerImpl.processDataRequest(DataManagerImpl.java:149)\n\tat gov.pnnl.goss.gridappsd.process.ProcessEvent.onMessage(ProcessEvent.java:213)\n\tat pnnl.goss.core.client.GossClient$1.run(GossClient.java:321)\n\tat java.lang.Thread.run(Thread.java:748)\n'}, 'responseComplete': True, 'id': 'fac4a616-2950-4b94-b0a2-91dcce5de816'}

poorva1209, 2019-11-14:
Seems like query is not written correctly. What query are you running ?

shpoudel, 2019-11-14:
Never mind. I figured it out. It was running well on "http://localhost:8889/bigdata/#query" but throwing error while running from my application terminal.

closed 2019-11-14 by shpoudel