GRIDAPPSD/gridappsd-forum #52: Proven error when trying to query sensor service data

blthayer, 2019-11-18:
@poorva1209 @ericstephan @craig8 

Platform version: `develop` (pulled morning of 2019-11-18)

### To reproduce:
1. Run a simulation with the following configuration:
```json
{
  "power_system_config": {
    "GeographicalRegion_name": "_73C512BD-7249-4F50-50DA-D93849B89C43",
    "SubGeographicalRegion_name": "_A1170111-942A-6ABD-D325-C64886DC4D7D",
    "Line_name": "_AAE94E4A-2465-6F5E-37B1-3E72183A4E44"
  },
  "application_config": {
    "applications": []
  },
  "simulation_config": {
    "start_time": "1358121600",
    "duration": "300",
    "simulator": "GridLAB-D",
    "timestep_frequency": "1000",
    "timestep_increment": "1000",
    "run_realtime": false,
    "simulation_name": "ieee8500",
    "power_flow_solver_method": "NR",
    "model_creation_config": {
      "load_scaling_factor": "1",
      "schedule_name": "ieeezipload",
      "z_fraction": "0",
      "i_fraction": "1",
      "p_fraction": "0",
      "randomize_zipload_fractions": false,
      "use_houses": false
    }
  },
  "test_config": {
    "events": [],
    "appId": ""
  },
  "service_configs": [
    {
      "id": "gridappsd-sensor-simulator",
      "user_options": {
        "random-seed": 42,
        "default-aggregation-interval": 30,
        "passthrough-if-not-specified": false,
        "default-perunit-confidence-band": 0.01,
        "default-perunit-drop-rate": 0
      }
    }
  ]
}
```
2. Get the simulation ID associated with the simulation. In my example, it'll be `264008561`.
3. Query Proven for sensor simulator output. Send the following (adapting for your simulation ID) to the topic `'/queue/goss.gridappsd.process.request.data.timeseries'`:
```json
{
  "queryMeasurement": "gridappsd-sensor-simulator",
  "queryFilter": {
    "simulation_id": "264008561",
    "measurement_mrid": [
      "_d8e8c24f-accc-4b19-95b9-f21253ea7911",
      "_94fe040a-72ad-4e51-b8db-8683332208d2",
      "_6cee1529-f71f-4d69-b9d6-fc4b52dff9b9",
      "_53a40ddc-ed5e-4ac1-9920-c260c419123a",
      "_2ae476db-bf11-4a35-9017-89c539fc5b98",
      "_05fd16e4-46be-43a4-a858-2eed79d23af6",
      "_3ed69627-c3b3-406b-aa83-4964f3ed058b",
      "_3a60c1ec-16aa-47f8-886b-7b80642830c3",
      "_d6e5e825-cf33-42a7-81d6-90be3b43afd3",
[... 4184 lines cut ...]
      "_3110749f-8634-4787-83a4-18befc517bec",
      "_03e562a3-b9a3-48bf-bd30-b33dd5ae2579",
      "_201a5763-7b43-4799-b629-9f8241a0e67b",
      "_dccfcc4a-bc16-4678-9606-47d92e21fa90",
      "_8e1bad7d-2e4c-4d03-bdc4-72ae16a24d8f",
      "_0ab3390f-250e-4f58-86e9-0dd2665c0ce0",
      "_2e31e76f-7ee8-4dba-827c-c6e15ad29c95",
      "_2fa1de79-4ef2-4221-85e7-a7fda80f6c00",
      "_cecd7022-7e77-4af8-9a5b-a12271d0532e",
      "_dc8547b5-6d66-467e-a682-269faface07e",
      "_d6ab6fd7-63ed-4792-a4af-0037c6e6336f",
      "_c7f9d14e-9c2c-4184-9595-534b9953def4",
      "_026bfb48-05e9-487d-9e34-9504f5341534",
      "_2b97e10e-c42f-457c-9a39-ba154db8663a",
      "_f3ea6e8a-f35b-4725-b72d-f365b1404d69",
      "_ff4909bf-4b25-4f4e-a455-e10c450c2b90",
      "_cad6a8cd-94b0-4486-8125-6fef3052c07f",
      "_4dd0be45-09b7-4147-aef3-99b256a81e6b",
      "_716f5ba2-9ffb-43f8-8231-0dd55c9bcd82",
      "_13a867ff-1952-469c-bd5f-3b363d6307f3",
      "_cf647839-908d-4477-8a8d-49601b84c3ad",
      "_ffa701d9-b6b6-4be7-ae93-0310b44dc1e3",
      "_18781ed1-28cd-4ff7-83ec-9c4316fe036c",
      "_86d43f41-a7cd-4f94-86ca-2a1d04567da8",
      "_b

blthayer, 2019-11-18:
@poorva1209 @ericstephan 
Update: If I split my one large query into 4 queries, it seems to work. So I'll remove the app-critical tag as I have a work-around.

blthayer, 2019-11-18:
Update 2: Splitting the query in half (rather than quarters) also seems to work. Though I am still getting the logging error.

ericstephan, 2019-11-18:
@blthayer I was wondering if there was a threshold in influx for query string size.    I will investigate breaking large queries into multiple queries and then returning one composite result back.

blthayer, 2019-11-18:
@ericstephan 
Thanks!

This isn't a huge priority for me - I need to do the breaking up on my side anyways in order to avoid excessive memory consumption. I just figured I'd let you all know what happens when you try to query everything at once.