GRIDAPPSD/gridappsd-forum #54: Proven should response back with proper error message

poorva1209, 2019-11-18:
Instead of 500 internal server error.

ericstephan, 2020-03-18:
@poorva1209 - I went through the code changes yesterday and will document a list of changes I am making along with the tests that need to be performed to make sure the response messages are adequate.  I also changed the timestamp to be a bit more robust allowing either quoted long or a long variable type.

ericstephan, 2020-03-19:
@poorva1209 - The following responses have been changed

**Writing Simulation Data**

- If a message type cannot be provided: ERROR MESSAGE "Invalid or missing message content type.  Simulation data format irregular or not detected as input, output, or alarm."

**Writing Simulation Output Data**

    If simulation timestamp is invalid ERROR MESSAGE
       "Invalid or missing message content type.  Invalid measurement timestamp."

    if simulationid is blank.  ERROR MESSAGE:  "Invalid or missing message content type.  Simulation id missing."

 **Writing Simulation Input Data**

 	If simulation timestamp is invalid ERROR MESSAGE
       "Invalid or missing message content type.  Invalid measurement timestamp."

 	if simulationid is blank.  ERROR MESSAGE:  "Invalid or missing message content type.  Simulation id missing."

 	if (forwardDifferenceArray == null) && (reverseDifferenceArray == null) ERROR MESSAGE:  "Invalid or missing message content type.  forwarDifferenceArray or reverseDifferenceArray measurement data missing."

**Querying Simulation Data**
      	If an error occurs in InfluxDB:  ERROR MESSAGE "Bad request made to time-series database.\" } " + qr.getError()

ericstephan, 2020-03-26:
@poorva1209 I finished updating messages for writing simulation input and output data.  I am currently working on query messages.  One of the problems with Influx is many times a lack of error response if a query fails.  To help address this I am creating a new debug function that the application developers can use to have Proven return the Influx Query statement it constructed.  This may only help power users that want to debug a native query, but at least it provides more of a response than no results.  @telsethag @bibiraju

ericstephan, 2020-03-26:
@poorva1209 @tonya1 I am ready to have the new release proven-cluster tested.  I am setting the new release version momentarily.  @telsethag @bibiraju

ericstephan, 2020-03-26:
@poorva1209 @tonya1 @bibiraju @telsethag It is tagged as v1.3.6

ericstephan, 2021-04-08:
@poorva1209 - I am not seeing the 500 error you reported when running the sample app.

poorva1209, 2021-04-08:
@ericstephan Which tag are you using for Proven? I am running develop tag for everything including Proven. I just now ran a simulation with 13 node model and sample app. Every 15th timestamp when sample app sends an input command I see 500 Internal server error.

closed 2022-02-03 by poorva1209