GRIDAPPSD/gridappsd-forum #55: fncs_goss_bridge error when running switch events

AAndersn, 2019-11-19:
@poorva1209, Starting this morning, I am encountering a fncs_goss_bridge error when trying to run the scripts for the first fault event on the develop branch. This is a new error that did not occur before. The test manager JSON script is unchanged.

```
Thread-38 ERROR [gov.pnnl.goss.gridappsd.log.LogManagerImpl] - 1574195706000|fncs_goss_bridge.py|402664722|ERROR|system|ERROR
An error occured while trying to translate the update message received

Thread-38 ERROR [gov.pnnl.goss.gridappsd.log.LogManagerImpl] - 1574195706000|fncs_goss_bridge.py|402664722|ERROR|system|ERROR
'NoneType' object has no attribute 'get'
```

```
{
    "source": "fncs_goss_bridge.py",
    "processId": "402664722",
    "timestamp": 1574195706000,
    "processStatus": "RUNNING",
    "logMessage": "incrementing to 122",
    "logLevel": "DEBUG",
    "storeToDb": true
}

remove
{
    "source": "fncs_goss_bridge.py",
    "processId": "402664722",
    "timestamp": 1574195706000,
    "processStatus": "ERROR",
    "logMessage": "'NoneType' object has no attribute 'get'",
    "logLevel": "ERROR",
    "storeToDb": true
}

remove
{
    "source": "fncs_goss_bridge.py",
    "processId": "402664722",
    "timestamp": 1574195706000,
    "processStatus": "ERROR",
    "logMessage": "An error occured while trying to translate the update message received",
    "logLevel": "ERROR",
    "storeToDb": true
}

remove
{
    "source": "fncs_goss_bridge.py",
    "processId": "402664722",
    "timestamp": 1574195706000,
    "processStatus": "RUNNING",
    "logMessage": "translating following message for fncs simulation 402664722 {\"simulation_id\": \"402664722\", \"message\": {\"timestamp\": 1373814120, \"difference_mrid\": \"_ce166b73-8409-4576-9a7c-4d58fc178e7b\", \"forward_differences\": [{\"object\": \"_302E3119-B3ED-46A1-87D5-EBC8496357DF\", \"attribute\": \"Switch.open\", \"value\": 1.0}, {\"object\": \"_2FA4B41B-C31B-4861-B8BB-941A8DFD1B41F\", \"attribute\": \"Switch.open\", \"value\": 1.0}, {\"object\": \"_A0E0AB93-FFC2-471B-B84C-19015CB15ED2\", \"attribute\": \"Switch.open\", \"value\": 1.0}], \"reverse_differences\": [{\"object\": \"_302E3119-B3ED-46A1-87D5-EBC8496357DF\", \"attribute\": \"Switch.open\", \"value\": 0.0}, {\"object\": \"_2FA4B41B-C31B-4861-B8BB-941A8DFD1B41\", \"attribute\": \"Switch.open\", \"value\": 0.0}, {\"object\": \"_A0E0AB93-FFC2-471B-B84C-19015CB15ED2\", \"attribute\": \"Switch.open\", \"value\": 0.0}]}}",
    "logLevel": "DEBUG",
    "storeToDb": true
}

remove
{
    "source": "fncs_goss_bridge.py",
    "processId": "402664722",
    "timestamp": 1574195706000,
    "processStatus": "RUNNING",
    "logMessage": "fncs events ['402664722']",
    "logLevel": "DEBUG",
    "storeToDb": true
}
```

[image]

I will try re-installing the develop branch to see if that corrects the problem and update the issue with any results.

poorva1209, 2019-11-19:
Can you check that you are using equipment mrids and not measurement mrids in the fault scripts? I have seen this error before with that cause.

AAndersn, 2019-11-19:
Yes, I am using equipment MRIDs. I am trying to see if the the equipment MRIDs were incorrectly updated when starting the platform. When I query blazegraph, I still get the same MRIDs as in the test script.

poorva1209, 2019-11-19:
ok. Yeah try again with updated develop. If this still happens then send me the link for script I can debug it.

AAndersn, 2019-11-19:
I have traced the problem down to the PV Farm. It appears that during the last update of the model, something with PV units broke. 

I verified that I had the latest release by deleting all docker containers and downloading new ones using

```
sudo docker rm -f $(docker ps -a -q)
sudo docker rmi -f $(docker images -q) 
./run.sh -t develop
```

I have attached the test JSON script. Simulation time is 2013-07-14 08:00:00 The equipment MRID is still the same in Blazegraph. [Event 1 Switch 3.zip](https://github.com/GRIDAPPSD/gridappsd-forum/files/3866543/Event.1.Switch.3.zip)

AAndersn, 2019-11-19:
FYI, the first two events using the same MRIDs as last week execute successfully. 
[Event 1 Switch 1+2.zip](https://github.com/GRIDAPPSD/gridappsd-forum/files/3866544/Event.1.Switch.1%2B2.zip)

poorva1209, 2019-11-19:
I am not sure what should be done here to resolve this. Do we need to update the DSS files for PV changes or re- create the measurements for PV units. Adding @shpoudel @Jeffrey-Simpson to the conversation.

temcdrm, 2019-11-19:
The OpenDSS files were updated to have balanced PV systems. Therefore, their mRIDs will have changed and their measurements need to be recreated.

poorva1209, 2019-11-19:
@tonya1 Please read Tom's previous comment and confirm that you have created all the needed measurements for latest power grid models.

AAndersn, 2019-11-19:
@temcdrm I have verified all the MRIDs through Blazegraph on the http://localhost:8889/bigdata/#query

```

PREFIX r:  <http://www.w3.org/1999/02/22-rdf-syntax-ns#>

PREFIX c:  <http://iec.ch/TC57/CIM100#>

SELECT ?eqtype ?eqname ?eqid WHERE {

VALUES ?fdrid {"_AAE94E4A-2465-6F5E-37B1-3E72183A4E44"}  
       
VALUES ?eqname {"ln1047pvfrm_sw"}

?fdr c:IdentifiedObject.mRID ?fdrid.

?eq c:Equipment.EquipmentContainer ?fdr.

?eq c:IdentifiedObject.name ?eqname.

?eq c:IdentifiedObject.mRID ?eqid.

?eq r:type ?rawtype.

  bind(strafter(str(?rawtype),"#") as ?eqtype)

?t c:Terminal.ConductingEquipment ?eq.

}

ORDER by ?eqtype ?eqname ?eqid 
```

shpoudel, 2019-11-19:
The balanced PVs were created for smaller units (i.e., those within the house level). The error is about the switch with the PV farm at 12.47 kV level. So, I don't think the balanced PVs mrids which gt changed because of updated OpenDSS file will affect the switch mrid for PV farm. Or, it does?

I received this kind of error a couple of times before while changing the tag and not updating the mrids in the event file which is not the case with @AAndersn. 

I will try to reproduce this error and let you know.

shpoudel, 2019-11-20:
@AAndersn: I checked with the following file and it worked for me. The simulation start time is : 2013-07-22 05:00:00 and fault is simulated one minute after the start time. One difference with your event file and mine is the way you define occuredDateTime and StopDateTime.

[image]

[Event_1_fault.zip](https://github.com/GRIDAPPSD/gridappsd-forum/files/3866666/Event_1_fault.zip)

AAndersn, 2019-11-20:
@shpoudel , Please use the JSON file I provided in my post on this issue. The original file from Jeff uses a different date and only one switch. After we added the swing-PQ nodes, it is necessary to  trip the CHP and PV farm in the event script order to simulate the scenario correctly.

AAndersn, 2019-11-20:
@shpoudel, To help clear any confusion, I have broken the three switch events into three separate JSON files (which are inlcuded above). The first two execute with no problems (opening the recloser and CHP gen breaker). The third (opening the PV farm breaker) results in the fncs_goss_bridge error.

shpoudel, 2019-11-20:
@AAndersn: The PV switch MRID is not consistent in forward and reverse difference in your  third event file. I corrected it and it is working as desired now.

AAndersn, 2019-11-20:
@shpoudel : Thanks for catching the stray character in the MRID. After staring at them for too long, they all start to look the same.

closed 2019-11-20 by AAndersn