GRIDAPPSD/gridappsd-forum #58: Error when querying proven

blthayer, 2019-11-27:
@craig8 will edit this issue or add a comment with directions for loading up the data into influx.

@poorva1209 @ericstephan 

Platform version: `develop` (as of 2019-11-27)

Topic: `goss.gridappsd.process.request.data.timeseries`

Query:
`{  "queryMeasurement": "simulation",  "queryFilter": {    "simulation_id": "1945523778"},  "responseFormat": "JSON"}`

Return:
```
{
    "error": {
        "message": "javax.ws.rs.InternalServerErrorException: HTTP 500 Internal Server Error\n\tat org.glassfish.jersey.client.JerseyInvocation.convertToException(JerseyInvocation.java:1098)\n\tat org.glassfish.jersey.client.JerseyInvocation.translate(JerseyInvocation.java:883)\n\tat org.glassfish.jersey.client.JerseyInvocation.lambda$invoke$1(JerseyInvocation.java:767)\n\tat org.glassfish.jersey.internal.Errors.process(Errors.java:316)\n\tat org.glassfish.jersey.internal.Errors.process(Errors.java:298)\n\tat org.glassfish.jersey.internal.Errors.process(Errors.java:229)\n\tat org.glassfish.jersey.process.internal.RequestScope.runInScope(RequestScope.java:414)\n\tat org.glassfish.jersey.client.JerseyInvocation.invoke(JerseyInvocation.java:765)\n\tat org.glassfish.jersey.client.JerseyInvocation$Builder.method(JerseyInvocation.java:456)\n\tat org.glassfish.jersey.client.JerseyInvocation$Builder.post(JerseyInvocation.java:357)\n\tat gov.pnnl.proven.api.exchange.RestExchange.addProvenData(RestExchange.java:164)\n\tat gov.pnnl.proven.api.producer.ProvenSession.sendMessage(ProvenSession.java:147)\n\tat gov.pnnl.proven.api.producer.Producer.sendMessage(Producer.java:343)\n\tat gov.pnnl.proven.api.producer.ProvenProducer.sendMessage(ProvenProducer.java:150)\n\tat gov.pnnl.goss.gridappsd.data.ProvenTimeSeriesDataManagerImpl.query(ProvenTimeSeriesDataManagerImpl.java:128)\n\tat gov.pnnl.goss.gridappsd.data.ProvenTimeSeriesDataManagerImpl.handle(ProvenTimeSeriesDataManagerImpl.java:117)\n\tat gov.pnnl.goss.gridappsd.data.DataManagerImpl.processDataRequest(DataManagerImpl.java:149)\n\tat gov.pnnl.goss.gridappsd.process.ProcessEvent.onMessage(ProcessEvent.java:213)\n\tat pnnl.goss.core.client.GossClient$1.run(GossClient.java:329)\n\tat java.lang.Thread.run(Thread.java:748)\n"
    },
    "responseComplete": true,
    "id": "b3ba3ade-e9d3-4ce1-9dd6-0895b496cffa"
}
```
Expected return: actual data

I suspect this issue is related to asking for too much data at once. I had a filter in place, but it turns out there's a mismatch between the MRIDs in this simulation data and what's in the platform right now.

craig8, 2019-12-02:
More updates on this

I queried the influx db using the following to retrieve known start/end times in the ingested data

```
> select * from simulation where measurement_mrid='_0000ff53-9ecf-442e-84d1-33395ea819a1'  order by time limit 20;
```

Resulting in the following
```
name: simulation
time                angle              hasSimulationMessageType magnitude         measurement_mrid                      simulation_id value
----                -----              ------------------------ ---------         ----------------                      ------------- -----
1563865202000000000 25.832866937648973 OUTPUT                   667.440926646063  _0000ff53-9ecf-442e-84d1-33395ea819a1 1945523778
1563865205000000000 25.832866467987124 OUTPUT                   667.4046967732635 _0000ff53-9ecf-442e-84d1-33395ea819a1 1945523778
1563865208000000000 25.832865440263877 OUTPUT                   667.3364962036346 _0000ff53-9ecf-442e-84d1-33395ea819a1 1945523778
1563865211000000000 25.832865769298174 OUTPUT                   667.3548544300825 _0000ff53-9ecf-442e-84d1-33395ea819a1 1945523778
1563865214000000000 25.83286679336338  OUTPUT                   667.4287212119381 _0000ff53-9ecf-442e-84d1-33395ea819a1 1945523778
1563865217000000000 25.8328670920962   OUTPUT                   667.4350479675752 _0000ff53-9ecf-442e-84d1-33395ea819a1 1945523778
1563865220000000000 25.832866732039616 OUTPUT                   667.4286653148031 _0000ff53-9ecf-442e-84d1-33395ea819a1 1945523778
1563865223000000000 25.832866735451116 OUTPUT                   667.4105400561843 _0000ff53-9ecf-442e-84d1-33395ea819a1 1945523778
1563865226000000000 25.8328678975421   OUTPUT                   667.5038161306446 _0000ff53-9ecf-442e-84d1-33395ea819a1 1945523778
1563865229000000000 25.83286669218147  OUTPUT                   667.3874726907572 _0000ff53-9ecf-442e-84d1-33395ea819a1 1945523778
1563865232000000000 25.83286590087007  OUTPUT                   667.3369509884399 _0000ff53-9ecf-442e-84d1-33395ea819a1 1945523778
1563865235000000000 25.832865819617552 OUTPUT                   667.3566501317995 _0000ff53-9ecf-442e-84d1-33395ea819a1 1945523778
1563865238000000000 25.8328655689241   OUTPUT                   667.3282842251425 _0000ff53-9ecf-442e-84d1-33395ea819a1 1945523778
1563865241000000000 25.83286566929482  OUTPUT                   667.3152169036254 _0000ff53-9ecf-442e-84d1-33395ea819a1 1945523778
1563865244000000000 25.83286578563117  OUTPUT                   667.3409055910998 _0000ff53-9ecf-442e-84d1-33395ea819a1 1945523778
1563865247000000000 25.832865832548332 OUTPUT                   667.3598466264386 _0000ff53-9ecf-442e-84d1-33395ea819a1 1945523778
1563865250000000000 25.83286573722149  OUTPUT                   667.3057946753657 _0000ff53-9ecf-442e-84d1-33395ea819a1 1945523778
1563865253000000000 25.83286600744314  OUTPUT                   667.3665504323614 _0000ff53-9ecf-442e-84d1-33395ea819a1 1945523778
1563865256000000000 25.832865935070924 OUTPUT                   667.366274489814  _0000ff53-9ecf-442e-84d1-33395ea819a1 1945523778
1563865259000000000 25.83286613572007  OUTPUT                   667.3729317778908 _0000ff53-9ecf-442e-84d1-33395ea819a1 1945523778
```

From the platform as @blthayer said I get the error posted above.
```
{"queryMeasurement":"simulation",
"queryFilter":{"simulation_id": "1945523778","startTime":"1563865202",
                        "endTime":"1563865259"},
"responseFormat":"JSON"}
```

In the influxdb  log I see the query

```
[httpd] 172.18.0.4 - root [02/Dec/2019:21:15:03 +0000] "GET /query?db=proven&epoch=s&p=%5BREDACTED%5D&q=select+%2A+from++%22simulation%22++where++simulation_id+%3D%271945523778%27++and+time+%3E%3D+1563865202000000000+++and+time+%3C%3D+1563865259000000000++&u=root HTTP/1.1" 200 85 "-" "okhttp/3.9.1" ce723086-1548-11ea-8011-0242ac120003 10181761
```

ericstephan, 2019-12-03:
I did some tests in the query you provided.  I waited 23 minutes for the results to return before aborting the query.  

I did some additional tests using the count function below. I ran this query 3 times, the first time it ran 2.5 minutes and the 2nd and 3rd time it took about a minute.  

Below you see count_angle and count_magnitude yielding 476 million records and the count_value yielding over 6 million records. for the simulation.  

> select count(*) from simulation where simulation_id = '1945523778';
name: simulation
time count_angle count_magnitude count_value
---- ----------- --------------- -----------
0    476601510   476601510       6146194

We have some options that I've discussed with @telsethag that Proven collects some summary information that can be provided to the application about a given simulation or more generically a given measurement.  Summary Information may include:
* total number of timesteps for a given simulation.
* min/max timestep for a given simulation.

I could also check into a possible automatic abort query if the query is taking too long due to the size of the request.

Eric

poorva1209, 2019-12-03:
@ericstephan Try adding a startTime and endTime to the query. The first problem is the 500 error - are you able to reproduce it? Once that is resolved then we can look into timing.

ericstephan, 2019-12-04:
Can you run this query?  I am testing it against Craig's database with no
problems.

{"queryType": "time-series", "queryMeasurement":
"simulation","queryFilter": {"simulation_id":
"1945523778","startTime":1563865202000000000,
"endTime":1563865202000000000},"responseFormat": "JSON"}

ericstephan, 2019-12-04:
When I attempted query from above, I realized that the time numbers are
quoted, which I thought was against the convention we were using to
represent numbers.  Remove the quotes around the startTime and endTime
values and the query should work.

Incorrect:

{"queryMeasurement":"simulation",
"queryFilter":{"simulation_id": "1945523778","startTime":"1563865202",
                        "endTime":"1563865259"},
"responseFormat":"JSON"}

Correct:

{"queryMeasurement":"simulation",
"queryFilter":{"simulation_id": "1945523778","startTime":1563865202,
                        "endTime":1563865259},
"responseFormat":"JSON"}

poorva1209, 2019-12-12:
Reworded the issue as this is not related to Proven. This should be resolved once @craig8  updated the preloaded data.

closed 2020-01-06 by craig8