GRIDAPPSD/gridappsd-forum #62: fncs fatal error when starting simulation

AAndersn, 2019-12-10:
Today, we have ran into a new error on both the EIOC pyvvo and my local machine. When starting a simulation, the fncs bridge runs into a fatal error:
 
[image]

On some simulation runs the simulation log stops there. On others, the simulation continues running, but no data is passed to the viz.

poorva1209, 2019-12-10:
@craig8 I see you merged changes in fncs goss bridge - Can that do this ?

AAndersn, 2019-12-11:
Update: Today the error is not present on new simulation runs. However, if the simulation is stopped using the square stop button and then restarted, the fncs goss brdige still encounters the fatal error:

[image]

Log message:

```

2019-12-11 21:27:22,449 Thread-203 ERROR [gov.pnnl.goss.gridappsd.log.LogManagerImpl] - 1576099642448|SimulationProcess|1105872583|ERROR|system|ERROR
FNCS Initialization Failed

2019-12-11 21:27:22,449 Thread-46 ERROR [gov.pnnl.goss.gridappsd.log.LogManagerImpl] - 1576099642448|SimulationProcess|1105872583|ERROR|system|ERROR
FNCS Initialization Failed
```

AAndersn, 2020-09-14:
Latest develop tag no longer has this problem. Simulations can be stopped, paused, restarted with no errors.

Is this due to switch to HELICS or was this bug fixed?

poorva1209, 2020-09-14:
This was due to some changes in the GridLAB-D latest release. This has been resolved. You can close this issue.

closed 2020-09-14 by AAndersn