GRIDAPPSD/gridappsd-forum #65: PV and inverter on secondary clarification.

Jeffrey-Simpson, 2020-04-02:
PV and inverter on secondary clarification.
When setting the p and q of inverters that are on the secondary side in the platform it looks like the values set are spread equally across both s1 and s2 instead of individually.  Is this the expected behavior?

For the following command  the object  "_EEE9E485-740A-4002-A18A-02EE8CC38236" q attribute should be : 0.0 and object  "_187D386A-8A3F-47DE-9532-850843328878" should be  7137.61, but they both end up being about 3500 (3536.2570359999995, 3548.684329). 

```json
{
  "command": "update",
  "input": {
    "simulation_id": 647716039,
    "message": {
      "timestamp": 1585596795,
      "difference_mrid": "8d8cde7a-e212-4795-aac0-d7da6d3d02fe",
      "reverse_differences": [
        {
          "object": "_68F0D6C3-D3D0-446E-89C8-BAAA0E1BC1B4",
          "attribute": "PowerElectronicsConnection.p",
          "value": 3524.1288082608266
        },
        {
          "object": "_68F0D6C3-D3D0-446E-89C8-BAAA0E1BC1B4",
          "attribute": "PowerElectronicsConnection.q",
          "value": -179.91393051258373
        },
        {
          "object": "_9517E89E-7F80-4769-93CB-FDC94D8ABA1E",
          "attribute": "PowerElectronicsConnection.p",
          "value": 3515.8791535907308
        },
        {
          "object": "_9517E89E-7F80-4769-93CB-FDC94D8ABA1E",
          "attribute": "PowerElectronicsConnection.q",
          "value": 179.91372855908824
        },
        {
          "object": "_EEE9E485-740A-4002-A18A-02EE8CC38236",
          "attribute": "PowerElectronicsConnection.p",
          "value": 4441.180523359725
        },
        {
          "object": "_EEE9E485-740A-4002-A18A-02EE8CC38236",
          "attribute": "PowerElectronicsConnection.q",
          "value": -179.9478822656219
        },
        {
          "object": "_187D386A-8A3F-47DE-9532-850843328878",
          "attribute": "PowerElectronicsConnection.p",
          "value": 4428.8231564863545
        },
        {
          "object": "_187D386A-8A3F-47DE-9532-850843328878",
          "attribute": "PowerElectronicsConnection.q",
          "value": 179.9477368459201
        }
      ],
      "forward_differences": [
        {
          "object": "_68F0D6C3-D3D0-446E-89C8-BAAA0E1BC1B4",
          "attribute": "PowerElectronicsConnection.p",
          "value": 3524.12
        },
        {
          "object": "_68F0D6C3-D3D0-446E-89C8-BAAA0E1BC1B4",
          "attribute": "PowerElectronicsConnection.q",
          "value": -1127.18
        },
        {
          "object": "_9517E89E-7F80-4769-93CB-FDC94D8ABA1E",
          "attribute": "PowerElectronicsConnection.p",
          "value": 3515.88
        },
        {
          "object": "_9517E89E-7F80-4769-93CB-FDC94D8ABA1E",
          "attribute": "PowerElectronicsConnection.q",
          "value": 3694.4
        },
        {
          "object": "_EEE9E485-740A-4002-A18A-02EE8CC38236",
          "attribute": "PowerElectronicsConnection.p",
          "value": 4441.18
        },
        {
          "object": "_EEE9E485-740A-4002-A18A-02EE8CC38236",
          "attribute": "PowerElectronicsConnection.q",
          "value": 0.0
        },
        {
          "object": "_187D386A-8A3F-47DE-9532-850843328878",
          "attribute": "PowerElectronicsConnection.p",
          "value": 4428.82
        },
        {
          "object": "_187D386A-8A3F-47DE-9532-850843328878",
          "attribute": "PowerElectronicsConnection.q",
          "value": 7137.61
        }
      ]
    }
  }
}

```

```
+--------------+------------------------------------------------------+------------------------------------------------------+-----------------------------------------------------+----------------------------------------------------+
|              | pv_47                                                | pv_48                                                | pv_1152                                             | pv_1153                                            |
|--------------+------------------------------------------------------+------------------------------------------------------+-----------------------------------------------------+----------------------------------------------------+
| size         | 3.7                                                  | 5.1                                                  | 2.7                                                 | 8.4                                                |
| fmrid        | _7D20FA43-8E88-4986-B64E-38F282E2B2BA                | _5B92B9CB-0B9D-4A40-BB13-CEE5FC1E63F3                | _B81F6356-B8E4-49FB-9D05-CFFA7C246AD9               | _BF1C4AEA-9235-4DBA-A45C-E7C15AD6BE5D              |
| id           | _68F0D6C3-D3D0-446E-89C8-BAAA0E1BC1B4                | _9517E89E-7F80-4769-93CB-FDC94D8ABA1E                | _EEE9E485-740A-4002-A18A-02EE8CC38236               | _187D386A-8A3F-47DE-9532-850843328878              |
| name         | pv_47                                                | pv_48                                                | pv_1152                                             | pv_1153                                            |
| busname      | SX2000711B.1                                         | SX2000711B.2                                         | SX2748143A.1                                        | SX2748143A.2                                       |
| meas_mrid    | _7675fb01-83ed-4d51-9c9f-35a1d3f5de24                | _19fb6c72-99f7-4903-9a46-e2dbcb75e8cd                | _f88ea336-85ce-4768-94f6-1c331b364a90               | _ec77c98c-0334-447a-9a37-5c69e6b0024f              |
| busphase     | ['1']                                                | ['2']                                                | ['1']                                               | ['2']                                              |
| numPhase     | 1                                                    | 1                                                    | 1                                                   | 1                                                  |
| last_time    | {'p': -3523.8130780000006, 'q': -5.431617999999665}  | {'p': -3516.186922, 'q': 5.4316180000001335}         | {'p': -4440.743877, 'q': -4.1061389999991516}       | {'p': -4429.256123, 'q': 4.106139000000979}        |
| current_time | {'p': -3524.2128550000007, 'q': -1349.5766119999998} | {'p': -3515.7871450000002, 'q': -1345.2617079999993} | {'p': -3569.374578000001, 'q': -3536.2570359999995} | {'p': -3545.073769, 'q': -3548.684329}             |
| polar        | {'p': 3773.7823571325166, 'q': -159.04595201164767}  | {'p': 3764.3709052055865, 'q': -159.06142147516093}  | {'p': 5024.4948704056, 'q': -135.26703912448937}    | {'p': 5016.0451049150315, 'q': -134.9708377662507} |
| index        | 559                                                  | 560                                                  | 593                                                 | 594                                                |
+--------------+------------------------------------------------------+------------------------------------------------------+-----------------------------------------------------+----------------------------------------------------+
```

temcdrm, 2020-04-02:
Yes, this is expected. Those residential inverters are generally 240-volt so we split the generation equally between s1 and s2.

Jeffrey-Simpson, 2020-04-02:
Okay thanks Tom. I will talk to Fei about how to handle this best.

closed 2022-03-22 by afisher1