GRIDAPPSD/gridappsd-forum #66: Not getting the develop tag recent functionalities

shpoudel, 2020-04-04:
@poorva1209: I am not getting the following two new functionalities in the develop tag as you mentioned:

1. The switch **open** and **close** status are still showing as 0/1 in the alarm instead of Open and Close.
2. The outage section is not changing the color to red as we saw in the EIOC system.

I did ./stop.sh -c, git pull, and ./run.sh -t develop. Please let me know if I am missing something.

I am waiting to verify the color mismatch such that I can create an event file for viz team to look into.

Thank you.

poorva1209, 2020-04-06:
@shpoudel Can you try running 
`docker-compose pull`

shpoudel, 2020-04-06:
Thank Poorva. It worked. I will send an email soon with the event file and detail explanation for the viz team to look into the small issue we had last time.

closed 2020-04-07 by poorva1209