GRIDAPPSD/gridappsd-forum #67: Browser slowing down with new develop branch

shpoudel, 2020-04-17:
@poorva1209 @bibiraju 

I am having an issue of **slowing down of the browser** and see that the simulation status in the viz seems to be lagged significantly. With this, the application performance and the corresponding simulation in the platform are not synchronized well.  I have attached an image in this post as well.

P.S: I have checked the real-time button in the simulation configuration.

Please let me know if you can think of any possible solution.

[image]

nhuyvan, 2020-04-27:
@bibiraju You should test my commit for this issue inside your VM instead of running the simulation on your Mac for a more accurate result

poorva1209, 2020-04-30:
@shpoudel Shiva can you test the latest viz develop tag to see if your browser still shows you slowed down message. If so, we will re-open this issue.

shpoudel, 2020-04-30:
@poorva1209 I was actually checking this while you were commenting on this issue here. And yes, I am still seeing the browser is slowing down with the same message.

shpoudel, 2020-04-30:
@poorva1209 Just to make sure about pulling the latest develop tag. I did the following steps recently. 
./stop.sh -c 
docker-compose pull (Error shown below for this)
./run.sh -t develop

I am getting following errors while doing **docker-compose** pull command:

WARNING: The GRIDAPPSD_TAG variable is not set. Defaulting to a blank string.
Pulling redis (redis:3.2.11-alpine)...
ERROR: Get https://registry-1.docker.io/v2/: net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)

So, I believe I am not running the latest develop tag, right? How can I get rid of that docker-compose pull command error?

poorva1209, 2020-04-30:
@tonya1 I thought this was just something with my VM but Shiva is also getting this error above. Can you try stop.sh -c and then run.sh -t develop. For me the containers and specially the MySQL container is not removing at all.

shpoudel, 2020-04-30:
I did /stop.sh -c and ./run.sh -t develop but the issue for browser is still there. 

Based on my experience, when pulling a latest tag, I usually get several pulling commands in the terminal if it is actually pulling the latest changes. But, now it just connects to the gridappsd container soon after doing run.sh -t develop. So, the latest develop tag is not pulled and browser issue still remains there in my local machine.

tonya1, 2020-05-04:
@shpoudel I have tried to reproduce this on my vm without success.  Would you wan to have a Teams meeting and share your screen?

poorva1209, 2020-05-05:
I tried to reproduce this error with latest develop tag and couldn't. I know I was getting this last week. @shpoudel What time would work for you for a quick a call today?

shpoudel, 2020-05-05:
@poorva1209 @tonya1 What about this afternoon? Anytime after 1:00 PM.

poorva1209, 2020-05-05:
@tonya1 Can you setup something with Shiva to debug this as I won't be working after 1pm?

tonya1, 2020-05-06:
@poorva1209 @shpoudel I was able to replicate the browser slowdown when I changed my VM to 8GB of memory with 4CPU using Firefox to run the 9500 model for 1200 seconds.  Firefox stopped at 5 seconds with a prompt to wait. 

I was able to run chrome without the error but I encountered a lot of errors trying to start the platform with the limited resources.  I had to kill and start the ./run-gridappsd.sh multiple times to get things running correctly.

I tried going to 6 cpu or 10 GB of memory and still saw the firefox wait error.

poorva1209, 2020-05-07:
Thanks @tonya1. Based on this I am moving the issue from review to in-progress. @nhuyvan @bibiraju keep on looking for issues that can slow down the browser. I also have some ideas, I will talk to you guys about it in work meeting.

closed 2020-09-29 by poorva1209