GRIDAPPSD/gridappsd-forum #68: Only one measurement object's output stored in database after simulation

blthayer, 2020-05-11:
When running a simulation with the 9500 node model, it would seem measurements from a single measurement object are being stored in the output.

Platform version: `v2020.04.0`

Steps to reproduce:
1. Run a simulation with the following configuration (9500 node for 20 seconds):
```json
{
  "power_system_config": {
    "GeographicalRegion_name": "_73C512BD-7249-4F50-50DA-D93849B89C43",
    "SubGeographicalRegion_name": "_A1170111-942A-6ABD-D325-C64886DC4D7D",
    "Line_name": "_AAE94E4A-2465-6F5E-37B1-3E72183A4E44"
  },
  "application_config": {
    "applications": []
  },
  "simulation_config": {
    "start_time": "1358121600",
    "duration": "20",
    "simulator": "GridLAB-D",
    "timestep_frequency": "1000",
    "timestep_increment": "1000",
    "run_realtime": false,
    "simulation_name": "ieee8500",
    "power_flow_solver_method": "NR",
    "model_creation_config": {
      "load_scaling_factor": "1",
      "schedule_name": "ieeezipload",
      "z_fraction": "0",
      "i_fraction": "1",
      "p_fraction": "0",
      "randomize_zipload_fractions": false,
      "use_houses": false
    }
  },
  "test_config": {
    "events": [],
    "appId": ""
  },
  "service_configs": []
}
```
2. Capture the simulation ID for use in the next step. I subscribed to simulation output for two measurement MRIDs, `_e78cd2b1-7c3a-432f-948a-cce98fd9e737` and `_fabafc17-3bb8-447d-87cd-a45e11868668`. I then simply print the corresponding measurements and get the following:
```
Time: 2013-01-14 00:00:02+00:00
Meas: [{'measurement_mrid': '_e78cd2b1-7c3a-432f-948a-cce98fd9e737', 'magnitude': 116.69820070682674, 'angle': -150.99733148057783}, {'measurement_mrid': '_fabafc17-3bb8-447d-87cd-a45e11868668', 'magnitude': 118.67570426324917, 'angle': 88.13061845760909}]
Time: 2013-01-14 00:00:05+00:00
Meas: [{'measurement_mrid': '_e78cd2b1-7c3a-432f-948a-cce98fd9e737', 'magnitude': 116.69821752371271, 'angle': -150.99733886467484}, {'measurement_mrid': '_fabafc17-3bb8-447d-87cd-a45e11868668', 'magnitude': 118.67572503551308, 'angle': 88.1306071917342}]
Time: 2013-01-14 00:00:08+00:00
Meas: [{'measurement_mrid': '_e78cd2b1-7c3a-432f-948a-cce98fd9e737', 'magnitude': 116.69821752371271, 'angle': -150.99733886467484}, {'measurement_mrid': '_fabafc17-3bb8-447d-87cd-a45e11868668', 'magnitude': 118.67572503551308, 'angle': 88.1306071917342}]
Time: 2013-01-14 00:00:11+00:00
Meas: [{'measurement_mrid': '_e78cd2b1-7c3a-432f-948a-cce98fd9e737', 'magnitude': 116.69821752371271, 'angle': -150.99733886467484}, {'measurement_mrid': '_fabafc17-3bb8-447d-87cd-a45e11868668', 'magnitude': 118.67572503551308, 'angle': 88.1306071917342}]
Time: 2013-01-14 00:00:14+00:00
Meas: [{'measurement_mrid': '_e78cd2b1-7c3a-432f-948a-cce98fd9e737', 'magnitude': 116.69821752371271, 'angle': -150.99733886467484}, {'measurement_mrid': '_fabafc17-3bb8-447d-87cd-a45e11868668', 'magnitude': 118.67572503551308, 'angle': 88.1306071917342}]
Time: 2013-01-14 00:00:17+00:00
Meas: [{'measurement_mrid': '_e78cd2b1-7c3a-432f-948a-cce98fd9e737', 'magnitude': 116.69821752371271, 'angle': -150.99733886467484}, {'measurement_mrid': '_fabafc17-3bb8-447d-87cd-a45e11868668', 'magnitude': 118.67572603498087, 'angle': 88.13060720748352}]
```
This proves the simulator is emitting output from more than just one measurement object.

3. After the simulation completes, query for simulation output (topic: `/queue/goss.gridappsd.process.request.data.timeseries`) with the following (update `simulation_id` with that obtained earlier):
```json
{
  "queryMeasurement": "simulation",
  "queryFilter": {
    "simulation_id": "358006346",
    "hasSimulationMessageType": "OUTPUT",
    "measurement_mrid": "_e78cd2b1-7c3a-432f-948a-cce98fd9e737"
  },
  "responseFormat": "JSON"
}
```
The following output is given:
```json
{
  "data": {},
  "responseComplete": true,
  "id": "774445831"
}
```
This proves the output associated with the measurement object with mrid `_e78cd2b1-7c3a-432f-948a-cce98fd9e737` is not in the database.

4. Repeat the following query, but completely remove the `measurement_mrid` field from the `queryFilter`. The following output is obtained:
```json
{
  "data": [
    {
      "hasSimulationMessageType": "OUTPUT",
      "measurement_mrid": "_fabafc17-3bb8-447d-87cd-a45e11868668",
      "angle": 88.13061845760909,
      "magnitude": 118.67570426324917,
      "simulation_id": "358006346",
      "time": 1358121602,
      "value": 1.0
    },
    {
      "hasSimulationMessageType": "OUTPUT",
      "measurement_mrid": "_fabafc17-3bb8-447d-87cd-a45e11868668",
      "angle": 88.1306071917342,
      "magnitude": 118.67572503551308,
      "simulation_id": "358006346",
      "time": 1358121605,
      "value": 1.0
    },
    {
      "hasSimulationMessageType": "OUTPUT",
      "measurement_mrid": "_fabafc17-3bb8-447d-87cd-a45e11868668",
      "angle": 88.1306071917342,
      "magnitude": 118.67572503551308,
      "simulation_id": "358006346",
      "time": 1358121608,
      "value": 1.0
    },
    {
      "hasSimulationMessageType": "OUTPUT",
      "measurement_mrid": "_fabafc17-3bb8-447d-87cd-a45e11868668",
      "angle": 88.1306071917342,
      "magnitude": 118.67572503551308,
      "simulation_id": "358006346",
      "time": 1358121611,
      "value": 1.0
    },
    {
      "hasSimulationMessageType": "OUTPUT",
      "measurement_mrid": "_fabafc17-3bb8-447d-87cd-a45e11868668",
      "angle": 88.1306071917342,
      "magnitude": 118.67572503551308,
      "simulation_id": "358006346",
      "time": 1358121614,
      "value": 1.0
    },
    {
      "hasSimulationMessageType": "OUTPUT",
      "measurement_mrid": "_fabafc17-3bb8-447d-87cd-a45e11868668",
      "angle": 88.13060720748352,
      "magnitude": 118.67572603498087,
      "simulation_id": "358006346",
      "time": 1358121617,
      "value": 1.0
    }
  ],
  "responseComplete": true,
  "id": "740398902"
}
```
Notice that even without an MRID filter, the only measurements that come back are from `_fabafc17-3bb8-447d-87cd-a45e11868668`.

poorva1209, 2020-05-12:
@ericstephan Looks like influx is storing simulation output for only one measurement mrid.  Please have a look at this today. Check the gridappsd-sensor-simulator too.

poorva1209, 2020-05-13:
I tested both simulation and sensor service output in influx, it has multiple measuremen_mrids now. closing this issue.

closed 2020-05-13 by poorva1209