GRIDAPPSD/gridappsd-forum #70: Influxdb/GOSS bombs out for modest query

blthayer, 2020-05-12:
Platform version: `v2020.04.0`
Influxdb version: `historical`

Summary: Asking for all historical data for only 1 minute causes an error.

To reproduce:
1. In the `docker-compose` file, modify the `influxdb` tag to be `historical`
2. Start platform as normal: `./run.sh -t v2020.04.0`
3. Send the following query on the 
```json
{
  "queryMeasurement": "gridappsd-sensor-simulator",
  "queryFilter": {
    "simulation_id": "890162203",
    "starttime": "1373727602",
    "endtime": "1373727662"
  },
  "responseFormat": "JSON"
}
```
Result: The query times out and the following gets dumped to the console:
```
LOOKING FOR CONVERTER PROVEN_GRIDAPPSD-SENSOR-SIMULATOR:JSON  IN   {PROVEN_LOADPROFILE:GRIDLABD_LOAD_SCHEDULE=gov.pnnl.goss.gridappsd.data.conversion.ProvenLoadScheduleToGridlabdLoadScheduleConverter@6ba5626c, PROVEN_WEATHER:GRIDLABD_WEATHER=gov.pnnl.goss.gridappsd.data.conversion.ProvenWeatherToGridlabdWeatherConverter@3dfc0430}
java.io.IOException: Connection reset by peer
2020-05-12 16:01:02,060 Thread-41 ERROR [pnnl.goss.core.client.GossClient] - publish error
javax.jms.JMSException: Connection reset by peer
	at org.fusesource.stomp.jms.StompJmsExceptionSupport.create(StompJmsExceptionSupport.java:59)
	at org.fusesource.stomp.jms.StompChannel.sendMessage(StompChannel.java:203)
	at org.fusesource.stomp.jms.StompJmsSession.send(StompJmsSession.java:708)
	at org.fusesource.stomp.jms.StompJmsSession.send(StompJmsSession.java:654)
	at org.fusesource.stomp.jms.StompJmsMessageProducer.send(StompJmsMessageProducer.java:157)
	at org.fusesource.stomp.jms.StompJmsMessageProducer.send(StompJmsMessageProducer.java:124)
	at pnnl.goss.core.client.DefaultClientPublisher.publish(DefaultClientPublisher.java:136)
	at pnnl.goss.core.client.GossClient.publish(GossClient.java:438)
	at gov.pnnl.goss.gridappsd.process.ProcessEvent.sendData(ProcessEvent.java:307)
	at gov.pnnl.goss.gridappsd.process.ProcessEvent.onMessage(ProcessEvent.java:215)
	at pnnl.goss.core.client.GossClient$1.run(GossClient.java:329)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: Connection reset by peer
	at sun.nio.ch.FileDispatcherImpl.write0(Native Method)
	at sun.nio.ch.SocketDispatcher.write(SocketDispatcher.java:47)
	at sun.nio.ch.IOUtil.writeFromNativeBuffer(IOUtil.java:93)
	at sun.nio.ch.IOUtil.write(IOUtil.java:65)
	at sun.nio.ch.SocketChannelImpl.write(SocketChannelImpl.java:468)
	at org.fusesource.hawtdispatch.transport.AbstractProtocolCodec.flush(AbstractProtocolCodec.java:198)
	at org.fusesource.hawtdispatch.transport.TcpTransport.flush(TcpTransport.java:678)
	at org.fusesource.hawtdispatch.transport.TcpTransport$5.run(TcpTransport.java:579)
	at org.fusesource.hawtdispatch.internal.HawtCustomDispatchSource$1.run(HawtCustomDispatchSource.java:127)
	at org.fusesource.hawtdispatch.internal.SerialDispatchQueue.run(SerialDispatchQueue.java:100)
	at org.fusesource.hawtdispatch.internal.pool.SimpleThread.run(SimpleThread.java:77)
	at sun.nio.ch.FileDispatcherImpl.write0(Native Method)
	at sun.nio.ch.SocketDispatcher.write(SocketDispatcher.java:47)
	at sun.nio.ch.IOUtil.writeFromNativeBuffer(IOUtil.java:93)
	at sun.nio.ch.IOUtil.write(IOUtil.java:65)
	at sun.nio.ch.SocketChannelImpl.write(SocketChannelImpl.java:468)
	at org.fusesource.hawtdispatch.transport.AbstractProtocolCodec.flush(AbstractProtocolCodec.java:198)
	at org.fusesource.hawtdispatch.transport.TcpTransport.flush(TcpTransport.java:678)
	at org.fusesource.hawtdispatch.transport.TcpTransport$5.run(TcpTransport.java:579)
	at org.fusesource.hawtdispatch.internal.HawtCustomDispatchSource$1.run(HawtCustomDispatchSource.java:127)
	at org.fusesource.hawtdispatch.internal.SerialDispatchQueue.run(SerialDispatchQueue.java:100)
	at org.fusesource.hawtdispatch.internal.pool.SimpleThread.run(SimpleThread.java:77)
```
Back of the napkin calculation:
Say the database is 70GB (though I think it's actually more like 30GB). We're grabbing 1 minute's worth of data, and there's just over 6 days worth of data.
`70 GB * 1 minute / (6 days * 24 hr/day * 60 min/hr) = 0.008GB`

It seems odd that 8MB causes the thing to bomb.

blthayer, 2020-05-12:
Note that after this error the platform must be restarted for future queries to work. Not fun.

blthayer, 2020-05-12:
Important note: when I try to query for the whole "week" for only four measurement MRIDs (necessary to get data for a single customer meter), I encounter the same error. As such, I'm bumping this to app-critical. Here's the query:
```json
{
  "queryMeasurement": "gridappsd-sensor-simulator",
  "queryFilter": {
    "simulation_id": "890162203",
    "starttime": "1373727602",
    "endtime": "1374271943",
    "measurement_mrid": [
      "_e78cd2b1-7c3a-432f-948a-cce98fd9e737",
      "_5a52bc83-d844-4ef4-8de5-d64ed6222192",
      "_90defe9f-f34a-47f2-b49b-277bed98a12f",
      "_7422078b-ef84-498b-ad74-de8a62c171ad"
    ]
  },
  "responseFormat": "JSON"
}
```
Note the above query does seem to work when edited to pull for only a single MRID.

poorva1209, 2020-05-12:
I tested by running a simulation for 2 minutes and then queried the sensor simulator data for 1 minutes and then 2 minutes. I got the data back both the times. Now. I will try with historical influxdb container.

poorva1209, 2020-12-01:
I queried historical data from influxdb for 1 nad 2 minutes and this query is working now. Closing this issue.

closed 2020-12-01 by poorva1209