GRIDAPPSD/gridappsd-forum #72: Common new developer issues and training needs

Jeffrey-Simpson, 2020-08-24:
I am creating this as a place to note training needs for developers. This should be filled out and added to ReadTheDocs

### Training needs for platform

These questions are typically coming from people who are used to talking to directly to powerflow simulator from python or matlab so the publish/subscribe is a new world to them.
The lack of 

Debug app in container. 
How the register app work and what is going on there.
How to log
How to get container name
How to use the docker logs --follow or --tail command
How to use docker cp

Difference messages

How publish subscribe works
How publish subscribe with the gridappsd-python
    The on_message function with topic. This especially is very confusing to developers who are used to reading files or getting values from OpenDSS

How to get the y-matrix and node names.

How to get model_dict.json

How to get specific measurements:
    Node Voltages
    Load Powers

shpoudel, 2020-08-25:
I agree with Jeff. Few more to add on:

1. What are different Ids associated with a component and which one to use to control a specific component?
2. More details and sample queries for several components would be useful for those with limited knowledge of SPARQL (me).
3. Sample code on how different APIs work.

poorva1209, 2022-11-02:
@AAndersn Can we put this in backlog or close if it is already done?

closed 2022-11-03 by poorva1209