GRIDAPPSD/gridappsd-forum #74: Getting Y matrix in real-time reflecting the topology changes

cmmooari, 2020-09-03:
Dear Gridapps-D developers,

Is it possible to get a "dynamic" Y matrix at every simulation interval during the simulation process, to reflect the changes on system topology (due to transformer tap changes and switch operations)?

Thank you very much!

poorva1209, 2020-09-03:
@areiman Can you provide an answer to this issue? Feel free to assign to Tom and Andy if needed.

areiman, 2020-12-10:
The "dynamic" Y matrix cannot be obtained directly at this time. 

However, measurements of switch and regulator positions are included in the simulation output and - along with the locations of these devices - can be used to dynamically update the Y matrix at each simulation time step.

A dynamic Ybus service is being considered for future implementation.

AAndersn, 2022-05-12:
The dynamic Y-bus service is now available in the ~develop release tag of the GridAPPS-D platform. That service publishes a new Y-bus matrix for all topology and transformer tap changes

areiman, 2022-11-03:
Closing - see dynamic Y-bus service.

closed 2022-11-03 by areiman