GRIDAPPSD/gridappsd-forum #76: Loads with zero power / outaged when starting simulation

AAndersn, 2020-09-22:
When starting a new simulation, random sets of lines and customer loads appear red (outaged?) in viz. This bug appears inconsistently between simulations and goes away after 30 - 45 seconds of simulation time. I have not noticed any correlations with any services or apps running.

[image]

This may be a problem with either the loads or the measurements rather than the viz because when plotted, there is no power flow on the "red" nodes:

[image]

It could be a power flow solver issue, because the immediate upstream line supplying that customer shows _some_ power flow, but much less than the nominal load power

[image]

After some time, the load values jump from zero to nominal, and all the equipment appears energized in the viz as well:

[image]

poorva1209, 2020-09-23:
@bibiraju Can you check what @AAndersn is experiencing here with simulation display on viz. 
If the values seem correct for displaying lines as red then Talk to @afisher1 to see if this a bug or should be like this.

nhuyvan, 2020-09-25:
Please see the red line segment right next to the transformer in the following picture, that line is highlighted red because its magnitude is 0, so it's marked as having no power, any magnitude value between -0.3 and 0.3 is marked as having no power, I'm also including the measurement data that was used for this line segment for your reference, that highlighted line segment has name **tpx337636c0**, I also observed magnitudes of 0 multiple times for **tpx337636c0**, and everytime that happened, this line was highlighted red, the same for the other lines

(The follow JSON is an aggregation of the data in "measurements" part of model dictionary and simulation output)

```json
{
    "name": "ACLineSegment_tpx337636c0_Power",
    "type": "VA",
    "magnitude": 0,
    "angle": 0,
    "mRID": "_a5a9fee7-0949-43a8-8f91-9fdf0ff3321d",
    "phases": "s1",
    "conductingEquipmentName": "tpx337636c0",
    "conductingEquipmentType": "ACLineSegment",
    "connectivityNode": "x2841619c",
    "conductingEquipmentMRID": "_FDF2BEEC-7C9B-427D-AA87-8F972F9618E4"
}
```

[image]

bibiraju, 2020-09-25:
@afisher1 Please see comments from Nhuy above. Viz is showing red lines as the value is 0. Let us know if something need to be changed in Viz.

AAndersn, 2020-09-25:
@bibiraju  @nhuyvan , I believe that everything in the viz is correct. The red line also corresponds to stripchart showing a value of 0 for the power plot. 

@afisher1 I think it is a problem with GridLab-D not setting up the loads right during the first several timesteps. After a minute, the simulation sets the loads to the correct value. 

In the third image I posted before, the stripchart shows the power consumption is zero for load 337715A0 (attached to node SX3254204 (low-side) and L3254204 (high-side). The only flow on line LN5744330-1 is the transformer magnetizing losses.

poorva1209, 2020-09-25:
@AAndersn @afisher1 Does this mean I have to wait for this to be resolved in GridLAB-D before I can generate historical data? Or this does not affect the data needed by VVO?

AAndersn, 2020-09-25:
@poorva1209 Let's go ahead and generate the mini data set and see if this causes a problem. My hope is that it won't be an issue because the bug only lasts for 1 minute, and we are doing a 15 minute average

shpoudel, 2020-09-29:
Unfortunately, I observed this issue for a longer period. I ran the simulation for 200 seconds and the power was zero for the entire simulation time with the triplex lines going red. 

[image]