GRIDAPPSD/gridappsd-forum #78: Default plots in visualization

shpoudel, 2020-09-30:
It seems like Voltage (min, max, and average) and Load Demand plots are added into visualization by default when the platform starts. However, the plot for the voltage looks a little off to me. The min and max values are somewhere between 0.75 p.u. to 0.59 p.u. I don't think the feeder voltage lies in between this range. Also, this range is not consistent with the total voltage violations count.

nhuyvan, 2020-10-01:
We asked Tom how to calculate min/average/max voltages , and this is basically what Tom said to do to find those 3 quantities: Get the percentage of nominal voltage for each ConnectivityNode by taking the average of Alo and Blo in the limits.json file, then divide each simulation output measurement's magnitude for measurement type of "PNV" by this average, then find the min/average/max voltages after applying the divisions. So it could be that this way of calculating the min/average/max isn't correct?

shpoudel, 2020-10-01:
I am not sure about Alo and Blo in the limits.json file. The Y-axis in the plot is a p.u., correct? Can you show me how the limits.json file looks like? Maybe Tom showed the easy way to calculate it and I might have understood it incorrectly.

shpoudel, 2020-10-01:
Never mind, I found the limit.json file. It is an average of alo and ahi that you should take not the average of Alo and Blo. Also, for some nodes, the sqrt(3) term might play a role. If the nominal voltage for the node is 208, the corresponding measurements for such a node are around 120. So, you should divide 208 by sqrt(3) before calculating the percentage of nominal voltage. @temcdrm , please correct me if I am wrong.

bibiraju, 2020-10-08:
@temcdrm We are taking the average of Alo and Blo. Is this correct? Please see comments from Shiva. We will work on fixing the sqrt(3) term as suggested above.

temcdrm, 2020-10-13:
Yes, you should take the average of Alo and Ahi.

closed 2020-11-03 by craig8