GRIDAPPSD/gridappsd-forum #79: No measurement/zero voltages for some spilt phase nodes

Jeffrey-Simpson, 2020-10-01:
I am getting zero voltages for these nodes X2748140A.2 , X3728040A.2, and X3160861C.2 in the 9500 model.
The reason for this is there is no corresponding measurement for any piece of equipment that is at that node so I set it zero.
I would expect there would be a line segment for S2 at these nodes but there is not. Is that assumption correct?  Should I be doing something else to get the node voltages?
 
I notice that the measurements defined in the model_dict.json for x2748140a versus x3066821a are different in that they there is a measurement for “s2” but is for the same ConductingEquipment_mRID so maybe it is an issue with the measurements being added?
 
grep ':"x2748140a' model_dict.json
{"name":"ACLineSegment_tpx391995a0_Voltage","mRID":"_23b96ebe-94c1-4fea-9359-255feca37b9d","ConductingEquipment_mRID":"_A02A838B-D41D-4C07-B877-79A541930DBB","Terminal_mRID":"_D6648A10-6AAD-4910-A179-51028BEB73C4","measurementType":"PNV","phases":"s1","MeasurementClass":"Analog","ConductingEquipment_type":"ACLineSegment","ConductingEquipment_name":"tpx391995a0","ConnectivityNode":"x2748140a","SimObject":"tpx_tpx391995a0"},
{"name":"ACLineSegment_tpx391995a0_Power","mRID":"_bdfd1e1a-6299-4f9d-b959-1b6c24a173fb","ConductingEquipment_mRID":"_A02A838B-D41D-4C07-B877-79A541930DBB","Terminal_mRID":"_D6648A10-6AAD-4910-A179-51028BEB73C4","measurementType":"VA","phases":"s1","MeasurementClass":"Analog","ConductingEquipment_type":"ACLineSegment","ConductingEquipment_name":"tpx391995a0","ConnectivityNode":"x2748140a","SimObject":"tpx_tpx391995a0"},
 
grep ':"x3066821a' model_dict.json
{"name":"ACLineSegment_tpx391994a0_Voltage","mRID":"_27dbda79-217a-4617-86ac-3d0ce003bcbd","ConductingEquipment_mRID":"_5151D5FB-F797-43B7-AACF-7B334DAAAE1D","Terminal_mRID":"_178516FF-0747-4569-8740-55387400604F","measurementType":"PNV","phases":"s1","MeasurementClass":"Analog","ConductingEquipment_type":"ACLineSegment","ConductingEquipment_name":"tpx391994a0","ConnectivityNode":"x3066821a","SimObject":"tpx_tpx391994a0"},
{"name":"ACLineSegment_tpx391994a0_Power","mRID":"_5e41970a-1c08-45f1-af8e-106d1d63e612","ConductingEquipment_mRID":"_5151D5FB-F797-43B7-AACF-7B334DAAAE1D","Terminal_mRID":"_178516FF-0747-4569-8740-55387400604F","measurementType":"VA","phases":"s2","MeasurementClass":"Analog","ConductingEquipment_type":"ACLineSegment","ConductingEquipment_name":"tpx391994a0","ConnectivityNode":"x3066821a","SimObject":"tpx_tpx391994a0"},
{"name":"ACLineSegment_tpx391994a0_Power","mRID":"_976d23f5-48f4-4d8e-901f-0e530f1595c8","ConductingEquipment_mRID":"_5151D5FB-F797-43B7-AACF-7B334DAAAE1D","Terminal_mRID":"_178516FF-0747-4569-8740-55387400604F","measurementType":"VA","phases":"s1","MeasurementClass":"Analog","ConductingEquipment_type":"ACLineSegment","ConductingEquipment_name":"tpx391994a0","ConnectivityNode":"x3066821a","SimObject":"tpx_tpx391994a0"},
{"name":"ACLineSegment_tpx391994a0_Voltage","mRID":"_ad8c3df3-56ac-447a-bfde-e00d64d9a7c0","ConductingEquipment_mRID":"_5151D5FB-F797-43B7-AACF-7B334DAAAE1D","Terminal_mRID":"_178516FF-0747-4569-8740-55387400604F","measurementType":"PNV","phases":"s2","MeasurementClass":"Analog","ConductingEquipment_type":"ACLineSegment","ConductingEquipment_name":"tpx391994a0","ConnectivityNode":"x3066821a","SimObject":"tpx_tpx391994a0"},
 
The line originally defined is :
https://github.com/GRIDAPPSD/Powergrid-Models/blob/develop/blazegraph/test/dss/WSU/TriplexLines.dss
New Line.Tpx391995A0     Bus1=X2748140A.1.2  Bus2=SX2748140A.1.2   linecode=4/0Triplex  length=50 units=ft
 
The model_base.dss line segment that is generated for X2748140A is:
new Line.tpx391995a0 phases=2 bus1=x2748140a.1 bus2=sx2748140a.1 length=50.001 linecode=4/0triplex units=ft
~ normamps=156.00 emergamps=195.00

AAndersn, 2020-10-26:
@Jeffrey-Simpson Please review if https://github.com/GRIDAPPSD/Powergrid-Models/pull/83 fixes the issue.

tonya1, 2020-11-13:
@Jeffrey-Simpson Can you test this container?  gridappsd/blazegraph:s2

Jeffrey-Simpson, 2020-11-16:
I test with the develop branch late Thursday and I had measurements for the s2 phases that were missing before and my app did not report any zero (or missing)  measurements which is good so I think the issue can be close.

If I try with that blazegraph:s2 container I get this issue though as an FYI:

Creating gridappsd-docker_proven_1     ... done

Creating gridappsd-docker_influxdb_1   ... done

Creating gridappsd-docker_redis_1      ... done

Creating gridappsd-docker_mysql_1      ... done

Creating gridappsd-docker_blazegraph_1 ... done

Creating gridappsd-docker_gridappsd_1  ... done

Creating gridappsd-docker_viz_1         ... done

Creating gridappsd-docker_test_app_1    ... done

Creating gridappsd-docker_sample_app_1  ... done

Creating gridappsd-docker_derdispatch_1 ... done

Getting blazegraph status

Checking blazegraph data

Blazegrpah data available (1958170)

Getting viz status

Containers are running

Connecting to the gridappsd container

docker exec -it gridappsd-docker_gridappsd_1 /bin/bash

OCI runtime exec failed: exec failed: container_linux.go:349: starting container process caused "exec: \"/bin/bash\": stat /bin/bash: no such file or directory": unknown

Thanks,
-Jeff

closed 2020-11-17 by poorva1209