GRIDAPPSD/gridappsd-forum #80: Query Historical Sensor Service Data

shpoudel, 2020-10-06:
@poorva1209 @craig8 : I was trying to use the sensor service data made available through gridappsd/influxdb:historical. It pulls 29.99 GB of an image (./run.sh) and while extracting it throws an error saying **no space left**. I see my system has a total of around 51GB space available. Any thoughts on this? Do the ~30GB data have all the simulation output saved for a week?

Also, I have a question regarding the pre-populated historical data generated from sensor service. When you run the simulation and store the historical data, can we select some object types and save the measurements associated with only those object types? I am asking this because, for any application, the entire simulation output might not be of use and if we want to store the entire simulation output for 2-3 weeks, we might always have a memory constraint.

[image]

shpoudel, 2020-10-14:
@poorva1209 @craig8 

I am following up on the same issue here. I pulled the updated image (gridappsd/influxdb:historical_1day). I used the following request with some filters. I am attaching a part of the response as well (first three dictionaries). I am expecting this: For one time step, there should be two measurements as I have two different mrids in the filter. But, we don't see such a pattern in the image. The alternative timestamp makes a 15 min time interval (1373551484 - 1373552384 = 900). Am I missing something here?

{'queryMeasurement': 'gridappsd-sensor-simulator', 
'queryFilter': {'simulation_id': '1905271421', 'starttime': '1373551205', 'endtime': '13736655590', 'measurement_mrid': ['_3fe87015-7e58-4271-ae78-523dc47b47ad', '_20f07f38-1160-4ca8-be34-f237d4f37b1b']}, 
'responseFormat': 'JSON'}

[image]

poorva1209, 2020-10-15:
Sensor service introduces missing data to inject realism so that could be one reason. But let me check influxdb directly for these mrids and timestamps and I will get back to you.

poorva1209, 2020-11-24:
@shpoudel Is this issue resolved in new historical data since we are not using randomization anymore.

shpoudel, 2020-11-24:
This can be closed. However, I have another related issue opened on the board.

https://github.com/GRIDAPPSD/GOSS-GridAPPS-D/issues/1390

closed 2020-11-24 by poorva1209