GRIDAPPSD/gridappsd-forum #81: PowerGrid Model API throws error with CSV, XML output

AAndersn, 2020-10-08:
Getting an `'error': 'Invalid json returned'` message from the PowerGrid Model API when requesting a response with a resultFormat other than JSON.

Passing a request with a JSON output format works correctly:

```
message = '{"requestType": "QUERY_MODEL_NAMES", "resultFormat": "JSON"}'
gridappsd_conn.get_response(topic, message)
```

This returns

```
{'data': {'modelNames': ['_49AD8E07-3BF9-A4E2-CB8F-C3722F837B62',
   '_4F76A5F9-271D-9EB8-5E31-AA362D86F2C3',
   '_503D6E20-F499-4CC7-8051-971E23D0BF79',
   '_5B816B93-7A5F-B64C-8460-47C17D6E4B0F',
   '_67AB291F-DCCD-31B7-B499-338206B9828F',
   '_77966920-E1EC-EE8A-23EE-4EFD23B205BD',
   '_9CE150A8-8CC5-A0F9-B67E-BBD8C79D3095',
   '_AAE94E4A-2465-6F5E-37B1-3E72183A4E44',
   '_C1C3E687-6FFD-C753-582B-632A27E28507',
   '_E407CBB6-8C8D-9BC9-589C-AB83FBF0826D']},
 'responseComplete': True,
 'id': '367747162'}
```

If I change result format to CSV or XML, I get

```
message = '{"requestType": "QUERY_MODEL_NAMES", "resultFormat": "CSV"}'
gridappsd_conn.get_response(topic, message)
```
This returns an error and then the requested data:
```
{'error': 'Invalid json returned',
 'header': {'content-length': '430',
  'expires': '0',
  'destination': '/temp-queue/response.20201008Oct1432',
  'GOSS_SUBJECT': 'system',
  'subscription': '809356',
  'priority': '4',
  'transformation': 'TEXT',
  'GOSS_HAS_SUBJECT': 'true',
  'message-id': 'ID:f516ef737acc-39375-1602107460711-4:9:-1:1:34',
  'persistent': 'true',
  'timestamp': '1602188072696'},
 'message': '{"data":_49AD8E07-3BF9-A4E2-CB8F-C3722F837B62,_4F76A5F9-271D-9EB8-5E31-AA362D86F2C3,_503D6E20-F499-4CC7-8051-971E23D0BF79,_5B816B93-7A5F-B64C-8460-47C17D6E4B0F,_67AB291F-DCCD-31B7-B499-338206B9828F,_77966920-E1EC-EE8A-23EE-4EFD23B205BD,_9CE150A8-8CC5-A0F9-B67E-BBD8C79D3095,_AAE94E4A-2465-6F5E-37B1-3E72183A4E44,_C1C3E687-6FFD-C753-582B-632A27E28507,_E407CBB6-8C8D-9BC9-589C-AB83FBF0826D,"responseComplete":true,"id":"1632321121"}'}
```

Same thing with XML:

```
message = '{"requestType": "QUERY_MODEL_NAMES", "resultFormat": "XML"}'
gridappsd_conn.get_response(topic, message)
```

```
{'error': 'Invalid json returned',
 'header': {'content-length': '561',
  'expires': '0',
  'destination': '/temp-queue/response.20201008Oct1518',
  'GOSS_SUBJECT': 'system',
  'subscription': '849150',
  'priority': '4',
  'transformation': 'TEXT',
  'GOSS_HAS_SUBJECT': 'true',
  'message-id': 'ID:f516ef737acc-39375-1602107460711-4:9:-1:1:35',
  'persistent': 'true',
  'timestamp': '1602188118761'},
 'message': '{"data":<?xml version="1.0" encoding="UTF-8" standalone="no"?>\n<modelNames>\n  <_49AD8E07-3BF9-A4E2-CB8F-C3722F837B62/>\n  <_4F76A5F9-271D-9EB8-5E31-AA362D86F2C3/>\n  <_503D6E20-F499-4CC7-8051-971E23D0BF79/>\n  <_5B816B93-7A5F-B64C-8460-47C17D6E4B0F/>\n  <_67AB291F-DCCD-31B7-B499-338206B9828F/>\n  <_77966920-E1EC-EE8A-23EE-4EFD23B205BD/>\n  <_9CE150A8-8CC5-A0F9-B67E-BBD8C79D3095/>\n  <_AAE94E4A-2465-6F5E-37B1-3E72183A4E44/>\n  <_C1C3E687-6FFD-C753-582B-632A27E28507/>\n  <_E407CBB6-8C8D-9BC9-589C-AB83FBF0826D/>\n</modelNames>\n,"responseComplete":true,"id":"84071283"}'}
```

MaximeBaudette, 2021-03-24:
This has been fixed for the XML version

AAndersn, 2022-02-05:
XML works correctly.
CSV version no longer throws error, but returns an empty dict {} object