GRIDAPPSD/gridappsd-forum #82: Voltage plot in display not working

therahuljha, 2020-10-17:
@poorva1209
I am observing that the new voltage plot is unable to reach the steady-state when there is a change in voltage control devices status. It gets stuck after the change as observed in the attached figure. 

[image]

poorva1209, 2020-10-17:
@bibiraju @nhuyvan Can you check if values are correct for these plots? If so, we can talk to Tom or Andy to check values.

nhuyvan, 2020-10-19:
How do I change the voltage control device statuses? And what should I expect?

shpoudel, 2020-11-06:
This hasn't been resolved so I am moving this to In Progress.
@nhuyvan  The values are correct. But, when the value gets changed, it doesn't settle to a new value. For example, open or close a switch, then you will see the effect when you plot its status.

nhuyvan, 2020-11-25:
@shpoudel I made a recording of a simulation against ieee123 https://drive.google.com/file/d/1q0fZgjcfqguMxuKm0Zp0jzmgc_alqNnA/view?usp=sharing (Hopefully, you are able to view it). I saw that when I opened/closed switch "sw4", the plots were changing, so I wasn't sure if this was correct or not, I am confused by what you said "But, when the value gets changed, it doesn't settle to a new value", can you elaborate? I can get on a call with if that's easier for you

shpoudel, 2020-11-25:
In the video, if you look closely, the plots are not settled after your second open command. It is expected in the demo you have that the voltage plot should be back with three parallel lines after you again close that switch. Does this make sense?

AAndersn, 2020-12-11:
Plot behavior now correct. Closing issue.

[image]

closed 2020-12-11 by AAndersn