GRIDAPPSD/gridappsd-forum #83: Wrong 'value' in alarm tab for capacitors

shpoudel, 2020-10-27:
For a capacitor, it is observed that when a close command is given, the alarm tab shows value as open. The logic for switch and capacitor is just the opposite which might be the cause for incorrect display in alarm.

poorva1209, 2021-01-04:
I changed alarm service to read capacitor Open - 0 and Open -1. 
We need to change this in Viz also. 
Will make a pull request for @shpoudel to test when both alarm and Viz changes are complete.

bibiraju, 2021-01-04:
@poorva1209 I have created a PR to change the value that Viz sends for Open and Close of capacitors. Viz now sends 0 for Open and 1 for Close. But I noticed that the value in Alarm tab is populated as received by the message from Platform. See below. So this message may need to be modified as well for the correct values.

[{"equipment_mrid": "_3DE55D2D-34D3-487E-9D6E-3A4DB1E38E47", "value": "Close", "created_by": "system", "equipment_name": "c90b", "phases": "B", "timestamp": 1609800759}] 

closed 2021-01-05 by poorva1209